introlab/rtabmap_ros #529: Rtabmap building from source conflict errors

derektan95, 2021-01-20:
Hi @matlabbe , 

I have a question about building this package from source, in order to use GTSAM optimizer and libpointmatcher. 

I made sure to follow all your instructions in the README.md. By firstly installing and uninstalling the binaries, these libopencv libraries are installed. These libraries are installed. 

```
ii  libopencv                                    4.1.1-2-gd5a58aa75                               arm64        Open Computer Vision Library
ii  libopencv-calib3d3.2:arm64                   3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Camera Calibration library
ii  libopencv-contrib3.2:arm64                   3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision contrlib library
ii  libopencv-core3.2:arm64                      3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision core library
ii  libopencv-dev                                4.1.1-2-gd5a58aa75                               arm64        Development files for Open Source Computer Vision Library
ii  libopencv-features2d3.2:arm64                3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann3.2:arm64                     3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-highgui3.2:arm64                   3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision High-level GUI and Media I/O library
ii  libopencv-imgcodecs3.2:arm64                 3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Image Codecs library
ii  libopencv-imgproc3.2:arm64                   3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Image Processing library
ii  libopencv-ml3.2:arm64                        3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Machine Learning library
ii  libopencv-objdetect3.2:arm64                 3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Object Detection library
ii  libopencv-photo3.2:arm64                     3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision computational photography library
ii  libopencv-shape3.2:arm64                     3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision shape descriptors and matchers library
ii  libopencv-stitching3.2:arm64                 3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision image stitching library
ii  libopencv-superres3.2:arm64                  3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Super Resolution library
ii  libopencv-video3.2:arm64                     3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Video analysis library
ii  libopencv-videoio3.2:arm64                   3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision Video I/O library
ii  libopencv-videostab3.2:arm64                 3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision video stabilization library
ii  libopencv-viz3.2:arm64                       3.2.0+dfsg-4ubuntu0.1                            arm64        computer vision 3D data visualization library

``` 

I did not install opencv by source, as i did not need the SIFT or SURF features. I proceeded with installed rtabmap package in my root folder, with the INSTALL-PREFIX set to a default path. However, I faced these errors when i performed catkin_make in my main workspace with the rtabmap_ros folder in it. 

```
/usr/bin/ld: warning: libopencv_photo.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_photo.so.3.2
/usr/bin/ld: warning: libopencv_stitching.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_stitching.so.3.2
/usr/bin/ld: warning: libopencv_video.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_video.so.3.2
/usr/bin/ld: warning: libopencv_videoio.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_videoio.so.3.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_calib3d.so.3.2
/usr/bin/ld: warning: libopencv_features2d.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_features2d.so.3.2
/usr/bin/ld: warning: libopencv_core.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_core.so.3.2
```

Yet at the same time, these errors occured. 

```
/usr/bin/ld: warning: libopencv_core.so.3.2, needed by /home/unitree/unitree_a1_repo/catkin_ws/devel/lib/librtabmap_ros.so, may conflict with libopencv_core.so.4.1
```

It seems like the installed libopencv files requires both version 3/4 shared object files, which are part of the initial installation Install and uninstall of rtabmap binaries. I wonder if this will cause issues along the way? Thanks!

thohemp, 2021-01-20:
You have opencv 4.1 and 3.2 installed. But you should not mix them up. 
Change this line: https://github.com/introlab/rtabmap/blob/master/CMakeLists.txt#L212
to 
`FIND_PACKAGE(OpenCV 3.2 REQUIRED QUIET)` to specify the version.

Make the same change for rtabmap_ros and try again.

derektan95, 2021-01-20:
Hi @thohemp ,

Thanks for your tip. What I'm confused by is this: 

There wasn't any openCV installed prior (I performed a sudo apt remove libopencv*). I performed the installation of rtab-map from binaries, and then removed it as per instructions. However, it seems like this step installs 2 versions of opencv, as you can see from the log posted in the first post. 

I think your fix might solve the issue. But I'm curious if the instructions will cause issues for others who does not use this fix?

thohemp, 2021-01-20:
I can't tell how version 4.1 got on your system. But any ros related stuff normally uses version 3.2 and I am pretty rtabmap does the same. Perhaps, version 4.1 was introduced by another package not related to this.

matlabbe, 2021-01-20:
Are you on a jetson? In latest jetpack they changed default libopencv-dev to use version 4 instead of 3.2, while ros packages are still linked to 3.2. Here more [info](https://github.com/introlab/rtabmap/issues/427#issuecomment-758897302).

derektan95, 2021-01-20:
Hi @thohemp and @matlabbe, 

Thank you both for your comments. That is a good catch. I'm actually using a Jetson TX2, which is why I'm facing the issue of OpenCV being version 4 by default. I've fixed this by carefully fixing this by building OpenCV from source, along with a few other dependencies. I will post my solutions as a possible guide once I'm on a computer. Thank you very much!

derektan95, 2021-01-21:
#### Install misc dependencies

```
sudo apt-get install ros-melodic-rtabmap ros-melodic-rtabmap-ros
sudo apt remove ros-melodic-rtabmap ros-melodic-rtabmap-ros       # Uninstall again
```

#### Installing OpenCV dependencies

```
cd ~/Libraries
git clone https://github.com/opencv/opencv          # Master branch
cd opencv
mkdir build && cd build
cmake -D CATKIN_INSTALL_PREFIX=/usr/local ..
make -j8 
sudo make install                                   # Install to usr/local directory
```

#### Installing OpenCV Bridge dependencies
These packages be cloned from `https://github.com/ros-perception/vision_opencv`. However, they have to be modified to fit to our use, if not there will be compile errors. For ROS Melodic, use the noetic branch and change python dependencies in CMakeLists.txt, or use the melodic branch and change some function headers (https://cyaninfinite.com/ros-cv-bridge-with-python-3/).

```
cd ~/Libraries
git clone https://github.com/ros-perception/vision_opencv
cd vision_opencv/cv_bridge                          
mkdir build && cd build
cmake -D CATKIN_INSTALL_PREFIX=/usr/local ..
make -j8
sudo make install                                   # Install to usr/local directory

cd ~/Libraries
cd vision_opencv/image_geometry                     
mkdir build && cd build
cmake -D CATKIN_INSTALL_PREFIX=/usr/local ..
make -j8
sudo make install                                   # Install to usr/local directory
```

#### Installing RTAB packages

```
# Installing RTAB Library
cd ~/Libraries
git clone https://github.com/introlab/rtabmap       # Master branch
cd rtabmap
mkdir build && cd build
cmake -D OpenCV_DIR=/usr/local/lib/cmake/opencv4  -D CATKIN_INSTALL_PREFIX=/usr/local ..
make -j8
sudo make install

# Installing RTAB ROS package
cd ~/unitree_a1_repo/catkin_ws/src
git clone https://github.com/introlab/rtabmap_ros
```

#### Linking RTAB-Map to newly compiled libraries
Add these 2 lines of code to specify where to link to the `cv_bridge` and `image_geometry` libraries, at the top of RTAB ROS's CMakeLists.txt. After this, you can compile the RTAB-Map package. It depends on where your cmake instruction files are located once these packages are installed.

```
set(cv_bridge_DIR /usr/local/share/cv_bridge/cmake)
set(image_geometry_DIR /usr/local/share/image_geometry/cmake)
#set(image_geometry_DIR /usr/local/lib/cmake/opencv4)         # Not sure if needed
```

derektan95, 2021-01-21:
Hi @matlabbe, 

Speaking of compilation from source, I've added the GTSAM and libpointmatcher functionality. While GTSAM is an option that we have to specify to use, how about libpointmatcher? Is it used by default if we build rtab-map with it? Thanks!

matlabbe, 2021-01-21:
Yes, libpointmatcher is used by default if rtabmap is built with it.
```
$ rtabmap --params | grep Icp/PM
Param: Icp/PM = "true"                                     [Use libpointmatcher for ICP registration instead of PCL's implementation.]
```

derektan95, 2021-01-27:
Hi @matlabbe, 

Thank you so much for all these clarification. Appreciate it!

-Derek

closed 2021-01-27 by derektan95