introlab/rtabmap_ros #530: Rtabmapviz on separate computer

derektan95, 2021-01-20:
Hi @matlabbe, 

I would like to clarify a confusion I currently have with rtabmapviz. I've installed Rtab-map on my Nvidia Jetson TX2, but I would like to view the results on another computer with an Ubuntu 16.04 (ROS Kinetic). While I installed Rtab-map on the TX2 from source, I installed Rtab-map on my computer via binaries for convenience sake, since it's purpose is simply to run rtabmapviz. 

My question is, would any advanced settings done on my rtabmapviz run on my other computer have any effects on the Rtab-map mapping node running on the Nvidia board? For instance, if the launch file on the Nvidia board is set to 1Hz, and if I were to change that to another value on Rtabmapviz running on my other computer, would this change override the initial settings on the launch file? 

By extension, I do not have settings like GTSAM installed on my other computer since it's built my binaries. Would the lack of the GTSAM functionality on the other computer override the GTSAM-enabled mapping node running on the Nvidia Board, since this setting can be tweaked either via code or from rtabmapviz? 

I have set the following code in my mapping launch file based on your graph optimization wiki tutorial: 

'''
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap">
   <param name="RGBD/OptimizeStrategy" type="string" value="2"/> 
   <param name="RGBD/OptimizeRobust" type="string" value="true"/>
   <param name="RGBD/OptimizeMaxError" type="string" value="0"/> 
   ...
</node>
'''

Thank you again so much for your help!

matlabbe, 2021-02-12:
Yes, you have to make sure that computers can correctly see each other on both side. This can be done explicitly by setting the `ROS_IP` on both computers.

Then, to make the advanced settings working, launch rtabmapviz in same namespace than rtabmap so it can see rtabmap parameters when opening Preferences. For example, if rtabmap is launched in rtabmap namespace, launch rtabmapviz like this:
```
export ROS_NAMESPACE=rtabmap
rosrun rtabmap_ros rtabmapviz
```

derektan95, 2021-02-26:
Hi Matlabbe, 

Thanks! It seems that by setting all advanced parameters in the launch file, another computer connected to the same ROS network can launch rtabmapviz with the same advanced parameters. Appreciate your reply. 

- Derek

closed 2021-02-26 by derektan95