introlab/rtabmap_ros #533: Compilation Error on ROS Noetic

ElClopitan, 2021-01-26:
[image]
Seem to have a compilation error on ROS Noetic. I am using noetic-devel for rtabmap built from source and rtabmap_ros.

matlabbe, 2021-01-26:
The noetic-devel branch of rtabmap doesn't match the version you have, lines 993 and 994 should be:
https://github.com/introlab/rtabmap_ros/blob/b7a66e41f16ae6f450590ef6a2efd654970acd02/src/MsgConversion.cpp#L993-L994

make sure you are on noetic-devel branch with latest changes.

ElClopitan, 2021-01-26:
Thank you! You were right that I wasn't up to date but I am still getting a different error that is outside rtabmap.
[image]
If you have any advice for this it would be greatly appreciated!

matlabbe, 2021-01-26:
This is a teb_local_planner issue. g2o has been updated their interface to c++14 recently, teb_local_planner may not have updated yet. I would recommend to build/install this version of g2o: https://github.com/RainerKuemmerle/g2o/releases/tag/20201223_git

ElClopitan, 2021-01-26:
Okay. Thank you!

closed 2021-01-26 by ElClopitan