introlab/rtabmap_ros #534: Superglue ROS

42Max42, 2021-01-28:
Hi,
I tryed Superglue with superpoint torch with rtabmap 0.20.7 and the results are great.
Then I tryed to run the same thing in ROS noetic (from tag 0.20.7-noetic), but when it loads Superpoint to match the first pair, it hangs for ever. I am using libtorch 1.7.1 cuda 11 (downloaded from pytorch.org, and uncompressed in /data/libtorch) and pytorch 1.7.1+cu110 installed with pip. It works fine using Superpoint in ROS with an other Vis/CorNNType.
Here is my launch file:
```
<launch>
  <group ns="rtabmap">
    <!-- Visual SLAM -->
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" args="--delete_db_on_start">

      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgb"    type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="subscribe_laserScan" type="bool" value="false"/>

      <remap from="rgbd_image"       to="/rgbd_image/compressed_relay"/>

      <param name="Vis/EstimationType" type="string" value="1"/>
      <param name="Vis/MinInliers" type="string" value="20"/>

      <param name="Kp/DetectorStrategy"    type="string" value="11"/>
      <param name="Vis/FeatureType" type="string" value="11"/>
      <param name="SuperPoint/Cuda"        type="string" value="true"/>
      <param name="SuperPoint/ModelPath" type="string" value="$(find rtabmap_ros)/cfg/superpoint.pt"/>

      <param name="Vis/CorNNType"      type="string" value="6"/>
      <param name="PyMatcher/Model" type="string" value="indoor"/>
      <param name="PyMatcher/Path" type="string" value="/data/rtabmap/corelib/src/pymatcher/rtabmap_superglue.py"/>
      <param name="PyMatcher/Threshold" type="string" value="0.2"/>
      <param name="PyMatcher/Iterations" type="string" value="20"/>
      <param name="PyMatcher/Cuda" type="string" value="true"/>

      <remap from="odom" to="/odom"/>

      <param name="frame_id"    type="string" value="/base_link"/>
      <param name="queue_size"  type="int"    value="30"/>
      <param name="approx_sync" type="bool"   value="true"/>

      <param name="Mem/IncrementalMemory" type="string" value="true"/>
    </node>

    <!-- Visualisation RTAB-Map -->
    <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">

      <param name="subscribe_rgb"    type="bool"   value="false"/>
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_stereo"    type="bool"   value="false"/>
      <param name="subscribe_laserScan" type="bool" value="false"/>

      <param name="subscribe_rgbd"   type="bool"   value="true"/>

      <param name="subscribe_odom_info" type="bool" value="false"/>

      <param name="queue_size"          type="int"    value="10"/>
      <param name="frame_id"            type="string" value="base_link"/>

      <remap from="rgbd_image"       to="/rgbd_image/compressed_relay"/>
      <remap from="odom"              to="/odom"/>
      <param name="approx_sync" type="bool"   value="true"/>
    </node>

  </group>

</launch>
```
I also had to
```
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/data/libtorch/lib/
```
to get rtabmap running with superpoint.
What is wrong?

Thanks in advance.

matlabbe, 2021-01-28:
Hi, you may add `--udebug` to `args="--delete_db_on_start"` to see more info. You can also add some print like [this](https://github.com/introlab/rtabmap/blob/839045a538396eb56e21ae472d9a4a20f8404437/corelib/src/pymatcher/rtabmap_superglue.py#L25) in the python code to see if the program is hanging somewhere in the python.

Note also that latest rtabmap version has some refactoring to python interface, which may help for the problem.

42Max42, 2021-01-29:
Hi,
Thank you for your answer.
I found that it is stuck here: https://github.com/magicleap/SuperGluePretrainedNetwork/blob/c0626d58c843ee0464b0fa1dd4de4059bfae0ab4/models/superglue.py#L224
I have checked the path is correct. It hangs when the device is gpu or cpu.
If I comment this line (which loads the weights) it passes https://github.com/introlab/rtabmap/blob/839045a538396eb56e21ae472d9a4a20f8404437/corelib/src/pymatcher/rtabmap_superglue.py#L39 but hangs later during matching and with gpu it remains stuck at this line.

With the new version (rtabmap and rtabmap_ros to the latest master), with just rtabmap (same config but with the correct path for  rtabmap_superglue.py)
```
Program started...
SuperGlue python init()
[ WARN] (2021-01-29 14:03:25.544) RtabmapThread.cpp:632::addData() Data buffer is full, the oldest data is removed to add the new one.
[ WARN] (2021-01-29 14:03:26.752) RtabmapThread.cpp:632::addData() Data buffer is full, the oldest data is removed to add the new one.
[ WARN] (2021-01-29 14:03:27.761) RtabmapThread.cpp:632::addData() Data buffer is full, the oldest data is removed to add the new one.
[ WARN] (2021-01-29 14:03:28.768) RtabmapThread.cpp:632::addData() Data buffer is full, the oldest data is removed to add the new one.
[ WARN] (2021-01-29 14:03:29.807) RtabmapThread.cpp:632::addData() Data buffer is full, the oldest data is removed to add the new one.
```
with both cpu and gpu, and it doesn't match anything.

But with rtabmap-matcher it works fine for matching.

In ros (pymatcher path adapted) it crashes at start without any error in logs or display.

```
PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /rtabmap/rtabmap/Kp/DetectorStrategy: 11
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/PyMatcher/Cuda: false
 * /rtabmap/rtabmap/PyMatcher/Iterations: 20
 * /rtabmap/rtabmap/PyMatcher/Model: indoor
 * /rtabmap/rtabmap/PyMatcher/Path: /data/rtabmap/cor...
 * /rtabmap/rtabmap/PyMatcher/Threshold: 0.2
 * /rtabmap/rtabmap/SuperPoint/Cuda: false
 * /rtabmap/rtabmap/SuperPoint/ModelPath: /data/ws/src/rtab...
 * /rtabmap/rtabmap/Vis/CorNNType: 6
 * /rtabmap/rtabmap/Vis/EstimationType: 1
 * /rtabmap/rtabmap/Vis/FeatureType: 11
 * /rtabmap/rtabmap/Vis/MinInliers: 20
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/frame_id: /base_link
 * /rtabmap/rtabmap/queue_size: 30
 * /rtabmap/rtabmap/rgbd_cameras: 1
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_laserScan: False
 * /rtabmap/rtabmap/subscribe_rgb: False
 * /rtabmap/rtabmap/subscribe_rgbd: True
 * /rtabmap/rtabmapviz/approx_sync: True
 * /rtabmap/rtabmapviz/frame_id: base_link
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/rgbd_cameras: 1
 * /rtabmap/rtabmapviz/subscribe_depth: False
 * /rtabmap/rtabmapviz/subscribe_laserScan: False
 * /rtabmap/rtabmapviz/subscribe_odom_info: False
 * /rtabmap/rtabmapviz/subscribe_rgb: False
 * /rtabmap/rtabmapviz/subscribe_rgbd: True
 * /rtabmap/rtabmapviz/subscribe_stereo: False

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

ROS_MASTER_URI=http://localhost:11311

process[rtabmap/rtabmap-1]: started with pid [6097]
process[rtabmap/rtabmapviz-2]: started with pid [6098]
[ INFO] [1611930266.482019774]: Starting node...
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[ INFO] [1611930266.517588281]: rtabmapviz: Using configuration from "/data/ws/src/rtabmap_ros/launch/config/rgbd_gui.ini"
[... 327 lines cut ...]
[ WARN] [1611930267.494454088]: Parameter Vis/Iterations not found
[ WARN] [1611930267.494646450]: Parameter Vis/MaxDepth not found
[ WARN] [1611930267.494837748]: Parameter Vis/MaxFeatures not found
[ WARN] [1611930267.495028357]: Parameter Vis/MeanInliersDistance not found
[ WARN] [1611930267.495215366]: Parameter Vis/MinDepth not found
[ WARN] [1611930267.495400940]: Parameter Vis/MinInliers not found
[ WARN] [1611930267.495584774]: Parameter Vis/MinInliersDistribution not found
[ WARN] [1611930267.495770547]: Parameter Vis/PnPFlags not found
[ WARN] [1611930267.495957332]: Parameter Vis/PnPRefineIterations not found
[ WARN] [1611930267.496144843]: Parameter Vis/PnPReprojError not found
[ WARN] [1611930267.496330833]: Parameter Vis/RefineIterations not found
[ WARN] [1611930267.496516787]: Parameter Vis/RoiRatios not found
[ WARN] [1611930267.496726288]: Parameter Vis/SubPixEps not found
[ WARN] [1611930267.496950160]: Parameter Vis/SubPixIterations not found
[ WARN] [1611930267.497139322]: Parameter Vis/SubPixWinSize not found
[ WARN] [1611930267.497316464]: Parameter g2o/Baseline not found
[ WARN] [1611930267.497490546]: Parameter g2o/Optimizer not found
[ WARN] [1611930267.497676534]: Parameter g2o/PixelVariance not found
[ WARN] [1611930267.497847983]: Parameter g2o/RobustKernelDelta not found
[ WARN] [1611930267.498020416]: Parameter g2o/Solver not found
[ INFO] [1611930267.498034849]: Parameters read = 0
[rtabmap/rtabmap-1] process has died [pid 6097, exit code -11, cmd /data/ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start --udebug rgbd_image:=/rgbd_image/compressed_relay odom:=/odom __name:=rtabmap __log:=/root/.ros/log/e8acf39a-623c-11eb-8707-0492260e8743/rtabmap-rtabmap-2.log].
log file: /root/.ros/log/e8acf39a-623c-11eb-8707-0492260e8743/rtabmap-rtabmap-2*.log

```

matlabbe, 2021-02-12:
rtabmap node coudl be launched with gdb:
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB
then type "bt" when it crashes to get the backtrace.

SuperGlue has been tested only with the standalone version. I should update my computer to Noetic soon, I could take a look afterwards for ROS (I don't even want to try on melodic, to not messed up my config between python 2 and 3...).

robot-mh, 2022-02-08:
> SuperGlue has been tested only with the standalone version. I should update my computer to Noetic soon, I could take a look afterwards for ROS (I don't even want to try on melodic, to not mess up my config between python 2 and 3...).

How is it now？ SuperGlue + which ROS version？ @matlabbe

matlabbe, 2022-11-19:
See https://github.com/introlab/rtabmap/issues/896#issuecomment-1257085425 for ROS example. The tricky part is to install pytorch with cuda correctly.

closed 2022-11-19 by matlabbe