introlab/rtabmap_ros #537: RTABMap-ros symbol lookup error on ros2.

Marsfan, 2021-01-30:
Just finished compiling RTABMap and rtabmap-ros for a Jetson Nano 2GB with ros2-eloquent (its not fun, as gcc love using all the RAM). I tried to run some of the ROS demos and got some symbol lookup errors: 
```
On a whim, I tried running rtabmap as root using `sudo`. Suprisingly, this worked and the interface popped up. 
```bash

[INFO] [launch]: All log files can be found below /home/heimdallr/.ros/log/2021-01-29-21-57-58-847678-heimdallr.local-8590
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rtabmap-1]: process started with pid [8601]
[INFO] [rtabmapviz-2]: process started with pid [8602]
[rtabmap-1] /home/heimdallr/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap: symbol lookup error: /usr/local/lib/librtabmap_core.so.0.20: undefined symbol: _ZN3g2o7Factory12registerTypeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt10shared_ptrINS_32AbstractHyperGraphElementCreatorEE
[ERROR] [rtabmap-1]: process has died [pid 8601, exit code 127, cmd '/home/heimdallr/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap -d --ros-args --params-file /tmp/launch_params_vfsmjhou -r scan:=/scan'].
[rtabmapviz-2] /home/heimdallr/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmapviz: symbol lookup error: /usr/local/lib/librtabmap_core.so.0.20: undefined symbol: _ZN3g2o7Factory12registerTypeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt10shared_ptrINS_32AbstractHyperGraphElementCreatorEE
[ERROR] [rtabmapviz-2]: process has died [pid 8602, exit code 127, cmd '/home/heimdallr/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmapviz --ros-args --params-file /tmp/launch_params_72ffuoma -r scan:=/scan'].
```

So I tried running the standalone rtabmap interface, and I got this error message: 
```bash
rtabmap: symbol lookup error: /usr/local/lib/librtabmap_core.so.0.20: undefined symbol: _ZN3g2o7Factory12registerTypeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt10shared_ptrINS_32AbstractHyperGraphElementCreatorEE

Why might I be getting this errors, and why would the standalong rtabmap only be running as root?

matlabbe, 2021-02-12:
AbstractHyperGraphElementCreator is g2o stuff. Which g2o version is installed? Maybe you have more than one version, from which it is built on and which is used on runtime. Maybe using root would change the paths the linker is looking for. Normally, you should not have to be root to use rtabmap.

Marsfan, 2021-02-15:
I built g2o by cloning it from github. The set of commands I made when building was: 

          ```bash
          git clone https://github.com/RainerKuemmerle/g2o.git
          cd g2o
          mkdir build
          cd build
          cmake -DBUILD_WITH_MARCH_NATIVE=OFF -DG2O_BUILD_APPS=OFF -DG2O_BUILD_EXAMPLES=OFF -DG2O_USE_OPENGL=OFF ..
          make -j3
          sudo make install
          ```

When I then tried to build rtabmap (not rtabmap-ros), I got the following cmake warning, but it did compile.

       ```cmake
       Latest g2o version detected with c++11 interface (config file:
       /usr/local/include/g2o/config.h).  Make sure g2o is built with
       "-DBUILD_WITH_MARCH_NATIVE=OFF" to avoid segmentation faults caused by
       Eigen.
       ```

However, as you can see, I did build with the recommended argument. Do you think this could be the source of the problem, and if so, how would you recommend that I fix it?

matlabbe, 2021-02-26:
If you built g2o from source, uninstall g2o binaries to make sure the right libraries are loaded on runtime:
```
sudo apt remove ros-eloquent-libg2o
```

For the 3D map view empty on Jetson, this is an open issue largely discussed here: https://github.com/introlab/rtabmap/issues/427

Marsfan, 2021-02-28:
@matlabbe Thanks for the help. Unfortunately, our team decided to switch back to ROS1 as there is much more documentation for it, more packages available, and because we can then just use existing RTABMap binaries. Since this _technically_ resolved _our_ problems, I am closing the issue. Feel free to re-open it if you think it should stay open. 

Glad to know that the 3D view is not likely a problem on my end.

closed 2021-02-28 by Marsfan