introlab/rtabmap_ros #546: make error after building from source

jenn-yu, 2021-02-20:
I've built rtabmap from source per instructions here:
https://github.com/introlab/rtabmap_ros but running `catkin_make` is failing with:

[image]

This is exactly what I've done:

```
$ mkdir -p ~/catkin_ws/src
$ catkin_init_workspace
$ cd ..
$ catkin_make
$ source devel/setup.bash
$ source /opt/ros/kinetic/setup.bash
$ sudo apt install ros-kinetc-rtabmap ros-kinetic-rtabmap-ros
$ sudo apt remove ros-kinetic-rtabmap ros-kinetic-rtabmap-ros
$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake ..
$ make
$ sudo make install
$ cd ~/catkin_ws
$ git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
$ catkin_make -j1
```

matlabbe, 2021-02-21:
pluginlib is a standard ros package. The sourcing should be `/opt/ros` first, then workspace:
```
$ source /opt/ros/kinetic/setup.bash
$ source ~/catkin_ws/devel/setup.bash
```

pavelmarkov, 2021-02-24:
I changed the _line 31_ in _rtabmap_ros/src/nodelets/icp_odometry.cpp_ to **#include <pluginlib/class_loader.h>** and it worked.

matlabbe, 2021-02-24:
Thx for the workaround, though not sure why it didn't work with pluginlib kinetic version. The class_loader header form h to hpp has been done in this commit (April 2018): https://github.com/ros/pluginlib/commit/84b0c46be4c3f47cf995998db517cfa4700b1514

Well, the class_loader.h is still in [pluginlib noetic](https://github.com/ros/pluginlib/tree/noetic-devel/pluginlib/include/pluginlib) for backward compatibility, I will change it back in icp_odometry.

commit 4af79a01 (2021-02-24): Fixed #546
closed 2021-02-24 by matlabbe