introlab/rtabmap_ros #548: Rtabmap and RGBD odometry crashes on subscription

peredwardsson, 2021-02-23:
RTAB-Map and RGBD odometry crashes when they receive data on their inputs and they can start placing input into the world with the proper transforms.

The system in question is an Nvidia Jetson AGX Xavier with a ZED2 camera. Using an rgbd_sync nodelet, the rgb and depth outputs from the zed are joined to an rtabmap_ros/RGBDImage at /camera/rgbd_image. The camera itself is located at /camera/zed_camera/, with the various outputs in child topics.

I launch RTAB-Map and RGBD odometry in the same launch file, found here. Params are left default for both cases as I've found they make no difference.

localization.launch
```
<launch>  

  <arg name="frame_id"                default="base_link" />
  <arg name="rgb_topic"               default="/camera/zed_camera/rgb/image_rect_color" />
  <arg name="rgbd_topic"              default="/camera/rgbd_image" />
  <arg name="depth_topic"             default="/camera/zed_camera/depth/depth_registered" />
  <arg name="camera_info_topic"       default="/camera/zed_camera/rgb/camera_info" />
  <arg name="imu_topic"               default="/camera/zed_camera/imu/data" />
  <arg name="imu_ignore_acc"          default="true" />
  <arg name="imu_remove_gravitational_acceleration" default="true" />
  <arg name="icp_scan"                default="/scan_filtered"/>
  <arg name="use_icp_odometry"        default="false"/>

  <!-- Localization-only mode -->
  <arg name="localization"      default="false"/>
  <arg     if="$(arg localization)" name="rtabmap_args"  default=""/>
  <arg unless="$(arg localization)" name="rtabmap_args"  default=""/>  

  <group ns="rtabmap">

    <!-- Visual Odometry using RGBD camera -->
    <node pkg="rtabmap_ros" type="rgbd_odometry" name="visual_odometry" args="--delete_db_on_start">
      <!-- <rosparam file="$(find hugo_configs)/config/rtabmap_vo.yaml" /> -->
      <remap from="rgb/image"                            to="$(arg rgb_topic)"/>
      <remap from="rgbd_image"                           to="$(arg rgbd_topic)"/>
      <remap from="depth/image"                          to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info"                      to="$(arg camera_info_topic)"/>
      <remap from="odom"                                 to="/vo"/>
    </node>

    <!-- SLAM -->
    <node pkg="rtabmap_ros" type="rtabmap" name="rtabmap" args="--delete_db_on_start" output="screen">
      <!-- <rosparam file="$(find hugo_configs)/config/rtabmap.yaml" /> -->
      <remap from="rgb/image"                          to="$(arg rgb_topic)"/>
      <remap from="rgbd_image"                         to="$(arg rgbd_topic)"/>
      <remap from="depth/image"                        to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info"                    to="$(arg camera_info_topic)"/>
      <remap from="odom"                               to="/odom_1/filtered"/>
      <remap from="scan"                               to="/scan_filtered"/>

    </node>
  </group>

</launch>
```

In the startup process, whenever 1) rtabmap is ready for input 2) input is available and 3) transforms are available, the node crashes. It can be reliably reproduced.

Here is a full roslaunch while launching the above launch file, as well as a robot_state_publisher and the appropriate camera settings for a ZED2 camera.
```
SUMMARY
========

PARAMETERS
 * /camera/zed_camera/auto_exposure_gain: True
 * /camera/zed_camera/auto_whitebalance: True
 * /camera/zed_camera/base_frame: base_link
 * /camera/zed_camera/brightness: 4
 * /camera/zed_camera/cam_pitch: 0.0
 * /camera/zed_camera/cam_pos_x: 0.0
 * /camera/zed_camera/cam_pos_y: 0.0
 * /camera/zed_camera/cam_pos_z: 0.0
[... 402 lines cut ...]
[ INFO] [1614093129.720121007]: /rtabmap/rtabmap: subscribe_rgb = true
[ INFO] [1614093129.720489345]: /rtabmap/rtabmap: subscribe_stereo = false
[ INFO] [1614093129.720824850]: /rtabmap/rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1614093129.721057278]: /rtabmap/rtabmap: subscribe_odom_info = false
[ INFO] [1614093129.721262696]: /rtabmap/rtabmap: subscribe_user_data = false
[ INFO] [1614093129.721429776]: /rtabmap/rtabmap: subscribe_scan = false
[ INFO] [1614093129.721634075]: /rtabmap/rtabmap: subscribe_scan_cloud = false
[ INFO] [1614093129.721798691]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1614093129.722035919]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1614093129.722503078]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1614093129.722946877]: Setup rgb-only callback
[ INFO] [1614093130.495014719]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /camera/zed_camera/rgb/image_rect_color \
   /camera/zed_camera/rgb/camera_info
[rtabmap/rtabmap-2] process has died [pid 23122, exit code -11, cmd /home/albert/hugo_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/zed_camera/rgb/image_rect_color rgbd_image:=/camera/rgbd_image depth/image:=/camera/zed_camera/depth/depth_registered rgb/camera_info:=/camera/zed_camera/rgb/camera_info odom:=/odom_1/filtered scan:=/scan_filtered __name:=rtabmap __log:=/home/albert/.ros/log/64af9fe4-75d7-11eb-b129-061ec70036de/rtabmap-rtabmap-2.log].
```
Note that as soon as the packages subscribe, they crash. The output looks fine upon inspection, and images, for instance, can be viewed in rviz without issues.

The two lines contained in the visual-odometry log above look like this.
```
nvbuf_utils: Could not get EGL display connection
[ INFO] [1614092536.878487380]: Initializing nodelet with 8 worker threads.
```
I have had issues getting further logs, but given instructions I will be happy to provide further logs as necessary.

peredwardsson, 2021-02-24:
Running with --udebug produces these as the last messages:

```
/rtabmap/visual_odometry subscribed to:
   /camera/rgbd_image
[DEBUG] (2021-02-24 16:03:40.013) OdometryF2F.cpp:241::computeTransform() Update key frame
[DEBUG] (2021-02-24 16:03:40.013) RegistrationVis.cpp:282::computeTransformationImpl() Vis/MinInliers=20
[DEBUG] (2021-02-24 16:03:40.013) RegistrationVis.cpp:283::computeTransformationImpl() Vis/InlierDistance=0.100000
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:284::computeTransformationImpl() Vis/Iterations=300
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:285::computeTransformationImpl() Vis/EstimationType=1
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:286::computeTransformationImpl() Vis/ForwardEstOnly=1
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:287::computeTransformationImpl() Vis/EpipolarGeometryVar=0.100000
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:288::computeTransformationImpl() Vis/PnPReprojError=2.000000
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:289::computeTransformationImpl() Vis/PnPFlags=0
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:290::computeTransformationImpl() Vis/CorType=0
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:291::computeTransformationImpl() Vis/CorFlowWinSize=16
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:292::computeTransformationImpl() Vis/CorFlowIterations=30
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:293::computeTransformationImpl() Vis/CorFlowEps=0.010000
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:294::computeTransformationImpl() Vis/CorFlowMaxLevel=3
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:295::computeTransformationImpl() Vis/CorNNDR=0.800000
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:296::computeTransformationImpl() Vis/CorNNType=1
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:297::computeTransformationImpl() Vis/CorGuessWinSize=40
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:298::computeTransformationImpl() Vis/CorGuessMatchToProjection=0
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:299::computeTransformationImpl() Feature Detector = 8
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:300::computeTransformationImpl() guess=xyz=0.000000,0.000000,0.000000 rpy=0.000000,0.000000,0.000000
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:313::computeTransformationImpl() Input(0): from=0 words, 0 3D words, 0 words descriptors,  0 kpts, 0 kpts3D, 0 descriptors, image=1280x720 models=1 stereo=0
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:326::computeTransformationImpl() Input(0): to=0 words, 0 3D words, 0 words descriptors, 0 kpts, 0 kpts3D, 0 descriptors, image=0x0 models=0 stereo=0
[DEBUG] (2021-02-24 16:03:40.014) RegistrationVis.cpp:346::computeTransformationImpl() 
[DEBUG] (2021-02-24 16:03:40.022) util2d.cpp:1162::computeRoi() roi ratios = 0.000000, 0.000000, 0.000000, 0.000000
[DEBUG] (2021-02-24 16:03:40.022) util2d.cpp:1163::computeRoi() roi = 0, 0, 1280, 720
[DEBUG] (2021-02-24 16:03:40.022) util2d.cpp:1190::computeRoi() roi = 0, 0, 1280, 720
[rtabmap/visual_odometry-1] process has died [pid 332, exit code -11, cmd /home/albert/hugo_ws/devel/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start --udebug rgb/image:=/camera/zed_camera/rgb/image_rect_color rgbd_image:=/camera/rgbd_image depth/image:=/camera/zed_camera/depth/depth_registered rgb/camera_info:=/camera/zed_camera/rgb/camera_info odom:=/vo __name:=visual_odometry __log:=/home/albert/.ros/log/64af9fe4-75d7-11eb-b129-061ec70036de/rtabmap-visual_odometry-1.log].
log file: /home/albert/.ros/log/64af9fe4-75d7-11eb-b129-061ec70036de/rtabmap-visual_odometry-1*.log
```

And following that,

```
/rtabmap/rtabmap subscribed to (approx sync):
   /camera/zed_camera/rgb/image_rect_color \
   /camera/zed_camera/rgb/camera_info
[DEBUG] (2021-02-24 16:04:38.773) Rtabmap.cpp:999::process() 
[ INFO] (2021-02-24 16:04:38.774) Rtabmap.cpp:1059::process() getting data...
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:2456::getLastWorkingSignature() 
[ INFO] (2021-02-24 16:04:38.774) Rtabmap.cpp:1205::process() Updating memory...
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:773::update() 
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:782::update() pre-updating...
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:5404::cleanUnusedWords() Removing 0 words (dictionary size=0)...
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:786::update() time preUpdate=0.043869 ms
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:4092::createSignature() 
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:4255::createSignature() Start dictionary update thread
[DEBUG] (2021-02-24 16:04:38.774) VWDictionary.cpp:470::update() incremental=1
[DEBUG] (2021-02-24 16:04:38.774) FlannIndex.cpp:52::release() 
[DEBUG] (2021-02-24 16:04:38.774) FlannIndex.cpp:80::release() 
[DEBUG] (2021-02-24 16:04:38.774) Memory.cpp:4263::createSignature() Received kpts=0 kpts3D=0, descriptors=0 _useOdometryFeatures=true
[DEBUG] (2021-02-24 16:04:38.774) VWDictionary.cpp:681::update() Dictionary updated! (size=0 added=0 removed=0)
[DEBUG] (2021-02-24 16:04:38.774) VWDictionary.cpp:689::update() 
[ INFO] (2021-02-24 16:04:38.774) Memory.cpp:4312::createSignature() Extract features
[DEBUG] (2021-02-24 16:04:38.775) Memory.cpp:4361::createSignature() rawDescriptorsKept=1, pose=1, maxFeatures=500, visMaxFeatures=1000
[... cut ...]
[rtabmap/rtabmap-2] process has died [pid 333, exit code -11, cmd /home/albert/hugo_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start --udebug rgb/image:=/camera/zed_camera/rgb/image_rect_color rgbd_image:=/camera/rgbd_image depth/image:=/camera/zed_camera/depth/depth_registered rgb/camera_info:=/camera/zed_camera/rgb/camera_info odom:=/odom_1/filtered scan:=/scan_filtered __name:=rtabmap __log:=/home/albert/.ros/log/64af9fe4-75d7-11eb-b129-061ec70036de/rtabmap-rtabmap-2.log].
log file: /home/albert/.ros/log/64af9fe4-75d7-11eb-b129-061ec70036de/rtabmap-rtabmap-2*.log
```

matlabbe, 2021-02-25:
To debug `Process has died` errors, follow step B on this issue: https://github.com/introlab/rtabmap_ros/issues/28
The gdb log can help to detect if it is a `-march=native` compilation issue (if error happens in g2o, gtsam or eigen).

peredwardsson, 2021-03-02:
Using the tips from issue #28, we found the error to be incorrect versions of opencv installed simultaneously. Cleaning up solved the issue.

closed 2021-03-02 by peredwardsson