introlab/rtabmap_ros #549: errors in stopping mapping

bill4u, 2021-02-26:
@matlabbe Thanks for your great work!
There are some problems as follows:
when I finish mapping and press the CTRL+C to stop the progress, the error comes: 
_**escalating to SIGTERM**_  
and doesn't have the commets like " Saving database/long-term memory... (located at /home/xxxx/.ros/rtabmap.db) "
but in the  .ros, it does have the rtabmap.db

Then I start the program to work in the localization mode, and the error prints like follows:
**_VMDictionary.cpp:714:addWordRef() Not found word_**

so how to finish the mapping progress? Does the CTRL+C is the right way ?

Althrough I try to fix the .db with the program rtabmap-recovery, but the localization result error is too large to use.

Would please give me some advices? Thanks very much!

matlabbe, 2021-02-28:
What are exactly the last lines in the terminal before the SIGTERM happens on rtabmap node? CTRL-C should indeed quit safely rtabmap, while finalizing the database.

For
> but the localization result error is too large to use.

you may have to move in area already mapped until global localization happen. It is also possible to send a 2D pose estimate from RVIZ (on `/rtabmap/initialpose` topic) to help rtabmap to localize faster.

bill4u, 2021-02-28:
@matlabbe Thanks for your replay.Here is the error code:
[INFO] [1614394941.859442592]: rtabmap(102): Rate=1.06s, Limit=0.000s, RTAB-MAP=0.0858S, Maps update=0.0043s pub=0.0005s ( local map=2, WM=2)
**^C**(here I press the CTRL+C)
[rtabmap/rtabmap-2] killing on exit
[rtabmap/rgbd_sync-1] killing on exit
[t265_to_laser-6] killing on exit
[t265_to_d400-5] killing on exit
[d400/realsense2_camera-4] killing on exit
[d400/realsense2_camera_manager-3] killing on exit
[t265/realsense2_camera-2] killing on exit
[t265/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
shutting down processing monitor...
... shutting down processing monitor complete
done
**[rtabmap/rtabmap-2] escalating to SIGTERM**(This is the error. And I also try to increase the DEFAULT_TIMEOUT_SIGINT  = 15.0 #seconds and DEFAULT_TIMEOUT_SIGTERM = 2.0 #seconds but it doesn't work. And the CPU load of the rtabmap thread is 1.5% after the CTRL+C)
shutting down processing monitor...
... shutting down processing monitor complete
done

bill4u, 2021-02-28:
And now, I use the command **rosservice call /rtabmap/backup** to save the map and use shell script to replace the rtabmap.db with rtabmap.db.back and it can work correctly. But the error is still happen after mapping and CTRL+C. In the **CoreWrapper::~CoreWrapper()** I have found the code to saving the map but I have no idea why it doesn't goto the related code...As the exit code int the terminal doesn't have the words like  "**Saving database/long-term memory... (located at %s)**"

bill4u, 2021-02-28:
And I have found that if I install the rtabmap with the ros-melodic-rtabmap-ros/ros-melodic-rtabmap, The problem won't happen and it will save the map correctly!!! 
If I build the source code myself and install it, Sometimes in some platform like I have two NVIDIA XAVIER NX. In XAVIER NX #1, it won't produce the error. But error comes in XAVIER NX #2. While I exchange two TF cards, The error produces in #2 always. Totaly comfused.....May rebuilding the rtabmap project and reinstall is the right way?

matlabbe, 2021-02-28:
You could check with `ldd` if rtabmap node is correctly loading the right rtabmap libraries (if there are more than one version on the computer):
```
ldd ~/caktin_ws/devel/lib/rtabmap_ros/rtabmap | grep rtabmap
```

This is what is done just after ros::spin() exits after a Ctrl-C:
https://github.com/introlab/rtabmap_ros/blob/fe17d86870eabcfeb1e4b88d455bbb60c0abec91/src/CoreWrapper.cpp#L786-L822

If printf of line 804 is not shown, it means it may be blocked in other lines above (if the destructor is actually called). You could add some printf to see where it blocks. You may add a printf just after the spin here too to debug if it can exit the spin:
https://github.com/introlab/rtabmap_ros/blob/fe17d86870eabcfeb1e4b88d455bbb60c0abec91/src/CoreNode.cpp#L96

bill4u, 2021-03-01:
@matlabbe Thanks for your advice. I forget that I have added an TCP link in the src code. But it doesn't deal with the CTRL+C action. So it blocks in the end.

closed 2021-03-01 by bill4u