introlab/rtabmap_ros #550: catkin_make -j1 error

Mikor-mkr, 2021-03-01:
Hi everyone, I am trying to install rtabmap_ros from source with libpointmatcher dependency. I've completed all the steps and when I reach the last step and I run `catkin_make -j VERBOSE = 1`  I get the following error.

Base path: /home/aven/catkin_ws
Source space: /home/aven/catkin_ws/src
Build space: /home/aven/catkin_ws/build
Devel space: /home/aven/catkin_ws/devel
Install space: /home/aven/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/aven/catkin_ws/build"
####
####
#### Running command: "make -j1 VERBOSE=1" in "/home/aven/catkin_ws/build"
####
/usr/bin/cmake -H/home/aven/catkin_ws/src -B/home/aven/catkin_ws/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/aven/catkin_ws/build/CMakeFiles /home/aven/catkin_ws/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/aven/catkin_ws/build'
make -f rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins.dir/build.make rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins.dir/depend
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/aven/catkin_ws/src /home/aven/catkin_ws/src/rtabmap_ros /home/aven/catkin_ws/build /home/aven/catkin_ws/build/rtabmap_ros /home/aven/catkin_ws/build/rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/aven/catkin_ws/build'
make -f rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins.dir/build.make rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins.dir/build
make[2]: Entering directory '/home/aven/catkin_ws/build'
make[2]: Nothing to be done for 'rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins.dir/build'.
make[2]: Leaving directory '/home/aven/catkin_ws/build'
[  1%] Built target rtabmap_costmap_plugins
make -f rtabmap_ros/CMakeFiles/rtabmap_point_cloud_assembler.dir/build.make rtabmap_ros/CMakeFiles/rtabmap_point_cloud_assembler.dir/depend
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/aven/catkin_ws/src /home/aven/catkin_ws/src/rtabmap_ros /home/aven/catkin_ws/build /home/aven/catkin_ws/build/rtabmap_ros /home/aven/catkin_ws/build/rtabmap_ros/CMakeFiles/rtabmap_point_cloud_assembler.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/aven/catkin_ws/build'
make -f rtabmap_ros/CMakeFiles/rtabmap_point_cloud_assembler.dir/build.make rtabmap_ros/CMakeFiles/rtabmap_point_cloud_assembler.dir/build
make[2]: Entering directory '/home/aven/catkin_ws/build'
make[2]: Nothing to be done for 'rtabmap_ros/CMakeFiles/rtabmap_point_cloud_assembler.dir/build'.
make[2]: Leaving directory '/home/aven/catkin_ws/build'
[  1%] Built target rtabmap_point_cloud_assembler
make -f rtabmap_ros/CMakeFiles/rtabmap_pointcloud_to_depthimage.dir/build.make rtabmap_ros/CMakeFiles/rtabmap_pointcloud_to_depthimage.dir/depend
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/aven/catkin_ws/src /home/aven/catkin_ws/src/rtabmap_ros /home/aven/catkin_ws/build /home/aven/catkin_ws/build/rtabmap_ros /home/aven/catkin_ws/build/rtabmap_ros/CMakeFiles/rtabmap_pointcloud_to_depthimage.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/aven/catkin_ws/build'
make -f rtabmap_ros/CMakeFiles/rtabmap_pointcloud_to_depthimage.dir/build.make rtabmap_ros/CMakeFiles/rtabmap_pointcloud_to_depthimage.dir/build
make[2]: Entering directory '/home/aven/catkin_ws/build'
make[2]: Nothing to be done for 'rtabmap_ros/CMakeFiles/rtabmap_pointcloud_to_depthimage.dir/build'.
make[2]: Leaving directory '/home/aven/catkin_ws/build'
[  2%] Built target rtabmap_pointcloud_to_depthimage
make -f rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_UserData.dir/build.make rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_UserData.dir/depend
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/aven/catkin_ws/src /home/aven/catkin_ws/src/rtabmap_ros /home/aven/catkin_ws/build /home/aven/catkin_ws/build/rtabmap_ros /home/aven/catkin_ws/build/rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_UserData.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/aven/catkin_ws/build'
make -f rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_UserData.dir/build.make rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_UserData.dir/build
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build/rtabmap_ros && ../catkin_generated/env_cached.sh /usr/bin/python2 /opt/ros/melodic/share/genmsg/cmake/../../../lib/genmsg/genmsg_check_deps.py rtabmap_ros /home/aven/catkin_ws/src/rtabmap_ros/msg/UserData.msg std_msgs/Header
make[2]: Leaving directory '/home/aven/catkin_ws/build'
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_UserData
make -f rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_NodeData.dir/build.make rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_NodeData.dir/depend
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/aven/catkin_ws/src /home/aven/catkin_ws/src/rtabmap_ros /home/aven/catkin_ws/build /home/aven/catkin_ws/build/rtabmap_ros /home/aven/catkin_ws/build/rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_NodeData.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/aven/catkin_ws/build'
make -f rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_NodeData.dir/build.make rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_NodeData.dir/build
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build/rtabmap_ros && ../catkin_generated/env_cached.sh /usr/bin/python2 /opt/ros/melodic/share/genmsg/cmake/../../../lib/genmsg/genmsg_check_deps.py rtabmap_ros /home/aven/catkin_ws/src/rtabmap_ros/msg/NodeData.msg rtabmap_ros/Point3f:rtabmap_ros/KeyPoint:geometry_msgs/Vector3:geometry_msgs/Pose:rtabmap_ros/EnvSensor:geometry_msgs/Transform:std_msgs/Header:rtabmap_ros/Point2f:geometry_msgs/Point:rtabmap_ros/GPS:rtabmap_ros/GlobalDescriptor:geometry_msgs/Quaternion
make[2]: Leaving directory '/home/aven/catkin_ws/build'
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
make -f rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_GetMap.dir/build.make rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_GetMap.dir/depend
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/aven/catkin_ws/src /home/aven/catkin_ws/src/rtabmap_ros /home/aven/catkin_ws/build /home/aven/catkin_ws/build/rtabmap_ros /home/aven/catkin_ws/build/rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_GetMap.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/aven/catkin_ws/build'
make -f rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_GetMap.dir/build.make rtabmap_ros/CMakeFiles/_rtabmap_ros_generate_messages_check_deps_GetMap.dir/build
make[2]: Entering directory '/home/aven/catkin_ws/build'
cd /home/aven/catkin_ws/build/rtabmap_ros && ../catkin_generated/env_cached.sh /usr/bin/python2 /opt/ros/melodic/share/genmsg/cmake/../../../lib/genmsg/genmsg_check_deps.py rtabmap_ros /home/aven/catkin_ws/src/rtabmap_ros/srv/GetMap.srv rtabmap_ros/Point3f:rtabmap_ros/Link:rtabmap_ros/KeyPoint:rtabmap_ros/MapGraph:geometry_msgs/Vector3:geometry_msgs/Pose:rtabmap_ros/EnvSensor:geometry_msgs/Transform:rtabmap_ros/MapData:rtabmap_ros/NodeData:std_msgs/Header:rtabmap_ros/Point2f:geometry_msgs/Point:rtabmap_ros/GPS:rtabmap_ros/GlobalDescriptor:geometry_msgs/Quaternion
make[2]: Leaving directory '/home/aven/catkin_ws/build'
[... 445 lines cut ...]
/home/aven/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp: In member function ‘virtual void rtabmap_ros::CoreWrapper::onInit()’:
/home/aven/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:582:34: error: ‘kGridGlobalAltitudeDelta’ is not a member of ‘rtabmap::Parameters’
  if(parameters_.find(Parameters::kGridGlobalAltitudeDelta()) != parameters_.end())
[... cut ...]
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o] Error 1
make[2]: Leaving directory '/home/aven/catkin_ws/build'
CMakeFiles/Makefile2:2162: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all] Error 2
make[1]: Leaving directory '/home/aven/catkin_ws/build'
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 VERBOSE=1" failed

Does anybody have any suggestion? I am thinking to make a new vm and start from scratch at this point.

matlabbe, 2021-03-01:
To have access to libpointmatcher, you should build [rtabmap](https://github.com/introlab/rtabmap) library with libpointmatcher. The error above is because you are building latest rtabmap_ros version against older rtabmap version. Uninstall rtabmap binaries:
```
sudo apt remove ros-melodic-rtabmap
```
and rebuild it from source.

Mikor-mkr, 2021-03-01:
Yeah , I figured it out that I got it all messed up and I started from scratch, thank you, I am closing the issue.

closed 2021-03-01 by Mikor-mkr