introlab/rtabmap_ros #551: How to deal with pose jump in localization mode?

bill4u, 2021-03-02:
@matlabbe Hi, Thanks for your time. My robot pose  jumpes in my house and I have no ideas how to deal with it. It's kind of you to offer some advices.

**My sensors**: D435i + T265 + 2d Lidar(pepperl funch R2000)
**Mapping command :** 
roslaunch rtabmap_ros rtabmap.launch args:="-d --Mem/UseOdomGravity true --Optimizer/GravitySigma 0.3"  odom_topic:=/t265/odom/sample frame_id:=t265_link rgbd_sync:=true depth_topic:=/d400/aligned_depth_to_color/image_raw rgb_topic:=/d400/color/image_raw camera_info_topic:=/d400/color/camera_info approx_rgbd_sync:=false visual_odometry:=false rviz:=true rtabmapviz:=false subscribe_scan:=true

There are some parameters **added to** the default **rtabmap.launch** file: 
_<param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
<param name="RGBD/ProximityBySpace"     type="string" value="true"/>
<param name="RGBD/ProximityByTime"      type="string" value="false"/>
<param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/>
<param name="Reg/Strategy"              type="string" value="1"/>
<param name="Vis/MinInliers"        type="string" value="12"/>
<param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
<param name="RGBD/OptimizeMaxError"     type="string" value="3"/>
<param name="Reg/Force3DoF"             type="string" value="true"/>
<param name="Grid/FromDepth"            type="string" value="false"/>
<param name="Mem/STMSize"               type="string" value="30"/>
<param name="RGBD/LocalRadius"          type="string" value="5"/>
<param name="Icp/CorrespondenceRatio"   type="string" value="0.4"/>
<param name="RGBD/AngularUpdate"           type="string" value="0.1"/>
<param name="RGBD/LinearUpdate"            type="string" value="0.1"/>_

The poses are generate from the command : _rosrun tf tf_echo /map /t265_link_

And the poses sometimes jump quite a lot, like 50cm. And in the RVIZ, the TOPIC: MapCloud appears in the RIVZ if the loop closure happens, just like in the RtabMapVIZ, the left middle color is yellow or green. While the poses jump when the color is red or the MapCloud disappear in the RVIZ.

My question is : 
      _How can I do to minimize the pose jump?_ 
      Do I need to add the IMU data from D435i?
      _And how can I take the advantage of the  precise result of Lidar?_ 
      In my setting, Lidar data just feed into the icp_odometry? 
Please, I need your help. How to smooth the result. Thanks a lot.

bill4u, 2021-03-02:
https://user-images.githubusercontent.com/12217369/109643584-62058200-7b8f-11eb-8964-830a17440970.mp4

bill4u, 2021-03-02:
And this is my record data:
I don't know if Baidu cloud disc is accessable for you?

**Link:** https://pan.baidu.com/s/1lKM1dATmJd-M5oJDcHiokQ 
**extract code:** 9qgx

matlabbe, 2021-03-02:
The jumps in the video are expected, because of the lidar noise and not perfect map. When you say:

> There are some parameters added to the default rtabmap.launch file

Make sure in the terminal log on startup that rtabmap is correctly loading those parameters (with Reg/Strategy=1, you should have quite accurate localization if the lidar can see walls). For 50 cm jumps, it is quite high though. 

> While the poses jump when the color is red or the MapCloud disappear in the RVIZ.

If MapCLoud disappears, it means that rtabmap detected an odometry reset, thus creating a new session. A red link means a global loop closure, which would correct odometry drift, thus the robot can jump even meters when this happens if the odometry has drifted a lot.

bill4u, 2021-03-03:
@matlabbe , You are right, After double checking, the poor map is the main problem. But I have no idea why the loop cloure not happen when the robot visit the same position. As in the map, the chair has triple RGBD cloud assembled.
[image]

The info of program is as follows:

[ INFO] [1614815299.601559806, 1613904692.843858298]: rztechmap (46): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.0865s, Maps update=0.0014s pub=0.0000s (local map=19, WM=19)
[ INFO] [1614815300.600044468, 1613904693.835937013]: rztechmap (47): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.0736s, Maps update=0.0009s pub=0.0000s (local map=20, WM=20)
[ INFO] [1614815301.693155633, 1613904694.930426443]: rztechmap (48): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.0778s, Maps update=0.0010s pub=0.0000s (local map=21, WM=21)
[ INFO] [1614815302.657438414, 1613904695.900737275]: rztechmap (49): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.0395s, Maps update=0.0006s pub=0.0000s (local map=22, WM=22)
[ INFO] [1614815303.899084406, 1613904697.140487045]: rztechmap (50): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.0623s, Maps update=0.0021s pub=0.0000s (local map=23, WM=23)
[ INFO] [1614815304.917955277, 1613904698.157717449]: rztechmap (51): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.0785s, Maps update=0.0016s pub=0.0000s (local map=24, WM=24)
**[ WARN] (2021-03-04 07:48:26.001) RegistrationVis.cpp:1099::computeTransformationImpl() All projected points are outside the camera. Guess (xyz=0.152261,0.381405,0.000000 rpy=0.000000,0.000000,0.762376) is wrong or images are not overlapping.**
[ INFO] [1614815306.013338719, 1613904699.257362901]: rztechmap (52): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1000s, Maps update=0.0017s pub=0.0000s (local map=25, WM=25)
[ INFO] [1614815307.161863192, 1613904700.399110117]: rztechmap (53): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1710s, Maps update=0.0016s pub=0.0000s (local map=26, WM=26)
[ INFO] [1614815308.155316631, 1613904701.392383295]: rztechmap (54): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1021s, Maps update=0.0015s pub=0.0000s (local map=27, WM=27)
[ INFO] [1614815309.146139933, 1613904702.389754975]: rztechmap (55): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1571s, Maps update=0.0017s pub=0.0000s (local map=28, WM=28)
[ INFO] [1614815310.223167089, 1613904703.462785382]: rztechmap (56): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1238s, Maps update=0.0019s pub=0.0000s (local map=29, WM=29)
[ INFO] [1614815311.306477020, 1613904704.548200015]: rztechmap (57): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1081s, Maps update=0.0021s pub=0.0000s (local map=30, WM=30)
[ INFO] [1614815312.384879056, 1613904705.623326890]: rztechmap (58): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1252s, Maps update=0.0020s pub=0.0000s (local map=31, WM=31)
[ INFO] [1614815313.472922612, 1613904706.714967048]: rztechmap (59): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1536s, Maps update=0.0016s pub=0.0000s (local map=32, WM=32)
[ INFO] [1614815314.509780678, 1613904707.747440232]: rztechmap (60): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1327s, Maps update=0.0040s pub=0.0000s (local map=33, WM=33)
[ INFO] [1614815315.635953105, 1613904708.879271544]: rztechmap (61): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1208s, Maps update=0.0022s pub=0.0000s (local map=34, WM=34)
[ INFO] [1614815316.697366126, 1613904709.941356518]: rztechmap (62): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1140s, Maps update=0.0027s pub=0.0000s (local map=35, WM=35)
[ INFO] [1614815317.792454548, 1613904711.034824977]: rztechmap (63): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.1294s, Maps update=0.0018s pub=0.0000s (local map=36, WM=36)
[ **WARN] (2021-03-04 07:48:38.912) Rztechmap.cpp:2997::process() Rejecting all added loop closures (1, first is 64 <-> 28) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 5.502707 (edge 28->64, type=2, abs error=0.026198 m, stddev=0.004761). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.**
**[ WARN] (2021-03-04 07:48:38.912) Rztechmap.cpp:3021::process() Rejecting all added loop closures (1, first is 64 <-> 28) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 4.800178 (edge 28->64, type=2, abs error=1.309407 deg, stddev=0.004761). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2021-03-04 07:48:38.912) Rztechmap.cpp:3046::process() Loop closure 64->28 rejected!**
[ INFO] [1614815318.914683712, 1613904712.156242523]: rztechmap (64): Rate=1.00s, Limit=0.000s, RZTECH-Map=0.2080s, Maps update=0.0016s pub=0.0000s (local map=37, WM=37)
[ WARN] (2021-03-04 07:48:40.075) Rztechmap.cpp:2997::process() Rejecting all added loop closures (1, first is 65 <-> 29) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 12.194030 (edge 29->65, type=2, abs error=0.040287 m, stddev=0.003304). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2021-03-04 07:48:40.075) Rztechmap.cpp:3021::process() Rejecting all added loop closures (1, first is 65 <-> 29) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 6.948079 (edge 29->65, type=2, abs error=1.315243 deg, stddev=0.003304). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3. [... cut ...]
[ WARN] (2021-03-04 07:49:02.133) Rztechmap.cpp:3021::process() Rejecting all added loop closures (1, first is 86 <-> 28) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 6.933837 (edge 28->86, type=1, abs error=2.009127 deg, stddev=0.005057). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2021-03-04 07:49:02.133) Rztechmap.cpp:3046::process() Loop closure 86->28 rejected!

closed 2021-03-04 by bill4u