introlab/rtabmap_ros #552: Ouster lidar no data received

Jaetriel, 2021-03-02:
Hello, I am quite new to rtabmap and ROS in general, I am trying to connect an OS1-128 lidar sensor with an external Tinkerforge Brick v2 IMU and visualize it using rtabmap and ROS. I started with just installing rtabmap and trying to connect the lidar first without the external IMU, just using the lidar's built-in one, and run the test_ouster_gen2.launch file with the basic configuration but I keep getting warnings that rtabmap did not receive any data.Here is the launch file I was using: 

<launch>
  
    

    
    <arg name="sensor_hostname"/>
    <arg name="udp_dest"/>

    <arg name="frame_id" default="os_sensor"/>
    <arg name="rtabmapviz"    default="true"/>
    <arg name="scan_20_hz"    default="false"/>
    <arg name="voxel_size"    default="0.15"/>  
    <arg name="use_sim_time"  default="false"/>
    <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>
    
    
    <include unless="$(arg use_sim_time)" file="$(find ouster_ros)/ouster.launch">
      <arg name="sensor_hostname" value="$(arg sensor_hostname)"/>
      <arg name="udp_dest" value="$(arg udp_dest)"/>
      <arg name="image" value="true"/>
      <arg     if="$(arg scan_20_hz)" name="lidar_mode" value="1024x20"/>
      <arg unless="$(arg scan_20_hz)" name="lidar_mode" value="2048x10"/>
    </include>

    
    <node pkg="nodelet" type="nodelet" name="imu_nodelet_manager" args="manager">
      <remap from="imu/data_raw" to="/os_cloud_node/imu"/>
      <remap from="imu/data" to="/os_cloud_node/imu/data"/>
    </node>
    <node pkg="nodelet" type="nodelet" name="imu_filter" args="load imu_filter_madgwick/ImuFilterNodelet imu_nodelet_manager">
      <param name="use_mag" value="false"/>
      <param name="world_frame" value="enu"/>
      <param name="publish_tf" value="false"/>
    </node> 
    <node pkg="nodelet" type="nodelet" name="imu_to_tf" args="load rtabmap_ros/imu_to_tf imu_nodelet_manager">
      <remap from="imu/data" to="/os_cloud_node/imu/data"/>
      <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
      <param name="base_frame_id" value="$(arg frame_id)"/>
    </node> 

    <group ns="rtabmap">
      <node pkg="rtabmap_ros" type="icp_odometry" name="icp_odometry" output="screen">
        <remap from="scan_cloud" to="/os_cloud_node/points"/>
        <param name="frame_id"        type="string" value="$(arg frame_id)"/>  
        <param name="odom_frame_id"   type="string" value="odom"/>
        <param     if="$(arg scan_20_hz)" name="expected_update_rate" type="double" value="25"/>
        <param unless="$(arg scan_20_hz)" name="expected_update_rate" type="double" value="15"/>

        <remap from="imu" to="/os_cloud_node/imu/data"/>
        <param name="guess_frame_id"   type="string" value="$(arg frame_id)_stabilized"/>
        <param name="wait_imu_to_init" type="bool" value="true"/>
     
        
        <param name="Icp/PointToPlane"        type="string" value="true"/>
        <param name="Icp/Iterations"          type="string" value="10"/>
        <param name="Icp/VoxelSize"           type="string" value="$(arg voxel_size)"/>
        <param name="Icp/DownsamplingStep"    type="string" value="1"/> 
        <param name="Icp/Epsilon"             type="string" value="0.001"/>
        <param name="Icp/PointToPlaneK"       type="string" value="20"/>
        <param name="Icp/PointToPlaneRadius"  type="string" value="0"/>
        <param name="Icp/MaxTranslation"      type="string" value="2"/>
        <param name="Icp/MaxCorrespondenceDistance" type="string" value="1"/>
        <param name="Icp/PM"                  type="string" value="true"/> 
        <param name="Icp/PMOutlierRatio"      type="string" value="0.1"/>
        <param name="Icp/CorrespondenceRatio" type="string" value="0.01"/>  

               
        <param name="Odom/ScanKeyFrameThr"       type="string" value="0.95"/>
[... 56 lines cut ...]
        <param name="frame_id" type="string" value="$(arg frame_id)"/>
        <param name="odom_frame_id" type="string" value="odom"/>
        <param name="subscribe_odom_info" type="bool" value="true"/>
        <param name="subscribe_scan_cloud" type="bool" value="true"/>
        <param name="approx_sync" type="bool" value="false"/>
        <remap from="scan_cloud" to="/os_cloud_node/points"/>
      </node>
  </group>

</launch>

And the warnings i keep getting after starting rtabmap with roslaunch rtabmap_ros test_ouster_gen2.launch sensor_hostname:=192.xxxxxx udp_dest:=192.xxxxxx : 

> [ WARN] [1614687309.213491141]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
> /rtabmap/rtabmapviz subscribed to (exact sync):
>    /os_cloud_node/points \
>    /rtabmap/odom_info
> [ WARN] [1614687311.238510555]: Still waiting for data on topic /imu/data_raw...
> [ WARN] [1614687312.882493616]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
> /rtabmap/rtabmap subscribed to (exact sync):
>    /rtabmap/odom \
>    /os_cloud_node/points
> 

I am not sure what exactly I am missing and what's going wrong, any help would be appreciated.

matlabbe, 2021-03-02:
check if `/os_cloud_node/points` is published by ouster node:
```
rostopic hz /os_cloud_node/points 
```

Jaetriel, 2021-03-02:
> check if `/os_cloud_node/points` is published by ouster node and that imu topic is correctly filtered:
> 
> ```
> rostopic hz /os_cloud_node/points /os_cloud_node/imu/data /os_cloud_node/imu
> ```

It seems that if i run the ros node for the ouster lidar by itself it publishes and i get the messages fine but when i run this launch file for ouster with rtabmap i get 
`subscribed to [/os_cloud_node/points]
subscribed to [/os_cloud_node/imu/data]
subscribed to [/os_cloud_node/imu]
no new messages`

matlabbe, 2021-03-02:
The code starting the ouster is here:
https://github.com/introlab/rtabmap_ros/blob/fbd4a144ae9d559ebf2c2086e7e2b3e46096dc47/launch/tests/test_ouster_gen2.launch#L26-L33

Maybe the lidar_mode is not compatible with your ouster, and verify the upd and sensor hostname parameters.

Jaetriel, 2021-03-03:
I checked the hostname and udp dest, and indeed there was a typo there, so now when I run it with the correct parameters I can see the topics are being published 
[image]

but I now get an error that icp_odometry has died and then still the same warnings of no data received

[image]

matlabbe, 2021-03-04:
Can you launch [icp_odometry with gdb](https://github.com/introlab/rtabmap_ros/issues/28) and show the backtrace when it crashes?
https://github.com/introlab/rtabmap_ros/blob/fbd4a144ae9d559ebf2c2086e7e2b3e46096dc47/launch/tests/test_ouster_gen2.launch#L52
```xml
      <node pkg="rtabmap_ros" type="icp_odometry" name="icp_odometry" output="screen" launch-prefix="xterm -e gdb -ex run --args">
```
Type `run` in gdb terminal to start the node. Type `bt` after it crashes.

If you can record a rosbag it could be nice to debug it here:
```
rosbag record -a
```

cheers,
Mathieu

matlabbe, 2021-05-02:
It depends which topic you are looking in RVIZ. For example cloud_map would be published at 1 Hz by default.

Your warning is because TF of odometry was more than 200 ms old. As your odometry is running at 6 Hz on average, when it drops under 5 Hz, then you are likely to have the warning.

Samehibrahim51, 2023-03-10:
Hi, I also have the same error using ouster os1- 16 beam, and can you tell me how did you solve it?