introlab/rtabmap_ros #553: Double free or corruption (out)

askokostic, 2021-03-02:
Hello,

At the beginning of February, rtabmap was working without any issues, but around two weeks ago I started having problems. When trying to launch anything I started getting `double free or corruption (out)`. 
If I only launch rtabmap without the simulation it does not crash, but as soon as it tries to subscribe to an existing topic it crashes.

Ubuntu 20.04
ROS2 Foxy
Everything is running inside a Docker container. 
The same error occurred on another pc when doing the same steps (same Dockerfile).

I have tried the suggestion from #505, but it crashes even faster. Changing `-DBUILD_WITH_MARCH_NATIVE` to `OFF` has the same effect. Installing Eigen and GTSAM from the source also didn't help. 
When launching the node with gdb I get a lot of Eigen callback errors.

**When executing:**

```
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/ros2_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models
export TURTLEBOT3_MODEL=waffle
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
ros2 launch rtabmap_ros turtlebot3_scan.launch.py
```
**I get this output:**

```
[INFO] [launch]: All log files can be found below /home/vscode/.ros/log/2021-02-25-16-36-15-064915-aleksa-Vostro-7590-5100
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rtabmap-1]: process started with pid [5105]
[INFO] [rtabmapviz-2]: process started with pid [5107]
[rtabmapviz-2] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[rtabmap-1] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[rtabmapviz-2] QStandardPaths: XDG_RUNTIME_DIR points to non-existing path '/run/user/1000', please create it with 0700 permissions.
[rtabmap-1] [INFO] [1614270976.170607016] [rtabmap]: rtabmap: subscribe_depth = false
[rtabmap-1] [INFO] [1614270976.170887317] [rtabmap]: rtabmap: subscribe_rgb = false
[rtabmap-1] [INFO] [1614270976.170908860] [rtabmap]: rtabmap: subscribe_stereo = false
[rtabmap-1] [INFO] [1614270976.170920231] [rtabmap]: rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmap-1] [INFO] [1614270976.170932497] [rtabmap]: rtabmap: subscribe_odom_info = false
[rtabmap-1] [INFO] [1614270976.170944118] [rtabmap]: rtabmap: subscribe_user_data = false
[rtabmap-1] [INFO] [1614270976.170960069] [rtabmap]: rtabmap: subscribe_scan = true
[rtabmap-1] [INFO] [1614270976.170975604] [rtabmap]: rtabmap: subscribe_scan_cloud = false
[rtabmap-1] [INFO] [1614270976.170991850] [rtabmap]: rtabmap: queue_size    = 10
[rtabmap-1] [INFO] [1614270976.171007709] [rtabmap]: rtabmap: approx_sync   = true
[rtabmap-1] [INFO] [1614270976.177933625] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-1] [INFO] [1614270976.178015719] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-1] [INFO] [1614270976.178039051] [rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-1] [INFO] [1614270976.178060963] [rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-1] [INFO] [1614270976.178076719] [rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-1] [INFO] [1614270976.178091851] [rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-1] [INFO] [1614270976.178106337] [rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-1] [INFO] [1614270976.178120815] [rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-1] [INFO] [1614270976.211075701] [rtabmap]: rtabmap: frame_id      = base_footprint
[rtabmap-1] [INFO] [1614270976.211142195] [rtabmap]: rtabmap: map_frame_id  = map
[rtabmap-1] [INFO] [1614270976.211160963] [rtabmap]: rtabmap: use_action_for_goal  = false
[rtabmap-1] [INFO] [1614270976.211180060] [rtabmap]: rtabmap: tf_delay      = 0.050000
[rtabmap-1] [INFO] [1614270976.211264954] [rtabmap]: rtabmap: tf_tolerance  = 0.100000
[rtabmap-1] [INFO] [1614270976.211285119] [rtabmap]: rtabmap: odom_sensor_sync   = false
[rtabmapviz-2] [INFO] [1614270976.216233277] [rtabmapviz]: rtabmapviz: subscribe_depth = false
[rtabmapviz-2] [INFO] [1614270976.216577444] [rtabmapviz]: rtabmapviz: subscribe_rgb = false
[rtabmapviz-2] [INFO] [1614270976.216611710] [rtabmapviz]: rtabmapviz: subscribe_stereo = false
[rtabmapviz-2] [INFO] [1614270976.216630162] [rtabmapviz]: rtabmapviz: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmapviz-2] [INFO] [1614270976.216659224] [rtabmapviz]: rtabmapviz: subscribe_odom_info = false
[rtabmapviz-2] [INFO] [1614270976.216678802] [rtabmapviz]: rtabmapviz: subscribe_user_data = false
[rtabmapviz-2] [INFO] [1614270976.216698061] [rtabmapviz]: rtabmapviz: subscribe_scan = true
[rtabmapviz-2] [INFO] [1614270976.216725072] [rtabmapviz]: rtabmapviz: subscribe_scan_cloud = false
[rtabmapviz-2] [INFO] [1614270976.216744381] [rtabmapviz]: rtabmapviz: queue_size    = 10
[rtabmapviz-2] [INFO] [1614270976.216764440] [rtabmapviz]: rtabmapviz: approx_sync   = true
[rtabmapviz-2] [INFO] [1614270976.221940982] [rtabmapviz]: rtabmapviz: Using configuration from "/home/vscode/.ros/rtabmapGUI.ini"
[rtabmap-1] [INFO] [1614270976.266343074] [rtabmap]: Setting RTAB-Map parameter "RGBD/NeighborLinkRefining"="True"
[rtabmap-1] [INFO] [1614270976.266749074] [rtabmap]: Setting RTAB-Map parameter "Reg/Strategy"="1"
[rtabmap-1] [WARN] [1614270976.282232905] [rtabmap]: Setting "Grid/FromDepth" parameter to false (default true) as "subscribe_scan" or "subscribe_scan_cloud" is true. The occupancy grid map will be constructed from laser scans. To get occupancy grid map from cloud projection, set "Grid/FromDepth" to true. To suppress this warning, add <param name="Grid/FromDepth" type="string" value="false"/>
[rtabmap-1] [INFO] [1614270976.282349111] [rtabmap]: Setting "Grid/RangeMax" parameter to 0 (default 5.000000) as "subscribe_scan" or "subscribe_scan_cloud" is true.
[... 781 lines cut ...]
WORKDIR /tmp
RUN set -x && \
  git clone https://github.com/introlab/rtabmap.git && \
  cd rtabmap && \
  git checkout ${RTABMAP_COMMIT} && \
  mkdir -p build && \
  cd build && \
  cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=${CMAKE_INSTALL_PREFIX} \
    .. && \
  make -j${NUM_THREADS} && \
  make install && \
  cd /tmp && \
  rm -rf * && \
  ldconfig
# ENV DBoW2_DIR=${CMAKE_INSTALL_PREFIX}/lib/cmake/DBoW2

ENV FASTRTPS_DEFAULT_PROFILES_FILE=/workspaces/ws_robogreen/fast_dds_config.xml

RUN echo "source /workspaces/ws_robogreen/install/setup.bash" >> /home/${USERNAME}/.bashrc \
    && usermod -aG 104 vscode \
    && usermod -aG video vscode \
    && usermod -aG dialout vscode 
```

matlabbe, 2021-03-02:
I tried with a minimal Dockerfile (from ros:foxy) with ros-foxy-desktop and ros-foxy-libg2o (had to[ update rtabmap library](https://github.com/introlab/rtabmap/commit/4e4207a6dd5ea42520a5db50990b1f4afa65ae14) to find that g2o on foxy), gtsam binaries (like in your dockerfile), libpointmatcher from source. 

```bash
~/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros# gdb ex run --args rtabmap

GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rtabmap...
(No debugging symbols found in rtabmap)
(gdb) run
Starting program: /root/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap 
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
double free or corruption (out)

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007fa561e23859 in __GI_abort () at abort.c:79
#2  0x00007fa561e8e3ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fa561fb8285 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007fa561e9647c in malloc_printerr (str=str@entry=0x7fa561fba670 "double free or corruption (out)") at malloc.c:5347
#4  0x00007fa561e98120 in _int_free (av=0x7fa561fe9b80 <main_arena>, p=0x5587f111efb0, have_lock=<optimized out>) at malloc.c:4314
#5  0x00007fa55ee968e1 in g2o::EdgeProjectP2MC_Intrinsics::~EdgeProjectP2MC_Intrinsics() () at /opt/ros/foxy/lib/libg2o_types_sba.so
#6  0x00007fa55965ec84 in g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::AbstractHyperGraphElementCreator*) ()
    at /opt/ros/foxy/lib/libg2o_core.so
#7  0x00007fa55ee8fb47 in  () at /opt/ros/foxy/lib/libg2o_types_sba.so
#8  0x00007fa56ae23b8a in  () at /lib64/ld-linux-x86-64.so.2
#9  0x00007fa56ae23c91 in  () at /lib64/ld-linux-x86-64.so.2
#10 0x00007fa56ae1313a in  () at /lib64/ld-linux-x86-64.so.2
#11 0x0000000000000001 in  ()
#12 0x00007fffeca30665 in  ()
#13 0x0000000000000000 in  ()
(gdb) 
```
Looks like it is crashing in `/opt/ros/foxy/lib/libg2o_types_sba.so`. I'll try to uninstall those g2o binaries and build an older g2o version.

matlabbe, 2021-03-02:
Installing older g2o version (I did go down to 20170730_git version) does the same error. With GTSAM dependency, it was working. I reinstalled latest g2o binaries, then installed GTSAM stable release instead of develop, then it worked!

Instead of:
```dockerfile
RUN add-apt-repository ppa:joseluisblancoc/gtsam-develop -y
RUN apt-get update \
    && apt-get -y install \
      libqt5svg5-dev \
      libgtsam-dev \
    #
    # Clean ups
    && apt-get autoremove -y \
    && apt-get clean -y \
    && rm -rf /var/lib/apt/lists/*
```

do:
```dockerfile
RUN add-apt-repository ppa:borglab/gtsam-release-4.0 -y
RUN apt-get update \
    && apt-get -y install \
      libqt5svg5-dev \
      libgtsam-dev \
      libgtsam-unstable-dev \
    #
    # Clean ups
    && apt-get autoremove -y \
    && apt-get clean -y \
    && rm -rf /var/lib/apt/lists/*
```

askokostic, 2021-03-02:
I'm getting these error messages when building the package

```
[Processing: rtabmap_ros]                               
--- stderr: rtabmap_ros                                 
CMake Warning:
  Manually-specified variables were not used by the project:

    DBoW2_DIR
    Pangolin_DIR
    g2o_DIR
    openvslam_DIR

Member name 'global' in the service request 'GetMap' is a reserved keyword in Python and is not supported at the moment. Please use a different name.
Member name 'global' in the service request 'PublishMap' is a reserved keyword in Python and is not supported at the moment. Please use a different name.
/workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘bool rtabmap_ros::convertScanMsg(const LaserScan&, const string&, const string&, const rclcpp::Time&, rtabmap::LaserScan&, tf2_ros::Buffer&, double, bool)’:
/workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp:1741:74: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘rtabmap::LaserScan’)
 1741 |   data = rtabmap::util3d::laserScan2dFromPointCloud(*pclScan, laserToOdom); // put back in laser frame
      |                                                                          ^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.inl.hpp:743:6: note: candidate: ‘cv::Mat& cv::Mat::operator=(const cv::Mat&)’
  743 | Mat& Mat::operator = (const Mat& m)
      |      ^~~
/usr/include/opencv4/opencv2/core/mat.inl.hpp:743:34: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const cv::Mat&’
  743 | Mat& Mat::operator = (const Mat& m)
      |                       ~~~~~~~~~~~^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.inl.hpp:3356:6: note: candidate: ‘cv::Mat& cv::Mat::operator=(const cv::MatExpr&)’
 3356 | Mat& Mat::operator = (const MatExpr& e)
      |      ^~~
/usr/include/opencv4/opencv2/core/mat.inl.hpp:3356:38: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const cv::MatExpr&’
 3356 | Mat& Mat::operator = (const MatExpr& e)
      |                       ~~~~~~~~~~~~~~~^
In file included from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.hpp:1241:10: note: candidate: ‘cv::Mat& cv::Mat::operator=(const Scalar&)’
 1241 |     Mat& operator = (const Scalar& s);
      |          ^~~~~~~~
/usr/include/opencv4/opencv2/core/mat.hpp:1241:36: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const Scalar&’ {aka ‘const cv::Scalar_<double>&’}
 1241 |     Mat& operator = (const Scalar& s);
      |                      ~~~~~~~~~~~~~~^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.inl.hpp:1405:6: note: candidate: ‘cv::Mat& cv::Mat::operator=(cv::Mat&&)’
 1405 | Mat& Mat::operator = (Mat&& m)
      |      ^~~
/usr/include/opencv4/opencv2/core/mat.inl.hpp:1405:29: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘cv::Mat&&’
 1405 | Mat& Mat::operator = (Mat&& m)
      |                       ~~~~~~^
/workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp:1749:74: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘rtabmap::LaserScan’)
 1749 |   data = rtabmap::util3d::laserScan2dFromPointCloud(*pclScan, laserToOdom); // put back in laser frame
      |                                                                          ^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /workspaces/ws_robogreen/src/rtabmap_ros/src/MsgConversion.cpp:28:
[... 113 lines cut ...]
  318 |  if(waitIMUToinit_ || odometry_->canProcessIMU())
      |                                  ^~~~~~~~~~~~~
      |                                  canProcessAsyncIMU
[... cut ...]
      |                 ^~~~~~~~~~~~~
      |                 canProcessAsyncIMU
make[2]: *** [CMakeFiles/rtabmap_ros.dir/build.make:63: CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/rtabmap_ros.dir/build.make:89: CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:893: CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rtabmap_ros [44.1s, exited with code 2]
```

askokostic, 2021-03-02:
I'm getting these error messages when building the package

```
[Processing: rtabmap_ros]                               
--- stderr: rtabmap_ros                                 
CMake Warning:
  Manually-specified variables were not used by the project:

    DBoW2_DIR
    Pangolin_DIR
    g2o_DIR
    openvslam_DIR

Member name 'global' in the service request 'GetMap' is a reserved keyword in Python and is not supported at the moment. Please use a different name.
Member name 'global' in the service request 'PublishMap' is a reserved keyword in Python and is not supported at the moment. Please use a different name.
/ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘bool rtabmap_ros::convertScanMsg(const LaserScan&, const string&, const string&, const rclcpp::Time&, rtabmap::LaserScan&, tf2_ros::Buffer&, double, bool)’:
/ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp:1741:74: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘rtabmap::LaserScan’)
 1741 |   data = rtabmap::util3d::laserScan2dFromPointCloud(*pclScan, laserToOdom); // put back in laser frame
      |                                                                          ^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /ros2_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.inl.hpp:743:6: note: candidate: ‘cv::Mat& cv::Mat::operator=(const cv::Mat&)’
  743 | Mat& Mat::operator = (const Mat& m)
      |      ^~~
/usr/include/opencv4/opencv2/core/mat.inl.hpp:743:34: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const cv::Mat&’
  743 | Mat& Mat::operator = (const Mat& m)
      |                       ~~~~~~~~~~~^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /ros2_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.inl.hpp:3356:6: note: candidate: ‘cv::Mat& cv::Mat::operator=(const cv::MatExpr&)’
 3356 | Mat& Mat::operator = (const MatExpr& e)
      |      ^~~
/usr/include/opencv4/opencv2/core/mat.inl.hpp:3356:38: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const cv::MatExpr&’
 3356 | Mat& Mat::operator = (const MatExpr& e)
      |                       ~~~~~~~~~~~~~~~^
In file included from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /ros2_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.hpp:1241:10: note: candidate: ‘cv::Mat& cv::Mat::operator=(const Scalar&)’
 1241 |     Mat& operator = (const Scalar& s);
      |          ^~~~~~~~
/usr/include/opencv4/opencv2/core/mat.hpp:1241:36: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const Scalar&’ {aka ‘const cv::Scalar_<double>&’}
 1241 |     Mat& operator = (const Scalar& s);
      |                      ~~~~~~~~~~~~~~^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /ros2_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv4/opencv2/core/mat.inl.hpp:1405:6: note: candidate: ‘cv::Mat& cv::Mat::operator=(cv::Mat&&)’
 1405 | Mat& Mat::operator = (Mat&& m)
      |      ^~~
/usr/include/opencv4/opencv2/core/mat.inl.hpp:1405:29: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘cv::Mat&&’
 1405 | Mat& Mat::operator = (Mat&& m)
      |                       ~~~~~~^
/ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp:1749:74: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘rtabmap::LaserScan’)
 1749 |   data = rtabmap::util3d::laserScan2dFromPointCloud(*pclScan, laserToOdom); // put back in laser frame
      |                                                                          ^
In file included from /usr/include/opencv4/opencv2/core/mat.hpp:3724,
                 from /usr/include/opencv4/opencv2/core.hpp:59,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /ros2_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:40,
                 from /ros2_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
[... 113 lines cut ...]
  318 |  if(waitIMUToinit_ || odometry_->canProcessIMU())
      |                                  ^~~~~~~~~~~~~
      |                                  canProcessAsyncIMU
/ros2_ws/src/rtabmap_ros/src/OdometryROS.cpp: In member function ‘void rtabmap_ros::OdometryROS::callbackIMU(sensor_msgs::msg::Imu_<std::allocator<void> >::SharedPtr)’:
/ros2_ws/src/rtabmap_ros/src/OdometryROS.cpp:357:18: error: ‘class rtabmap::Odometry’ has no member named ‘canProcessIMU’; did you mean ‘canProcessAsyncIMU’?
  357 |   if(!odometry_->canProcessIMU() &&
[... cut ...]
      |                 ^~~~~~~~~~~~~
      |                 canProcessAsyncIMU
make[2]: *** [CMakeFiles/rtabmap_ros.dir/build.make:63: CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/rtabmap_ros.dir/build.make:89: CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:893: CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rtabmap_ros [44.1s, exited with code 2]
```

matlabbe, 2021-03-02:
This is a change done yesterday, pull latest rtabmap_ros ros2 branch.

askokostic, 2021-03-02:
Console output:

```
[INFO] [launch]: All log files can be found below /home/vscode/.ros/log/2021-03-02-21-14-02-494762-aleksa-Vostro-7590-36189
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rtabmap-1]: process started with pid [36194]
[INFO] [rtabmapviz-2]: process started with pid [36196]
[ERROR] [rtabmapviz-2]: process has died [pid 36196, exit code -11, cmd '/workspaces/ws_robogreen/install/rtabmap_ros/lib/rtabmap_ros/rtabmapviz --ros-args --params-file /tmp/launch_params_yvkr1e5x -r scan:=/scan'].
[ERROR] [rtabmap-1]: process has died [pid 36194, exit code -11, cmd '/workspaces/ws_robogreen/install/rtabmap_ros/lib/rtabmap_ros/rtabmap -d --ros-args --params-file /tmp/launch_params_n045s7ys -r scan:=/scan'].
```

gdb call stack (of [rtabmap-1]):

```
libc.so.6!free (Unknown Source:0)
libg2o_types_slam2d.so!g2o::EdgeSE2PointXYCalib::~EdgeSE2PointXYCalib() (Unknown Source:0)
libg2o_core.so!g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::AbstractHyperGraphElementCreator*) (Unknown Source:0)
libg2o_types_slam2d.so!__static_initialization_and_destruction_0(int, int) [clone .constprop.0] (Unknown Source:0)
ld-linux-x86-64.so.2![Unknown/Just-In-Time compiled code] (Unknown Source:0)
```

matlabbe, 2021-03-02:
It seems crashing on another Factory::registerType. Let me rebuild your dockerfile with gtsam stable release and `-DBUILD_WITH_MARCH_NATIVE=OFF`.

askokostic, 2021-03-02:
[This](https://github.com/introlab/rtabmap_ros/issues/553#issuecomment-789162658) and `DBUILD_WITH_MARCH_NATIVE=OFF` seemed to have worked. It launches without any errors now. I will try out the other launch files to make sure everything works. Thank you very much!

matlabbe, 2021-03-02:
Yeah, just finished testing your dockerfile with those changes and I don't have errors. Well I had an exception from libpointmacher about not existing parameter Force4DoF, which has been added recently. This [commit](https://github.com/introlab/rtabmap/commit/98c69c4578905cf6ff089e1769f05463cc848756) fixed that problem.

closed 2021-03-04 by askokostic