introlab/rtabmap_ros #554: rtabmap launch file: Bayes Filter Error

s3668134, 2021-03-09:
Hello,

I've gotten this error every time I have launched rviz and rtabmap: using the basic rtabmap.launch file and I am unsure as to why. 
I am using rtabmap to view Intel RealSense and I pass it with these arguments in the roslaunch command.

- rtabmap_args:="—delete_db_on_start" 
- depth_topic:=/camera/aligned_depth_to_color/image_raw 
- rgb_topic:=/camera/color/image_raw 
- camera_info_topic:=/camera/color/camera_info 
- approx_sync:=false 

`[FATAL] (2021-03-09 21:04:52.313) BayesFilter.cpp:357::generatePrediction() No 0 margin neighbor for signature 19 !?!?`

If this information helps, here it is
Hardware: Jetson Nano
OS: Ubuntu 18.04 w/ ROS Melodic
Using QT4, VTK 6.30, PCL 1.8 

How could I go about solving this issue?

s3668134, 2021-03-12:
@matlabbe Hello, thank you for your help so far.

I was able to pull the latest changes and set the Mem/InitWMWithAllNodes to true, however, rtabmap/cloud_map will not show up even though running `rostopic hz /rtabmap/cloud_map` shows that we are getting messages from it. How could I get the cloud_map to display in rviz?

matlabbe, 2021-03-12:
What is your command when launching rtabmap? Do you start from a fresh database? Do you restart in localization mode or in SLAM mode? 

cloud_map may be empty until the robot is localized in the previous map.

s3668134, 2021-03-12:
My command to launch rtabmap is

`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="—delete_db_on_start" depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info approx_sync:=false`

We start from a fresh database, and we restart in SLAM mode.

matlabbe, 2021-03-12:
Make sure you do double-dash `"--delete_db_on_start"` or simply `"-d"` to start from fresh database. Then if the map is still empty, are there warnings on terminal?

s3668134, 2021-03-12:
There were no warnings on terminal when I was executing the command. I can run rtabmapviz and get the 3d points, it just will not display on rviz.