introlab/rtabmap_ros #555: RTABMAP stopping with L515 and T265

NorwegianPi, 2021-03-12:
I am trying to run create a cloud_map by using L515 as camera and T265 for odometry.

| System   | |
| ------------- | ------------- |
| Ubuntu 18.04  |   |
| RTABMAP | built from source  |
| Real Sense | built from source  |
| IntelNuc | built from source  |
| ROS Melodic | apt  |

When i launch the realsense cameras i do get a warning `frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically`, but cameras seems to run fine.
When i launch RTABMAP it seems to run og for 20-30s before stopping publishing. There is no error message realsense or rtabmap window.

If I try to restart rtabmap or try to start rtabmapp I get this warning:

```
[ WARN] [1615559798.353315754]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /t265/odom/sample \
   /rtabmap/rgbd_image
```
Nothing is published to `/rtabmap/rgbd_image` topic.

if i set `<arg name="subscribe_rgbd" default="false"/>`, i get another warning. 

``` xml
[ WARN] [1615561088.313176827]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /t265/odom/sample \
   /l515/color/image_raw \
   /l515/aligned_depth_to_color/image_raw \
   /l515/color/camera_info

```
All these topics have a rate of 30 (200 for odom).

Does anyone have and idea what might be causing this behaviour?

I have tried the launch file metioned by @pauldeee. [here](https://github.com/introlab/rtabmap/issues/574), but can not get it to work when T265 is enabled

**realsense.launch:**

``` xml

<launch>
  <arg name="device_type_t265"    		default="t265"/>
  <arg name="device_type_l515"    		default="l515"/>
  <arg name="device_type_d435"    		default="d4.5"/>
  
  <arg name="serial_no_t265"    			default=""/>
  <arg name="serial_no_l515"    			default=""/>
  <arg name="serial_no_d435"    			default=""/>
  
  <arg name="t265_ns"              			default="t265"/>
  <arg name="l515_ns"              			default="l515"/>
  <arg name="d435_ns"              			default="d435"/>

  <arg name="tf_prefix_t265"         default="$(arg t265_ns)"/>
  <arg name="tf_prefix_l515"         default="$(arg l515_ns)"/>
  <arg name="tf_prefix_d435"         default="$(arg d435_ns)"/>

  <arg name="initial_reset"             default="false"/>
  <arg name="enable_fisheye"            default="false"/>

  <arg name="color_width"               default="1280"/>
  <arg name="color_height"              default="720"/>
  <arg name="depth_width"               default="640"/>
  <arg name="depth_height"              default="480"/>
  <arg name="ordered_pc"                default="false"/>
  <arg name="clip_distance"             default="-2"/>
[... 853 lines cut ...]
[ INFO] [1615559437.574384325]: rtabmap (22): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1267s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559438.119128604]: rtabmap (23): Rate=0.50s, Limit=0.000s, Conversion=0.0017s, RTAB-Map=0.1293s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559438.639744817]: rtabmap (24): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1260s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559439.178648379]: rtabmap (25): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1313s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559439.710400016]: rtabmap (26): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1312s, Maps update=0.0002s pub=0.0001s (local map=1, WM=1)
[ INFO] [1615559440.240332056]: rtabmap (27): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1277s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559440.777700222]: rtabmap (28): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1307s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559441.308780334]: rtabmap (29): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1303s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559441.842133943]: rtabmap (30): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1313s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559442.382133610]: rtabmap (31): Rate=0.50s, Limit=0.000s, Conversion=0.0012s, RTAB-Map=0.1383s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559442.902529360]: rtabmap (32): Rate=0.50s, Limit=0.000s, Conversion=0.0018s, RTAB-Map=0.1254s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559443.435852451]: rtabmap (33): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1257s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559444.002007380]: rtabmap (34): Rate=0.50s, Limit=0.000s, Conversion=0.0025s, RTAB-Map=0.1242s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559444.568656062]: rtabmap (35): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1269s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559445.109529040]: rtabmap (36): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1349s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559445.664314149]: rtabmap (37): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1234s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559446.212887077]: rtabmap (38): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1382s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559446.813997925]: rtabmap (39): Rate=0.50s, Limit=0.000s, Conversion=0.0015s, RTAB-Map=0.1383s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559447.405197908]: rtabmap (40): Rate=0.50s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1306s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559448.073490820]: rtabmap (41): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1342s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559448.670185963]: rtabmap (42): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1312s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559449.303063974]: rtabmap (43): Rate=0.50s, Limit=0.000s, Conversion=0.0012s, RTAB-Map=0.1317s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1615559450.026916230]: rtabmap (44): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1219s, Maps update=0.0001s pub=0.0001s (local map=1, WM=1)
[ INFO] [1615559450.657874369]: rtabmap (45): Rate=0.50s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.1208s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
```

matlabbe, 2021-03-28:
To debug this, you may check if the topics are still published or not when rtabmap stops updating:
```
rostopic hz /t265/odom/sample \
  /l515/aligned_depth_to_color/image_raw \
  /l515/color/image_raw \
  /l515/color/camera_info
```
If one of those stops publishing, it would be a realsense issue. Verify the USB connection, with my laptop it happens often that when I move the USB, it messed up the image capture.

NorwegianPi, 2021-03-31:
Publishing rate seems to be as expected

```
                topic                     rate   min_delta   max_delta   std_dev    window
==========================================================================================
/t265/odom/sample                        200.3   3.815e-06   0.04921     0.01381    13248 
/l515/aligned_depth_to_color/image_raw   29.91   0.01486     0.288       0.007445   13248 
/l515/color/image_raw                    30.03   0.01525     0.05901     0.004085   1979  
/l515/color/camera_info                  30.02   0.01515     0.05887     0.00403    1979 
```

NorwegianPi, 2021-04-07:
During testing I noticed some strange behaviour from the L515. It seemed to generate different error messages depending on if it was screwed on to a bracket or held by hand. I aquired another L515 to test with and everything seems to works as expected. 
The strange thing is that the camera seemed to work fine in the RealSense SDK and with realsense_ros.

Will do some checking on FW versions and try the same FW on both cameras.
Will close this issue as it is not related to RTABMAP.

Thank you for your time @matlabbe

closed 2021-04-07 by NorwegianPi