introlab/rtabmap_ros #557: [FATAL] (2021-03-18 10:33:46.208) DBDriverSqlite3.cpp:306::getDatabaseVersionQuery() Condition (rc == SQLITE_DONE) not met! [DB error (0.20.7): no such table: Admin]

Today20180409, 2021-03-18:
When I use depth camera and lidar for SLAM，the program will run normally at the beginning,and I can get a grid map . But the following errors are reported and rtabmap stopped after a while:
[FATAL] (2021-03-18 10:33:46.208) DBDriverSqlite3.cpp:306::getDatabaseVersionQuery() Condition (rc == SQLITE_DONE) not met! [DB error (0.20.7): no such table: Admin]
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-03-18 10:33:46.208) DBDriverSqlite3.cpp:306::getDatabaseVersionQuery() Condition (rc == SQLITE_DONE) not met! [DB error (0.20.7): no such table: Admin]
[rtabmap-1] process has died [pid 14012, exit code -6, cmd /home/udrive/udrive_v1_1_1/ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start --uinfo rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info odom:=/odom_icp __name:=rtabmap __log:=/home/udrive/.ros/log/ee110e18-878d-11eb-8e02-00a73d681a98/rtabmap-1.log].
log file: /home/udrive/.ros/log/ee110e18-878d-11eb-8e02-00a73d681a98/rtabmap-1*.log

I have tried run the program source with different ways of source  compilation and binary installation, but the result was the same. my launch files as following:
<launch>
	<arg name="rgb_image" default="/camera/rgb/image_rect_color"/>
	<arg name="depth_image" default="/camera/depth_registered/image_raw"/>
	<arg name="rgb_camera_info" default="/camera/rgb/camera_info"/>
	<arg name="scan_topic" default="/scan"/>
	<arg name="odom_topic" default="/odom_icp"/>
   <arg name="rtabmap" default="true"/>

   <arg name="frame_id"                default="base_link"/>     
	
	
   <arg name="gftt_block_size" default="5" /> 
   <arg name="gftt_min_distance" default="5" /> 

	<arg name="cfg"                     default="" /> 
	<arg name="args"                    default=""/>              
	<arg name="rtabmap_args"            default="$(arg args)"/>   
	<arg name="launch_prefix"           default=""/>              
	<arg name="output"                  default="screen"/>        

	<arg name="odom_frame_id"           default="odom"/>                
	<arg name="map_frame_id"            default="map"/>
	<arg name="ground_truth_frame_id"   default=""/>     
	<arg name="ground_truth_base_frame_id" default=""/>  
	<arg name="namespace"               default="rtabmap"/>
	<arg name="database_path"           default="~/.ros/rtabmap.db"/>
	<arg name="queue_size"              default="10"/>
	<arg name="wait_for_transform"      default="0.2"/>
	
	<arg name="vo_frame_id"              default="$(arg odom_topic)"/> 
	<arg name="publish_tf_odom"          default="false"/>
	<arg name="odom_tf_angular_variance" default="1"/>             
	<arg name="odom_tf_linear_variance"  default="1"/>             
	<arg name="odom_args"                default=""/>              
	<arg name="odom_sensor_sync"         default="false"/>
	<arg name="odom_guess_frame_id"        default=""/>
	<arg name="odom_guess_min_translation" default="0"/>
	<arg name="odom_guess_min_rotation"    default="0"/>
	<arg name="odom_max_rate"            default="0"/>
	<arg name="odom_expected_rate"       default="0"/>
	<arg name="imu_topic"                default="/imu/data"/>          
	<arg name="wait_imu_to_init"         default="false"/>
	

   <node if="$(arg rtabmap)"  name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start --uinfo" >
		<remap from="rgb/image" to="$(arg rgb_image)"/>
		<remap from="depth/image" to="$(arg depth_image)"/>
		<remap from="rgb/camera_info" to="$(arg rgb_camera_info)"/>
		<remap from="odom" to="$(arg odom_topic)"/>

		<param name="subscribe_depth" type="bool" value="true"/>
		<param name="subscribe_scan" type="bool" value="true"/>
		<param name="subscribe_scan_cloud " type="bool" value="false"/>

		<param name="frame_id" type="string" value="$(arg frame_id)"/>
		<param name="map_frame_id" type="string" value="$(arg map_frame_id)"/>
		<param name="odom_frame_id" type="string" value="$(arg odom_frame_id)"/>
		<param name="odom_tf_linear_variance"  type="double" value="0.001"/>
	   <param name="odom_tf_angular_variance" type="double" value="0.001"/>

		<param name="approx_sync" type="bool" value="true"/>
		<param name="Grid/FromDepth" type="bool" value="false"/>

		<param name="wait_for_transform" type="bool" value="true"/>

		
		<param name="RGBD/NeighborLinkRefining" type="string" value="true"/>  
		<param name="RGBD/ProximityBySpace"     type="string" value="true"/>  
		<param name="RGBD/ProximityByTime"      type="string" value="false"/> 
		<param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/> 
		<param name="Reg/Strategy"              type="string" value="1"/>     
		<param name="Vis/MinInliers"        type="string" value="12"/>   
		<param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/> 
		<param name="RGBD/OptimizeMaxError"     type="string" value="3"/>	    
		<param name="Reg/Force3DoF"             type="string" value="true"/>  
		<param name="Grid/FromDepth"            type="string" value="false"/> 
		<param name="Mem/STMSize"               type="string" value="30"/>    
		<param name="RGBD/LocalRadius"          type="string" value="5"/>     
		<param name="Icp/CorrespondenceRatio"   type="string" value="0.4"/>   

		<param name="GFTT/BlockSize" type="string" value="$(arg gftt_block_size)"/>
		<param name="GFTT/MinDistance" type="string" value="$(arg gftt_min_distance)"/>
    </node> 
	 
    <arg name="icp_odometry"         default="false"/>
    <node if="$(arg icp_odometry)"  pkg="rtabmap_ros" type="icp_odometry" name="icp_odometry" output="$(arg output)" args="$(arg rtabmap_args) $(arg odom_args)" launch-prefix="$(arg launch_prefix)">
      <remap from="scan"                   to="$(arg scan_topic)"/>
      <remap from="odom"                   to="$(arg odom_topic)"/>
      <remap from="imu"                    to="$(arg imu_topic)"/>
      
      
      <param name="frame_id"                    type="string" value="$(arg frame_id)"/>
      <param name="odom_frame_id"               type="string" value="$(arg odom_frame_id)"/>
      <param name="publish_tf"                  type="bool"   value="$(arg publish_tf_odom)"/>
      <param name="ground_truth_frame_id"       type="string" value="$(arg ground_truth_frame_id)"/>
      <param name="ground_truth_base_frame_id"  type="string" value="$(arg ground_truth_base_frame_id)"/>
      <param name="wait_for_transform_duration" type="double" value="$(arg wait_for_transform)"/>
      <param name="wait_imu_to_init"            type="bool"   value="$(arg wait_imu_to_init)"/>
      <param name="config_path"                 type="string" value="$(arg cfg)"/>
      <param name="queue_size"                  type="int"    value="$(arg queue_size)"/>
      <param name="guess_frame_id"              type="string" value="$(arg odom_guess_frame_id)"/>
      <param name="guess_min_translation"       type="double" value="$(arg odom_guess_min_translation)"/>
      <param name="guess_min_rotation"          type="double" value="$(arg odom_guess_min_rotation)"/>
      <param name="expected_update_rate"        type="double" value="$(arg odom_expected_rate)"/>
      <param name="max_update_rate"             type="double" value="$(arg odom_max_rate)"/>
    </node>

</launch>
 
I have been bothered by this problem for a long time. Can anyone help me? 
thank you.

matlabbe, 2021-03-18:
Is the hard drive full at some point? Does it take a particular time to happen? On initialization or when closing?

Today20180409, 2021-03-19:
Thank you for your reply！
I've tested many times and found no regularity, not at the beginning or at the end。
I don't know if the hard disk is full when the program exits, and I'll test it.
What is the hard drive capacity requirement for running this program?

Today20180409, 2021-03-19:
@matlabbe  I've tested it to make sure there's a lot of space left on the hard disk

matlabbe, 2021-03-19:
`getDatabaseVersion` would be called from there https://github.com/introlab/rtabmap_ros/blob/8d053bc43f0045378406f0aa0c694244337e4b42/src/MapsManager.cpp#L544
but Admin table should always exists. There is the database schema: https://github.com/introlab/rtabmap/blob/21737f993723c2e5f92471ca153fb92edb92fd51/corelib/src/resources/DatabaseSchema.sql.in#L118-L119

Could you share the database created when this happens?

Today20180409, 2021-03-19:
Thank you again for your reply. I'm sorry, I'm a novice. I don't understand what you mean when you say database is specified? Does it refer to this file of  (~/.ros/rtabmap.db)? @matlabbe

matlabbe, 2021-03-19:
yes