introlab/rtabmap_ros #558: pcl:RadiusOutlierRemoval throttles point_cloud_assembler's output

victorll998, 2021-03-19:
Hi everyone,

I am new to ROS and have been testing with rtabmap nodelets with realsense d455 these days.

I found when I enable [RadiusOutlierRemoval](https://github.com/introlab/rtabmap_ros/blob/d586a8f0879d955e5754d18cfeeab8e1061183e7/src/nodelets/point_cloud_assembler.cpp#L316) in point_cloud_assembler nodelets, the output of /assembled_cloud drops from 3 hz to nearly 0 hz. I tested with stand-alone RadiusOutlierRemoval and it's supposed not to affect the output stream that significantly.

I wonder if anyone has encountered this before or any idea how to solve it. Thanks in advance. 

Parameters setting for point_cloud_assembler nodelets:
| | |
|--|--|
|frameId | camera_link|
|max_cloud | 5|
|fixed_frame_id| odom|
|noise_radius| 0.8|
|noise_min_neighbors|2|
|assembling_time (*i don't think it is used in the nodelet though it got defined*) | 1|

matlabbe, 2021-03-19:
What is the resolution or size of the cloud you are feeding to point_cloud_assembler?

victorll998, 2021-03-21:
Hi @matlabbe I used point_xyzrbgd nodelet to generate the cloud. It doesn't specify any height or width. But on average measurement the cloud has a height of 1 and a width of around 4000. The 5 scans fed into point_cloud_assembler to apply  RadiusOutlierRemoval make the total resolution of cloud 1*20000.

matlabbe, 2021-05-02:
A noise_radius of 0.8 is quite large, though I don't know what is the scale of your point cloud. I would use also use voxel_size before doing the radius filtering to downsample the point cloud.