introlab/rtabmap_ros #559: Failed to load nodelet '/camera/rgbd_sync`

thohemp, 2021-03-23:
I want to run the rgbd_sync nodelet for remote mapping, but it crashes when I try to start it.

Log:
`?[31m[ERROR] [1616508026.117511300]: Failed to load nodelet [/camera/rgbd_sync] of type [rtabmap_ros/rgbd_sync] even after refreshing the cache: Failed to load library /home/pal/deployed_ws/lib//librtabmap_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librtabmap_core.so.0.20: cannot open shared object file: No such file or directory)`

The library exists at that exact place (despite the odd double //). I tried to recompile and to reset the PATH, but it that doesn't solve it. Any idea for a solution?

matlabbe, 2021-05-02:
So this works:
`rosrun rtabmap_ros rgbd_sync`

but not this?
`rosrun nodelet nodelet standalone rtabmap_ros/rgbd_sync`

The first one is a wrapper of the nodelet, which is strange that one works and not the other.

MadhuriPatil1694, 2024-01-29:
I am also getting same error did you find any solution for this

matlabbe, 2024-01-30:
Which command did you use? Rgbd_sync is now in rtabmap_sync package, not rtabmap_ros.