introlab/rtabmap_ros #561: rtabmap_ros is not receiving data via topics from RealSense camera

alexandru-oana, 2021-03-30:
I'm running a D435i RealSense camera on ROS foxy, but rtabmap_ros is complaining that is not receiving any data.

```
rtabmapviz-3] rtabmapviz subscribed to (exact sync):
[rtabmapviz-3]    odom,
[rtabmapviz-3]    /camera/color/image_raw,
[rtabmapviz-3]    /camera/aligned_depth_to_color/image_raw,
[rtabmapviz-3]    rgb/camera_info
[rgbd_odometry-1] [WARN] [1617042581.953327412] [rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rgbd_odometry-1] rgbd_odometry subscribed to (exact sync):
[rgbd_odometry-1]    /camera/color/image_raw,
[rgbd_odometry-1]    /camera/aligned_depth_to_color/image_raw,
[rgbd_odometry-1]    rgb/camera_info
[rtabmap-2] [WARN] [1617042582.892121577] [rtabmap]: rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.

```
I've used the provided generic realsense_d400.launch.py launch file.

```
ros2 launch realsense2_camera rs_launch.py 
ros2 launch rtabmap_ros realsense_d400.launch.py
```

I've checked using **_ros2 topic echo_** that the realsense2_camera is publishing data; and they are also _**visibile in rviz2**_, but they don't reach rtabmap_ros2 node.

Any suggestion ?

matlabbe, 2021-04-04:
Hi, I updated the usage here to make sure you are using aligned depth to color topic: 
https://github.com/introlab/rtabmap_ros/blob/bbc953484c941d1a691267ddcbfe53ae3df00fbc/launch/ros2/realsense_d400.launch.py#L5-L6

Another thing is that your "rgb/camera_info" remap didn't work:
https://github.com/introlab/rtabmap_ros/blob/bbc953484c941d1a691267ddcbfe53ae3df00fbc/launch/ros2/realsense_d400.launch.py#L21

I had the same remap bug, but after trying many things, the remap started to work (don't know why?), now I cannot reproduce the bug anymore.

alexandru-oana, 2021-04-05:
Thanks ! Works now with the aligned_depth=true !

closed 2021-04-05 by alexandru-oana