introlab/rtabmap_ros #563: Trouble launching multiple zed2 cameras

peleito, 2021-04-03:
I'm trying to launch two ZED2 cameras for outdoor mapping. I have built rtabmap_ros from source with -DRTABMAP_SYNC_MULTI_RGBD=ON. 

My launch file:
```
<?xml version="1.0"?>
<launch>

   <!-- Choose visualization -->
   <arg name="rviz"       default="false" />

   <!-- <arg name="wait_for_transform"  default="true" /> -->
   
   
   <!-- Localization-only mode -->
    <arg name="localization"            default="false"/>

    <arg     if="$(arg localization)" name="args" default=""/>
    <arg unless="$(arg localization)" name="args" default="--delete_db_on_start"/>

   <!-- sync rgb/depth images per camera -->
   <group ns="zedA">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager1" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync1" args="load rtabmap_ros/rgbd_sync nodelet_manager1">
      <remap from="rgb/image"       to="zed_node_A/rgb/image_rect_color"/>
      <remap from="depth/image"     to="zed_node_A/depth/depth_registered"/>
      <remap from="rgb/camera_info" to="zed_node_A/rgb/camera_info"/>
      <param name="approx"     value="false"/>
    </node>
   </group>
   
   <group ns="zedB">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager2" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync2" args="load rtabmap_ros/rgbd_sync nodelet_manager2">
      <remap from="rgb/image"       to="zed_node_B/rgb/image_rect_color"/>
      <remap from="depth/image"     to="zed_node_B/depth/depth_registered"/>
      <remap from="rgb/camera_info" to="zed_node_B/rgb/camera_info"/>
      <param name="approx"     value="false"/>
    </node>
   </group>

  <group ns="rtabmap">
    <!-- Visual SLAM (robot side) -->
    <!-- args: "delete_db_on_start" and "udebug" -->
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" >
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="rgbd_cameras"     type="int"    value="2"/>
      <param name="frame_id"         type="string" value="base_link"/>
      <param name="gen_scan"         type="bool"   value="true"/>
      <param name="queue_size"       type="int"    value="100"/>

      <remap from="odom"              to="/odometry/filtered"/>	
      <remap from="rgbd_image0"       to="/zedA/rgbd_image"/>
      <remap from="rgbd_image1"       to="/zedB/rgbd_image"/>

      <param name="Grid/FromDepth"     type="string" value="false"/>
      <param name="Vis/EstimationType" type="string" value="0"/>
      
      <!-- localization mode -->
      <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
      <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
      <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>
    </node>
    
  </group>
  
  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/rgbd.rviz"/>

</launch>
[... 131 lines cut ...]
av788@av788-desktop:~/catkin_ws$ rostopic hz /zedB/rgbd_image
subscribed to [/zedB/rgbd_image]
average rate: 15.054
min: 0.058s max: 0.072s std dev: 0.00376s window: 15
average rate: 15.039
min: 0.025s max: 0.111s std dev: 0.01179s window: 30
average rate: 14.987
min: 0.025s max: 0.111s std dev: 0.00993s window: 45
^Caverage rate: 15.030
min: 0.025s max: 0.111s std dev: 0.00954s window: 49
av788@av788-desktop:~/catkin_ws$ rostopic hz /odometry/filtered
subscribed to [/odometry/filtered]
no new messages
no new messages
^Cno new messages
```
So I tried changing my launch file by switching the odom remap to the odometry data from a single camera.
`<remap from="odom"                 to="/zedB/zed_node_B/odom"/>`
After that change I now get this error, which has me lost on the next steps. Maybe my launch file change was not correct and I should go back.
```
[ INFO] [1617411817.944935601]: rtabmap 0.20.9 started...
[rtabmap/rtabmap-5] process has died [pid 20892, exit code -11, cmd /home/av788/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start odom:=/zedB/zed_node_B/odom rgbd_image0:=/zedA/rgbd_image rgbd_image1:=/zedB/rgbd_image __name:=rtabmap __log:=/home/av788/.ros/log/5d4efa1a-9418-11eb-a06c-fc44827e1bb1/rtabmap-rtabmap-5.log].
log file: /home/av788/.ros/log/5d4efa1a-9418-11eb-a06c-fc44827e1bb1/rtabmap-rtabmap-5*.log
```
Any help would be greatly appreciated.

matlabbe, 2021-04-03:
Can you change this line:
```xml
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" >
```
to 
```xml
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" launch-prefix="xterm -e gdb --args">
```

Type "run" in gdb terminal, then when it crashes, type "bt" (backtrace) to see where what caused the seg fault.

I tried with this launch:
```xml
<?xml version="1.0"?>
<launch>

   <!-- Choose visualization -->
   <arg name="rviz"       default="false" />
   
   <!-- Localization-only mode -->
    <arg name="localization"            default="false"/>

    <arg     if="$(arg localization)" name="args" default=""/>
    <arg unless="$(arg localization)" name="args" default="--delete_db_on_start"/>

   <!-- sync rgb/depth images per camera -->
   <group ns="zedA">    
    <param name="zed_node_A/pos_tracking/publish_tf" value="true"/>
    <param name="zed_node_A/pos_tracking/publish_map_tf" value="false"/>  
    <include file="$(find zed_wrapper)/launch/include/zed_camera.launch.xml">
       <arg name="node_name"         value="zed_node_A" />
       <arg name="camera_name"       value="zedA" />
       <arg name="camera_id"         value="0" />
       <arg name="camera_model"      value="zed2" />
    </include> 
    <node pkg="nodelet" type="nodelet" name="rgbd_sync1" args="standalone rtabmap_ros/rgbd_sync nodelet_manager1">
      <remap from="rgb/image"       to="zed_node_A/rgb/image_rect_color"/>
      <remap from="depth/image"     to="zed_node_A/depth/depth_registered"/>
      <remap from="rgb/camera_info" to="zed_node_A/rgb/camera_info"/>
      <param name="approx"     value="false"/>
    </node>
   </group>
   
   <group ns="zedB">
    <param name="zed_node_B/pos_tracking/publish_tf" value="false"/>
    <param name="zed_node_B/pos_tracking/publish_map_tf" value="false"/>
    <include file="$(find zed_wrapper)/launch/include/zed_camera.launch.xml">
       <arg name="node_name"         value="zed_node_B" />
       <arg name="camera_name"       value="zedB" />
       <arg name="camera_id"         value="1" />
       <arg name="camera_model"      value="zedm" />
       <arg name="base_frame"        value="zedB_base_link" />
    </include>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync2" args="standalone rtabmap_ros/rgbd_sync nodelet_manager2">
      <remap from="rgb/image"       to="zed_node_B/rgb/image_rect_color"/>
      <remap from="depth/image"     to="zed_node_B/depth/depth_registered"/>
      <remap from="rgb/camera_info" to="zed_node_B/rgb/camera_info"/>
      <param name="approx"     value="false"/>
    </node>
   </group>
   
   <node pkg="tf" type="static_transform_publisher" name="zedA_to_zedB"
        args="0 0 0 3.14159 0 0 base_link zedB_base_link 100" /> 

  <group ns="rtabmap">
    <!-- Visual SLAM (robot side) -->
    <!-- args: "delete_db_on_start" and "udebug" -->
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" >
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="rgbd_cameras"     type="int"    value="2"/>
      <param name="frame_id"         type="string" value="base_link"/>
      <param name="gen_scan"         type="bool"   value="true"/>
      <param name="queue_size"       type="int"    value="100"/>

      <remap from="odom"              to="/zedA/zed_node_A/odom"/>	
      <remap from="rgbd_image0"       to="/zedA/rgbd_image"/>
      <remap from="rgbd_image1"       to="/zedB/rgbd_image"/>

      <param name="Grid/FromDepth"     type="string" value="false"/>
      <param name="Vis/EstimationType" type="string" value="0"/>
      
      <!-- localization mode -->
      <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
      <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
      <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>
    </node>
    
  </group>
  
  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/rgbd.rviz"/>

</launch>
```
To use only zedA as odometry, we have to remove/commend those [lines](https://github.com/stereolabs/zed-ros-wrapper/blob/6c11b36aeb1208614933493d793e33974be84576/zed_wrapper/params/common.yaml#L47-L48) from common.yaml (see above that those zed parameters `publish_tf` and `publish_map_tf` are set in this launch file). Setting VGA resolution [here](https://github.com/stereolabs/zed-ros-wrapper/blob/6c11b36aeb1208614933493d793e33974be84576/zed_wrapper/params/common.yaml#L26) also helps to use less power. Note that in the launch above, I used a zedm and a zed2 that were stick one over the other at 180 deg (that static transform could be more precise though).

peleito, 2021-04-04:
Thank you so much for your response, I am very grateful. I tried your launch file, also with a ZED2 and ZEDm for consistency, to try something that you know works, made the adjustments the the common.yaml file, and I changed:
`<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" >`
to
`<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" launch-prefix="xterm -e gdb --args">`
I proceded to run in gdb and backtraced as you suggested. I'm not quite sure what any of this means, as I am new to most of this stuff. If there is any other information that I could provide which would be helpful, plese let me know.
```
Starting program: /home/av788/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start odom:=/zedA/zed_node_A/odom rgbd_image0:=/zedA/rgbd_image rgbd_image1:=/zedB/rgbd_image __name:=rtabmap __log:=/home/av788/.ros/log/67260af4-958c-11eb-8e90-fc44827e1bb1/rtabmap-rtabmap-9.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
(gdb) [New Thread 0x7fffa58b9700 (LWP 16174)]
[New Thread 0x7fffa50b8700 (LWP 16175)]
[New Thread 0x7fffa08b7700 (LWP 16176)]
[New Thread 0x7fff9e0b6700 (LWP 16177)]
[New Thread 0x7fff9b8b5700 (LWP 16178)]
[New Thread 0x7fff990b4700 (LWP 16179)]
[New Thread 0x7fff988b3700 (LWP 16180)]
[New Thread 0x7fff940b2700 (LWP 16181)]
[New Thread 0x7fff918b1700 (LWP 16182)]
[New Thread 0x7fff8f0b0700 (LWP 16183)]
[New Thread 0x7fff8c8af700 (LWP 16184)]
[New Thread 0x7fff8a0ae700 (LWP 16185)]
[New Thread 0x7fff898ad700 (LWP 16186)]
[New Thread 0x7fff890ac700 (LWP 16187)]
[New Thread 0x7fff888ab700 (LWP 16188)]
[New Thread 0x7fff880aa700 (LWP 16189)]
[New Thread 0x7fff878a9700 (LWP 16190)]
[New Thread 0x7fff870a8700 (LWP 16191)]
[New Thread 0x7fff868a7700 (LWP 16192)]
[New Thread 0x7fff860a6700 (LWP 16193)]
[New Thread 0x7fff858a5700 (LWP 16194)]
[New Thread 0x7fff850a4700 (LWP 16195)]
[New Thread 0x7fff848a3700 (LWP 16196)]
[ INFO] [1617571600.823494664]: Starting node...
[New Thread 0x7fff83860700 (LWP 16198)]
[New Thread 0x7fff8305f700 (LWP 16199)]
[New Thread 0x7fff8285e700 (LWP 16200)]
[New Thread 0x7fff8205d700 (LWP 16201)]
[New Thread 0x7fff8185c700 (LWP 16202)]
[New Thread 0x7fff8105b700 (LWP 16203)]
[New Thread 0x7fff8085a700 (LWP 16204)]
[New Thread 0x7fff73fff700 (LWP 16205)]
[New Thread 0x7fff737fe700 (LWP 16206)]
[New Thread 0x7fff72ffd700 (LWP 16207)]
[New Thread 0x7fff727fc700 (LWP 16208)]
[New Thread 0x7fff71ffb700 (LWP 16209)]
[New Thread 0x7fff717fa700 (LWP 16210)]
[New Thread 0x7fff70ff9700 (LWP 16211)]
[New Thread 0x7fff707f8700 (LWP 16212)]
[New Thread 0x7fff6fff7700 (LWP 16213)]
[New Thread 0x7fff6f7f6700 (LWP 16214)]
[ INFO] [1617571600.863423672]: Initializing nodelet with 12 worker threads.
[New Thread 0x7fff6eff5700 (LWP 16215)]
[New Thread 0x7fff67933700 (LWP 16216)]
[ INFO] [1617571601.299379541]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1617571601.299404137]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1617571601.299413505]: /rtabmap/rtabmap(maps): map_cleanup                = true
[ INFO] [1617571601.299423690]: /rtabmap/rtabmap(maps): map_always_update          = false
[ INFO] [1617571601.299434116]: /rtabmap/rtabmap(maps): map_empty_ray_tracing      = true
[ INFO] [1617571601.299443956]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1617571601.299453920]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1617571601.299465598]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1617571601.320653526]: /rtabmap/rtabmap(maps): octomap_tree_depth         = 16
[ INFO] [1617571601.330735495]: rtabmap: frame_id      = base_link
[ INFO] [1617571601.330759940]: rtabmap: map_frame_id  = map
[ INFO] [1617571601.330769444]: rtabmap: use_action_for_goal  = false
[ INFO] [1617571601.330782442]: rtabmap: tf_delay      = 0.050000
[ INFO] [1617571601.330792081]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1617571601.330801462]: rtabmap: odom_sensor_sync   = false
[ INFO] [1617571601.331038760]: rtabmap: gen_scan  = true
[ INFO] [1617571601.331055410]: rtabmap: gen_scan_max_depth  = 4.000000
[... 58 lines cut ...]
#6  0x00007fff6dfdcf9d in rtabmap_ros::CoreWrapper::process(ros::Time const&, rtabmap::SensorData&, rtabmap::Transform const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Mat const&, rtabmap::OdometryInfo const&, double) ()
    at /home/av788/catkin_ws/devel/lib//librtabmap_sync.so
#7  0x00007fff6dfe10a5 in rtabmap_ros::CoreWrapper::commonDepthCallbackImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::shared_ptr<rtabmap_ros::UserData_<std::allocator<void> > const> const&, std::vector<boost::shared_ptr<cv_bridge::CvImage const>, std::allocator<boost::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<boost::shared_ptr<cv_bridge::CvImage const>, std::allocator<boost::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<sensor_msgs::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::CameraInfo_<std::allocator<void> > > > const&, sensor_msgs::LaserScan_<std::allocator<void> > const&, sensor_msgs::PointCloud2_<std::allocator<void> > const&, boost::shared_ptr<rtabmap_ros::OdomInfo_<std::allocator<void> > const> const&, std::vector<rtabmap_ros::GlobalDescriptor_<std::allocator<voi [... cut ...]
#22 0x00007ffff79b125d in nodelet::detail::CallbackQueue::callOne() () at /opt/ros/melodic/lib/libnodeletlib.so
#23 0x00007ffff79b3c05 in nodelet::detail::CallbackQueueManager::workerThread(nodelet::detail::CallbackQueueManager::ThreadInfo*) () at /opt/ros/melodic/lib/libnodeletlib.so
#24 0x00007ffff3166bcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#25 0x00007ffff401d6db in start_thread (arg=0x7fff8105b700) at pthread_create.c:463
#26 0x00007ffff435671f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

peleito, 2021-04-09:
I tried running it again after a running:
```
sudo apt update
sudo apt upgrade
```
and making a clean build of my catkin_ws but got a different error this time.
```
(gdb) run
Starting program: /home/av788/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start odom:=/zedA/zed_node_A/odom rgbd_image0:=/zedA/rgbd_image rgbd_image1:=/zedB/rgbd_image __name:=rtabmap __log:=/home/av788/.ros/log/720b3ae6-9890-11eb-a7ef-fc44827e1bb1/rtabmap-rtabmap-9.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffb6f28700 (LWP 5962)]
[New Thread 0x7fffb6727700 (LWP 5963)]
[New Thread 0x7fffb1f26700 (LWP 5964)]
[New Thread 0x7fffaf725700 (LWP 5965)]
[New Thread 0x7fffacf24700 (LWP 5966)]
[New Thread 0x7fffaa723700 (LWP 5967)]
[New Thread 0x7fffa7f22700 (LWP 5968)]
[New Thread 0x7fffa5721700 (LWP 5969)]
[New Thread 0x7fffa2f20700 (LWP 5970)]
[New Thread 0x7fffa071f700 (LWP 5971)]
[New Thread 0x7fff9df1e700 (LWP 5972)]
[New Thread 0x7fff9b71d700 (LWP 5973)]
[New Thread 0x7fff9af1c700 (LWP 5977)]
[New Thread 0x7fff9a71b700 (LWP 5978)]
[New Thread 0x7fff99f1a700 (LWP 5979)]
[New Thread 0x7fff99719700 (LWP 5980)]
[New Thread 0x7fff98f18700 (LWP 5981)]
[New Thread 0x7fff98717700 (LWP 5982)]
[New Thread 0x7fff97f16700 (LWP 5983)]
[New Thread 0x7fff97715700 (LWP 5984)]
[New Thread 0x7fff96f14700 (LWP 5985)]
[New Thread 0x7fff96713700 (LWP 5986)]
[New Thread 0x7fff95f12700 (LWP 5987)]
[ INFO] [1617903181.137162267]: Starting node...
[New Thread 0x7fff94ecf700 (LWP 5989)]
[New Thread 0x7fff8ffff700 (LWP 5990)]
[New Thread 0x7fff87fff700 (LWP 5991)]
[New Thread 0x7fff8f7fe700 (LWP 5992)]
[New Thread 0x7fff8effd700 (LWP 5993)]
[New Thread 0x7fff8e7fc700 (LWP 5994)]
[New Thread 0x7fff8dffb700 (LWP 5995)]
[New Thread 0x7fff8d7fa700 (LWP 5996)]
[New Thread 0x7fff8cff9700 (LWP 5997)]
[New Thread 0x7fff877fe700 (LWP 5998)]
[New Thread 0x7fff86ffd700 (LWP 5999)]
[New Thread 0x7fff867fc700 (LWP 6000)]
[New Thread 0x7fff85ffb700 (LWP 6001)]
[New Thread 0x7fff857fa700 (LWP 6002)]
[New Thread 0x7fff84ff9700 (LWP 6003)]
[New Thread 0x7fff7ffff700 (LWP 6004)]
[New Thread 0x7fff7f7fe700 (LWP 6005)]
[ INFO] [1617903181.176436993]: Initializing nodelet with 12 worker threads.
[New Thread 0x7fff7effd700 (LWP 6006)]
[New Thread 0x7fff75686700 (LWP 6007)]
[ INFO] [1617903184.479741733]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1617903184.479796604]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1617903184.479824032]: /rtabmap/rtabmap(maps): map_cleanup                = true
[ INFO] [1617903184.479849145]: /rtabmap/rtabmap(maps): map_always_update          = false
[ INFO] [1617903184.479872906]: /rtabmap/rtabmap(maps): map_empty_ray_tracing      = true
[ INFO] [1617903184.479898645]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1617903184.479924374]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1617903184.479946192]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1617903184.488912396]: /rtabmap/rtabmap(maps): octomap_tree_depth         = 16
[ INFO] [1617903184.506776351]: rtabmap: frame_id      = base_link
[ INFO] [1617903184.506800005]: rtabmap: map_frame_id  = map
[ INFO] [1617903184.506811878]: rtabmap: use_action_for_goal  = false
[ INFO] [1617903184.506824881]: rtabmap: tf_delay      = 0.050000
[ INFO] [1617903184.506838749]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1617903184.506850999]: rtabmap: odom_sensor_sync   = false
[... 99 lines cut ...]
[Thread 0x7fff98717700 (LWP 5982) exited]
[Thread 0x7fff98f18700 (LWP 5981) exited]
[Thread 0x7fff99719700 (LWP 5980) exited]
[Thread 0x7fff99f1a700 (LWP 5979) exited]
[Thread 0x7fff9a71b700 (LWP 5978) exited]
[Thread 0x7fff9af1c700 (LWP 5977) exited]
[Thread 0x7fff9b71d700 (LWP 5973) exited]
[Thread 0x7fff9df1e700 (LWP 5972) exited]
[Thread 0x7fffa071f700 (LWP 5971) exited]
[Thread 0x7fffa2f20700 (LWP 5970) exited]
[Thread 0x7fffa5721700 (LWP 5969) exited]
[Thread 0x7fffa7f22700 (LWP 5968) exited]
[Thread 0x7fffaa723700 (LWP 5967) exited]
[Thread 0x7fffacf24700 (LWP 5966) exited]
[Thread 0x7fffaf725700 (LWP 5965) exited]
[Thread 0x7fffb1f26700 (LWP 5964) exited]
[Thread 0x7fffb6727700 (LWP 5963) exited]
[Thread 0x7fffb6f28700 (LWP 5962) exited]
[Thread 0x7ffff7f18a00 (LWP 5955) exited]
[Inferior 1 (process 5955) exited with code 0177]
(gdb) bt
No stack.
(gdb) 
```
I'm guessing this is due to some sort of library mismatch or because I installed rtabmap or one of its dependencies wrong, but I could be wrong. I don't know if it is best to just start over, delete rtabmap, and reinstall from source, or if this can easily be fixed.