introlab/rtabmap_ros #564: Error launching rtabmap with D435i

MrOCW, 2021-04-05:
Running on Jetson Xavier NX, built from source with ORB-SLAM enabled.

$ roslaunch realsense2_camera rs_camera.launch align_depth:=true unite_imu_method:="linear_interpolation" enable_gyro:=true enable_accel:=true

$ rosrun imu_filter_madgwick imu_filter_node \
    _use_mag:=false \
    _publish_tf:=false \
    _world_frame:="enu" \
    /imu/data_raw:=/camera/imu \
    /imu/data:=/rtabmap/imu

$ roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false \
    wait_imu_to_init:=true \
    imu_topic:=/rtabmap/imu
    rtabmapviz:=false \
    rviz:=true

[ERROR] [1617606586.398766976]: Failed to load nodelet [/points_xyzrgb] of type [rtabmap_ros/point_cloud_xyzrgb] even after refreshing the cache: Failed to load library /home/jetson/catkin_ws/devel/lib//librtabmap_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libg2o.so: cannot open shared object file: No such file or directory)

[ERROR] [1617606586.402163862]: The error before refreshing the cache was: Failed to load library /home/jetson/catkin_ws/devel/lib//librtabmap_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libg2o.so: cannot open shared object file: No such file or directory)
[points_xyzrgb-4] process has died [pid 10304, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet standalone rtabmap_ros/point_cloud_xyzrgb rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay cloud:=voxel_cloud __name:=points_xyzrgb __log:=/home/jetson/.ros/log/89f2a28a-95dd-11eb-afde-70665563d293/points_xyzrgb-4.log].
log file: /home/jetson/.ros/log/89f2a28a-95dd-11eb-afde-70665563d293/points_xyzrgb-4*.log

[ERROR] [1617606588.045238294]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapCloud' failed to load.  Error: Failed to load library /home/jetson/catkin_ws/devel/lib//librtabmap_rviz_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libg2o.so: cannot open shared object file: No such file or directory)

[ERROR] [1617606588.053910206]: PluginlibFactory: The plugin for class 'rtabmap_ros/Info' failed to load.  Error: Failed to load library /home/jetson/catkin_ws/devel/lib//librtabmap_rviz_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libg2o.so: cannot open shared object file: No such file or directory)

[ERROR] [1617606588.061254961]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapGraph' failed to load.  Error: Failed to load library /home/jetson/catkin_ws/devel/lib//librtabmap_rviz_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libg2o.so: cannot open shared object file: No such file or directory)

MrOCW, 2021-04-12:
Solved.

closed 2021-04-12 by MrOCW

matlabbe, 2021-05-04:
Hi, while this is solved, you specified "ORB-SLAM enabled". If you want to use ORB_SLAM odometry, you have to add `--Odom/Strategy 5` in `rtabmap_args`.