introlab/rtabmap_ros #566: Ouster example not compatible with recent ouster-lidar package update

inceptionjs, 2021-04-08:
The test_ouster.launch file in the tests sub-folder does not seem to run properly when using the most recent ouster-lidar package.

The main issue seems to be that Ouster changed the names of nodes and arguments in the ouster.launch file (which used to be os1.launch in previous versions).

[https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/ouster.launch](url)

I am unsure as to how to properly update variables in the test_ouster.launch file and other related files to launch rtabmap with the ouster lidar.

Any help would be much appreciated.

matlabbe, 2021-04-08:
Can you try https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_ouster_gen2.launch ?

inceptionjs, 2021-04-15:
This worked! Thank you.

On a side note, when installing rtabmap with libpointmatcher, are there any extra steps that should be taken, or should I just follow the installation tutorials? I ask because I had to reinstall rtabmap as it did not recognize libpointmatcher even though I had it installed.

Also, are there any parameters you would suggest tweaking if I were to use the LiDAR outdoors? I did some initial testing and the results were not that great, especially for the imu data.

matlabbe, 2021-05-04:
Is libpointmacher installed? (make install)

When building rtabmap, cmake should be then able to find it, unless there is something different on how libpointmatcher is installed in latest version.

For outdoor, you may take a look at this post: http://official-rtab-map-forum.67519.x6.nabble.com/Ouster-drive-under-the-rain-td6496.html

leramleram, 2023-02-23:
I want to follow up on this topic.
I´m trying to figure out if rtabmap could be a solution for our purposes (inside/outside mapping and localization)

It seems rtabmap´s ouster test launch files are not working any more with the actual ouster packages.
(Neither ouster.launch nor ouster_gen2.launch)
Problem starts already when launching the rtab test_ouster_gen2 node.
It doesn´t find `ouster.launch`

matlabbe, 2023-02-26:
You may try changing `ouster.launch` for `sensor.launch`. You may have to remap the pointcloud and imu topics. I cannot test it but from 
https://github.com/introlab/rtabmap_ros/blob/3cde10ee3be29560b5c4af18bd0d28fc2323ae8c/launch/tests/test_ouster_gen2.launch#L50-L63
it could look like this:
```xml
    <arg name="imu_topic"     default="/ouster/imu"/>
    <arg name="scan_topic"    default="/ouster/points"/>

    <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>

    <!-- Ouster -->
    <include unless="$(arg use_sim_time)" file="$(find ouster_ros)/launch/sensor.launch">
      <arg name="viz" value="false"/>
      <arg name="sensor_hostname" value="$(arg sensor_hostname)"/>
      <arg name="udp_dest" value="$(arg udp_dest)"/>
      <arg name="image" value="true"/>
      <arg     if="$(arg scan_20_hz)" name="lidar_mode" value="1024x20"/>
      <arg unless="$(arg scan_20_hz)" name="lidar_mode" value="1024x10"/>
      <arg if="$(arg ptp)" name="timestamp_mode" value="TIME_FROM_PTP_1588"/>
    </include>
```

leramleram, 2023-02-26:
Yeah, that´s what I did.
Would be great to have it done globally in this repo.

What I also noticed is that the timebase from the ouster sensor and ros are different.
When time from ptp1588 is not set, the timebase for the ouster is time from startup but ros calculates in epoch.

Are pull requests welcome?

matlabbe, 2023-02-27:
Sure! pull requests are welcome.

For the ouster stamps, you indeed make sure you are using PTP if integrating with other ROS packages using ros time. I think this example can work without PTP (rtabmap would use stamps inside the topics), but I don't have access to an ouster anymore to check this...

leramleram, 2023-02-27:
Regarding the timestamps: it works out of the box when timestamp mode is set to `TIME_FROM_ROS_TIME` when PTP is not set.

I suggest it like this:
[image]

matlabbe, 2023-04-15:
Added it here: https://github.com/introlab/rtabmap_ros/commit/3e140a04e6d4827cb913825c19a06dc8626f790c

closed 2023-04-15 by matlabbe