introlab/rtabmap_ros #569: Is there a parameter to change the resolution of /rtabmap/octomap_binary

TouchDeeper, 2021-04-13:
Hello,

I list the parameters by:
```
rosrun rtabmap_ros rtabmap --params
rosrun rtabmap_ros stereo_odometry --params
```
But I can not find the parameters to modify the resolution of `/rtabmap/octomap_binary`.
Could you please provide a quick way to modify the resolution of `/rtabmap/octomap_binary`? Thanks a lot.

matlabbe, 2021-05-04:
Parameter `Grid/CellSize` will be used to set OctoMap resolution at Tree Depth=16 (highest resolution). `/rtabmap/octomap_binary` returns always the whole OctoMap. See also doc of http://docs.ros.org/en/hydro/api/octomap_msgs/html/msg/Octomap.html

However, the PointCloud2 outputs of `/rtabmap/octomap_*` can be generated at a specified depth using `octomap_tree_depth` parameter.

TouchDeeper, 2021-05-12:
@matlabbe
Thanks for your reply.

closed 2021-05-12 by TouchDeeper