introlab/rtabmap_ros #570: Value of queue_size in L515 + T265

sumitsarkar1, 2021-04-13:
1. Ubuntu 18 with ROS Melodic
2. librealsense and rtabmap and rtabmap_ros build from source
3. Using T265 with L515 rtabmap_ros stops abruptly after some time
4. When queue_size is 10 (default value) it stops early..like after 30 seconds to 40 seconds...for larger values like 200 it continues (I did not check how long it continues)..however in that case pause button does not stop the streaming which I guess is not the live streaming but the already recorded data due to large buffer (I may be wrong in this assumption)
5. Is this T265 +L515 dependent on queue_size which is little difficult to believe because with stand alone RTabMap L515 and T265 works flawlessly for long hours
6. I followed #555 and #574 and #1078
7. The rtabmap_ros gets abruptly paused...no new map is build in the gui and in the terminal its just some update info lines shows up and then suddenly no update info line starts appearing 
`[ INFO] [1618329073.723286023]: rtabmap (27): Rate=1.00s, Limit=0.000s, Conversion=0.0009s, RTAB-Map=0.0878s, Maps update=0.0015s pub=0.0000s (local map=21, WM=21)
`
8. The topics are still alive . I have checked it with rviz and rostopic hz and have constant update rates ...the same when the L515 and T265 are launched
9. Again , with stand alone RTabMap T265 + L515 works flawless. I just have a remote application to implement for which I need to stream the topics over wireless and do the mapping at a remote compute centre
10.  Following are my launch files . Please suggest if I am missing something

L515 + T265 launch file :

`<launch>
  <arg name="device_type_t265"    		default="t265"/>
  <arg name="device_type_l515"    		default="l515"/>
  <arg name="device_type_d435"    		default="d4.5"/>
  
  <arg name="serial_no_t265"    			default="2322110234"/>
  <arg name="serial_no_l515"    			default="f0231832"/>
  <arg name="serial_no_d435"    			default=""/>
  
  <arg name="t265_ns"              			default="t265"/>
  <arg name="l515_ns"              			default="l515"/>
  <arg name="d435_ns"              			default="d435"/>

  <arg name="tf_prefix_t265"         default="$(arg t265_ns)"/>
  <arg name="tf_prefix_l515"         default="$(arg l515_ns)"/>
  <arg name="tf_prefix_d435"         default="$(arg d435_ns)"/>

  <arg name="initial_reset"             default="false"/>
  <arg name="enable_fisheye"            default="false"/>

  <arg name="color_width"               default="960"/>
  <arg name="color_height"              default="540"/>
  <arg name="depth_width"               default="640"/>
  <arg name="depth_height"              default="480"/>
  <arg name="ordered_pc"                default="false"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>

  <param name="/t265/realsense2_camera/frames_queue_size" type="int" value="30"/>
  <param name="/t265/realsense2_camera/enable_relocalization" type="bool" value="false"/>
  <group ns="$(arg t265_ns)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="device_type"           value="$(arg device_type_t265)"/>
      <arg name="serial_no"             value="$(arg serial_no_t265)"/>
      <arg name="tf_prefix"         	  value="$(arg tf_prefix_t265)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
      <arg name="enable_fisheye1"       value="$(arg enable_fisheye)"/>
      <arg name="enable_fisheye2"       value="$(arg enable_fisheye)"/>
      <arg name="topic_odom_in"         value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"       value="$(arg calib_odom_file)"/>
      <arg name="enable_pose"           value="true"/>
      <arg name="unite_imu_method"	value="linear_interpolation"/>
	
      
    </include>
  </group>
  
  
  <group ns="$(arg l515_ns)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="device_type"           value="$(arg device_type_l515)"/>
      <arg name="serial_no"             value="$(arg serial_no_l515)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_l515)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
[... 410 lines cut ...]
  
  
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
  </node>

</launch>`

launching parameters for rtabmap_ros 

`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="-d --Mem/UseOdomGravity true --Optimizer/GravitySigma 0.3 --delete_db_on_start" odom_topic:=/t265/odom/sample frame_id:=t265_link rgbd_sync:=true depth_topic:=/l515/aligned_depth_to_color/image_raw    rgb_topic:=/l515/color/image_raw camera_info_topic:=/l515/color/camera_info    approx_rgbd_sync:=false visual_odometry:=false queue_size:=200
`

matlabbe, 2021-05-04:
If there is no error, it may means that rtabmap is not able to synchronize topics form T265 and L515 after some time. Maybe the timestamp set in odometry topic drift in comparison to timsestamps set in L515 topics.

You may compare `/rtabmap/rgbd_image/header/stamp` with `/t265/odom/sample/header/stamp` if they are always in sync. This may be why increasing queue_size helps to get data synchronized a little longer but will eventually fail.

With the standalone, we restamp pose and image topics from both camera at the time of receiving it on the computer. Check if there are parameters to chose computer time instead of camera time for realsense cameras. There is a realsense parameter called `RS2_OPTION_GLOBAL_TIME_ENABLED` that may be related.

gongyue666, 2021-11-17:
I face the same problem, the parameter "RS2_OPTION_GLOBAL_TIME_ENABLED" should be set in rs_l515_and_t265.launch?

sumitsarkar1, 2021-12-01:
@gongyue666 any luck in setting that flag

gongyue666, 2021-12-01:
the warnings are disappeared, but the final result is not very good in my case

sumitsarkar1, 2021-12-01:
final result as in the 3D map or still the streaming cannot be stopped?

gongyue666, 2021-12-01:
final result as in the 3D map performed badly, easily lost~

shubhamwagh, 2021-12-01:
@gongyue666 How do you set **RS2_OPTION_GLOBAL_TIME_ENABLED** in the launch file?

shubhamwagh, 2021-12-01:
_rtabmap_ stand-alone application works flawlessly for long hours with L515 + T265. But _rtabmap-ros_ struggles to run the same configuration and for some reason, it is dependent **queue_size** value. I did try to put very high value of queue-size but the final map quality is not great as compared to running rtabmap standalone application.

gongyue666, 2021-12-02:
even D435+T265 is better than L515+T265

sumitsarkar1, 2021-12-07:
@shubhamwagh were you able to run standalone RTabmap via ssh? Like the onboard compute (on the robot) runs SLAM using RTabMap and you can start and stop the mapping using RTabmap GUI from a remote computer. And ya L515 maps are more "crisp" (like walls at a distance more than 5 mtrs) than depth camera maps

shubhamwagh, 2021-12-07:
@sumitsarkar1 , yes I am able to run standalone RTAB map via ssh display forward. You can check the [gist](https://gist.github.com/shubhamwagh/5bdda4f8fc7cab6bb4d38b604d3d3a73#file-ssh_display_forward-md) I have created to forward display over ssh. You can follow the steps to forward the display to the client.

The setup of my system is as follows:

Robot  (Server):
Single Board computer (Latte Panda Alpha)
L515 + T265 connected to Panda.

Laptop (Client):
ssh robot and run rtabmap. Rtabmap GUI opens on client side.

Robot and Laptop are wired connected via router.

sumitsarkar1, 2021-12-07:
@shubhamwagh thanx a lot...I had some success using ssh and Nautilus...however over wireless there is latency ...but thanx for the forward display suggestions..I will try them

shubhamwagh, 2022-01-14:
@sumitsarkar1 ,  I got RTAB-MAP for L515 + T265 working with ROS now.

Looks like by default global_time_sync for L515 is set to false. Add the following rosparams before launching l515. Then it works fine without an issue :D.

```xml
  <param name="/l515/motion_module/global_time_enabled" type="bool" value="true"/>
  <param name="/l515/l500_depth_sensor/global_time_enabled" type="bool" value="true"/>
  <param name="/l515/rgb_camera/global_time_enabled" type="bool" value="true"/>
```

Here is my complete launch file for sensor launch

```xml
<launch>
  <arg name="device_type_t265"    		default="t265"/>
  <arg name="device_type_l515"    		default="l515"/>
  
  <arg name="serial_no_t265"    			default=""/>
  <arg name="serial_no_l515"    			default=""/>
  
  <arg name="t265_ns"              			default="t265"/>
  <arg name="l515_ns"              			default="l515"/>

  <arg name="tf_prefix_t265"         default="$(arg t265_ns)"/>
  <arg name="tf_prefix_l515"         default="$(arg l515_ns)"/>

  <arg name="initial_reset"             default="false"/>
  <arg name="enable_fisheye"            default="false"/>

  <arg name="color_width"               default="1280"/>
  <arg name="color_height"              default="720"/>
  <arg name="depth_width"               default="640"/>
  <arg name="depth_height"              default="480"/>
  <arg name="ordered_pc"                default="false"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="topic_odom_in"             default=""/>
  <arg name="calib_odom_file"           default=""/>

  
  <param name="/t265/tracking_module/frames_queue_size" type="int" value="16"/>
  <param name="/t265/realsense2_camera/enable_relocalization" type="bool" value="false"/>
  
  <param name="/l515/motion_module/global_time_enabled" type="bool" value="true"/>
  <param name="/l515/l500_depth_sensor/global_time_enabled" type="bool" value="true"/>
  <param name="/l515/rgb_camera/global_time_enabled" type="bool" value="true"/>
  
  <!--remap from="/$(arg tf_prefix_t265)/odom/sample" to="/odom" /-->

  <group ns="$(arg t265_ns)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="device_type"           value="$(arg device_type_t265)"/>
      <arg name="serial_no"             value="$(arg serial_no_t265)"/>
      <arg name="tf_prefix"         	  value="$(arg tf_prefix_t265)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
      <arg name="enable_fisheye1"       value="$(arg enable_fisheye)"/>
      <arg name="enable_fisheye2"       value="$(arg enable_fisheye)"/>
      <arg name="topic_odom_in"         value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"       value="$(arg calib_odom_file)"/>
      <arg name="gyro_fps"            value="-1"/>
      <arg name="accel_fps"           value="-1"/>
      <arg name="enable_gyro"         value="true"/>
      <arg name="enable_accel"        value="true"/>
      <arg name="enable_pose"         value="true"/>
      <arg name="enable_sync"           value="true"/>

      <!-- <arg name="ordered_pc"            value="$(arg ordered_pc)"/> -->
    </include>
  </group>
  	      <!-- <arg name="enable_infra"          default="true"/> -->
  
  <!-- Launch L515 through nodeler -->
  <group ns="$(arg l515_ns)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="device_type"           value="$(arg device_type_l515)"/>
      <arg name="serial_no"             value="$(arg serial_no_l515)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_l515)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
      <arg name="align_depth"           value="true"/>
      <arg name="filters"               value="pointcloud"/>
      <arg name="color_width"           value="$(arg color_width)"/>
      <arg name="color_height"          value="$(arg color_height)"/>
      <arg name="depth_width"           value="$(arg depth_width)"/>
      <arg name="depth_height"          value="$(arg depth_height)"/>
      <arg name="clip_distance"         value="$(arg clip_distance)"/>
      <arg name="ordered_pc"            value="$(arg ordered_pc)"/>
      <arg name="enable_sync"           value="true"/>
    </include>
  </group>

  <!-- CAMERA ALIGNMENT -->
  <node pkg="tf" type="static_transform_publisher" name="t265_to_l515" args="0.00305 -0.0003 -0.04305 0.0 0.0 1.57 /$(arg tf_prefix_t265)_link /$(arg tf_prefix_l515)_link 100"/>
  

</launch>
```

shubhamwagh, 2022-01-14:
Following are the issues I referred to, to solve this problem:
1) https://github.com/IntelRealSense/realsense-ros/issues/2127
2) https://github.com/IntelRealSense/realsense-ros/issues/1607#issuecomment-760128960

Hope this helps someone who wants to work with L515 + T265 sensor configuration using rtabmap_ros

closed 2022-02-03 by matlabbe

sumitsarkar1, 2023-01-04:
@shubhamwagh what was your launching command? I am using the following

`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="-d --Mem/UseOdomGravity true --Optimizer/GravitySigma 0.3 --delete_db_on_start" odom_topic:=/t265/odom/sample frame_id:=t265_link rgbd_sync:=true depth_topic:=/l515/aligned_depth_to_color/image_raw    rgb_topic:=/l515/color/image_raw camera_info_topic:=/l515/color/camera_info    approx_rgbd_sync:=true visual_odometry:=false queue_size:=200`