introlab/rtabmap_ros #571: Using URDF model in RTABMap

MrOCW, 2021-04-14:
Hi @matlabbe, I created a .urdf file for my robot car. How do I use it in rtabmap? Are there specific TF frames to add as I only have base_link, camera_link, and 4 wheel links. Read from the other forum that there should be a /map and /odom? Do I need to create these in the URDF file as well?

MrOCW, 2021-04-14:
Apologies I have figured it out. Just had to change camera_link to base_link. Thanks!

closed 2021-04-14 by MrOCW

matlabbe, 2021-05-04:
`rtabmap.launch` uses `frame_id:=camera_link` by default (mostly for the hand-held camera mapping examples which all have camera_link as base frame), but `frame_id` should really be set to base frame of your robot, in your case `base_link`.