introlab/rtabmap_ros #572: After mapping for awhile, process has died

MrOCW, 2021-04-14:
Hi @matlabbe , after using ORB2-RTAB, my RViz freezes and I get this message on my terminal

[rtabmap/rgbd_odometry-1] process has died [pid 29136, exit code -11, cmd /home/jetsonxaviernx/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry --GFTT/MinDistance 10 --Optimizer/GravitySigma 0.0 --Odom/Strategy 5 --OdomORBSLAM/VocPath /home/jetson/catkin_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt --OdomORBSLAM/Bf 46.356317139 --ORB/Gpu true --Reg/Force3DoF true --RGBD/NeighborLinkRefining true --Rtabmap/DetectionRate 3 --Rtabmap/LoopThr 0.1 --Odom/ResetCountdown 5 rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay odom:=odom imu:=/rtabmap/imu __name:=rgbd_odometry __log:=/home/jetson/.ros/log/d37b4292-9d01-11eb-be01-70665563d293/rtabmap-rgbd_odometry-1.log].
log file: /home/jetson/.ros/log/d37b4292-9d01-11eb-be01-70665563d293/rtabmap-rgbd_odometry-1*.log

May I know what is happening here?

MrOCW, 2021-04-15:
After disabling Odom/ResetCountdown, everything works fine now. Likely reason is due to ORB-SLAM ORBvoc.txt taking awhile to read and causes the 'process has died' issue.

closed 2021-04-15 by MrOCW

matlabbe, 2021-05-04:
Ok good to know! It should not die though, I'll take a look.

wz8530, 2023-04-19:
@MrOCW I have the same problem, how to solve it?
The error that occurs is as follows:
/rtabmap/rgbd_odometry subscribed to (exact sync):
   /camera/color/image_raw \
   /camera/depth/image_rect_raw \
   /camera/color/camera_info
[ WARN] (2023-04-19 16:03:21.414) Odometry.cpp:317::process() Updated initial pose from xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000 to xyz=0.000000,0.000000,0.000000 rpy=0.005809,-0.293085,-2.636529 with IMU orientation
[ WARN] (2023-04-19 16:03:21.414) Transform.cpp:551::getTransform() No transform found for stamp 1681891401.373435! Earliest is 1681891401.385509
[ WARN] [1681891401.832057313]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/depth/image_rect_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info
[rtabmap/rgbd_odometry-1] process has died [pid 1733, exit code -11, cmd /home/wz/project/ws_rtabmap-0.21.0/devel/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start --Odom/Strategy 5 --OdomORBSLAM/VocPath /home/wz/project/ORB_SLAM3_v0.4/Vocabulary/ORBvoc.txt rgb/image:=/camera/color/image_raw depth/image:=/camera/depth/image_rect_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay odom:=odom imu:=/rtabmap/imu __name:=rgbd_odometry __log:=/home/wz/.ros/log/9757f6ac-de86-11ed-a470-70b5e871fa63/rtabmap-rgbd_odometry-1.log].
log file: /home/wz/.ros/log/9757f6ac-de86-11ed-a470-70b5e871fa63/rtabmap-rgbd_odometry-1*.log

The /rtabmap/odom topic does not have any data, is the error related to this?

matlabbe, 2023-04-20:
See A from https://github.com/introlab/rtabmap_ros/issues/28 to get more info. Note that ORBSLAM integration is not actively supported, so refer to [this](https://github.com/introlab/rtabmap/blob/f8f6b7788a835021abb487d0a8088b87d2c0cb52/docker/jfr2018/latest/Dockerfile#L156-L158) for version that I know worked. For ORB_SLAM3, you may look at this [post](https://github.com/introlab/rtabmap/issues/655#issuecomment-851037772).

wz8530, 2023-04-25:
> See A from #28 to get more info. Note that ORBSLAM integration is not actively supported, so refer to [...]

Hi @matlabbe , I have tried rtabmap+ORB2/ORB3_v0.4 and found this issue when running in rgbd mode. But when I run in Stereo mode, it works normally.
Then, I enabled gdb debugging for this issue, as shown in the following figure:
[image]

According to the gdb prompt, the problem occurs in the StereoCameraModel::baseline() function or opencv_core.so.4.5 library, according to the [blog](https://blog.csdn.net/a395381306/article/details/101521675) prompt, I modified the StereoCameraModel::baseline() function and after recompiling, it can run in the rgbd mode of rtabmap+ORB2/3 normally.

Modify as follows:
Change `double baseline() const {return right_.fx()!=0.0 && left_.fx() != 0.0 ? left_.Tx() / left_.fx() - right_.Tx()/right_.fx():0.0;}` to `double baseline() const {return 0.05;}`

I read the parameters `right_.Tx()=-19.265989303588867`, `right_.fx()=385.83758544921875`, `left_.Tx()=0` and `left_.fx()=385.83758544921875` from the d435i camera P matrix, and found that the formula calculation result is also 0.0449. Why are the results different?

matlabbe, 2023-05-06:
The gdb trace shows that opencv is asserting on an empty matrix. I see also that you are using a local opencv version. Did you rebuild cv_bridge with that version? otherwise this can create opencv issues like this when the library is loading the wrong opencv library on runtime or that rtabmap_ros is linking to more than one OpenCV version (because of cv_bridge). If you really need OpenCV 4.5 (and the not the binaries used by cv_bridge), uninstall opencv binaries from your computer, and rebuild all ros packages you need depending on it.

wz8530, 2023-05-08:
> The gdb trace shows that opencv is asserting on an empty matrix. I see also that you are using a local opencv version. Did you rebuild cv_bridge with that version? otherwise this can create opencv [...]

@matlabbe , I need OpenCV >=4.4.0（GFTT+BRIEF），and I rebuild all ros packages that I need depending on it. As shown below, i switched to opencv4.5.3 and `catkin_make`（the cv_bridge and image_geometry compiled using opencv4.5.3 has been installed in the system）:
[image]

I can find that all the libraries linked to opencv3.2.0 have been changed to opencv4.5.3, so in my opinion, it may not be a problem with the library. Additionally, I will recompile the entire ROS using opencv4.5.3 to test it.