introlab/rtabmap_ros #577: Mapping error on remote computer

locvu98, 2021-05-05:
Hello, I'm trying to use Realsense D435 camera and Raspberry Pi 4 for handheld mapping as suggested in this guide: http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping
Here's my setup:
- [Laptop] Add these commands to bashrc:
𝗲𝘅𝗽𝗼𝗿𝘁 𝗥𝗢𝗦_𝗛𝗢𝗦𝗧𝗡𝗔𝗠𝗘=𝟭𝟵𝟮.𝟭𝟲𝟴.𝟭.𝟳
𝗲𝘅𝗽𝗼𝗿𝘁 𝗥𝗢𝗦_𝗠𝗔𝗦𝗧𝗘𝗥_𝗨𝗥𝗜=𝗵𝘁𝘁𝗽://${𝗥𝗢𝗦_𝗛𝗢𝗦𝗧𝗡𝗔𝗠𝗘}:𝟭𝟭𝟯𝟭𝟭
- [Raspberry Pi 4B] Add these commands to bashrc:
𝗲𝘅𝗽𝗼𝗿𝘁 𝗥𝗢𝗦_𝗛𝗢𝗦𝗧𝗡𝗔𝗠𝗘=𝟭𝟵𝟮.𝟭𝟲𝟴.𝟭.𝟭𝟮
𝗲𝘅𝗽𝗼𝗿𝘁 𝗥𝗢𝗦_𝗠𝗔𝗦𝗧𝗘𝗥_𝗨𝗥𝗜=𝗵𝘁𝘁𝗽://𝟭𝟵𝟮.𝟭𝟲𝟴.𝟭.𝟳:𝟭𝟭𝟯𝟭𝟭
- [Laptop] Run 𝗿𝗼𝘀𝗰𝗼𝗿𝗲
- [Raspberry Pi 4B] Run this command
𝗿𝗼𝘀𝗹𝗮𝘂𝗻𝗰𝗵 𝗿𝗲𝗮𝗹𝘀𝗲𝗻𝘀𝗲𝟮_𝗰𝗮𝗺𝗲𝗿𝗮 𝗿𝘀_𝗰𝗮𝗺𝗲𝗿𝗮.𝗹𝗮𝘂𝗻𝗰𝗵 𝗮𝗹𝗶𝗴𝗻_𝗱𝗲𝗽𝘁𝗵:=𝘁𝗿𝘂𝗲
- [Laptop] Run this command 
𝗿𝗼𝘀𝗹𝗮𝘂𝗻𝗰𝗵 𝗿𝘁𝗮𝗯𝗺𝗮𝗽_𝗿𝗼𝘀 𝗿𝘁𝗮𝗯𝗺𝗮𝗽.𝗹𝗮𝘂𝗻𝗰𝗵 \
    𝗿𝘁𝗮𝗯𝗺𝗮𝗽_𝗮𝗿𝗴𝘀:="--𝗱𝗲𝗹𝗲𝘁𝗲_𝗱𝗯_𝗼𝗻_𝘀𝘁𝗮𝗿𝘁" \
    𝗱𝗲𝗽𝘁𝗵_𝘁𝗼𝗽𝗶𝗰:=/𝗰𝗮𝗺𝗲𝗿𝗮/𝗮𝗹𝗶𝗴𝗻𝗲𝗱_𝗱𝗲𝗽𝘁𝗵_𝘁𝗼_𝗰𝗼𝗹𝗼𝗿/𝗶𝗺𝗮𝗴𝗲_𝗿𝗮𝘄 \
    𝗿𝗴𝗯_𝘁𝗼𝗽𝗶𝗰:=/𝗰𝗮𝗺𝗲𝗿𝗮/𝗰𝗼𝗹𝗼𝗿/𝗶𝗺𝗮𝗴𝗲_𝗿𝗮𝘄 \
    𝗰𝗮𝗺𝗲𝗿𝗮_𝗶𝗻𝗳𝗼_𝘁𝗼𝗽𝗶𝗰:=/𝗰𝗮𝗺𝗲𝗿𝗮/𝗰𝗼𝗹𝗼𝗿/𝗰𝗮𝗺𝗲𝗿𝗮_𝗶𝗻𝗳𝗼 \
    𝗮𝗽𝗽𝗿𝗼𝘅_𝘀𝘆𝗻𝗰:=𝗳𝗮𝗹𝘀𝗲
However, I have encountered this issue:
[image]
When I ran 𝗿𝗼𝘀𝘁𝗼𝗽𝗶𝗰 𝗵𝘇 /𝗰𝗮𝗺𝗲𝗿𝗮/𝗰𝗼𝗹𝗼𝗿/𝗶𝗺𝗮𝗴𝗲_𝗿𝗮𝘄 /𝗰𝗮𝗺𝗲𝗿𝗮/𝗰𝗼𝗹𝗼𝗿/𝗰𝗮𝗺𝗲𝗿𝗮_𝗶𝗻𝗳𝗼 on my laptop's terminal and raspberry pi's terminal, it seemed like these topics didn't have the same rate.
[image]
How can I fix this problem? Thank you very much.

matlabbe, 2021-05-05:
Hi, you may take a look at this page: http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping

Check your network bandwidth, subscribing to raw images from remote could require at 30 Hz 720p : 79 MB per sec. At least compare with `/camera/color/image_raw/compressed`.

locvu98, 2021-05-10:
Thank you for replying!
I followed your guide http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping
However, on the remote computer, topic [/camera/data_throttled_image/compressed] does not appear to be published.
How can I fix this issue?

locvu98, 2021-05-17:
sorry for late replying, but rostopic hz /camera/data_throttled_image/compressed is not published on the local computer.

matlabbe, 2021-05-17:
You may have to debug why `/camera/data_throttled_image/compressed` is not published. Is `/camera/data_throttled_image` published? Look at the node publishing it (with `rosnode info`), if its inputs are also published.