introlab/rtabmap_ros #578: The SLAM result is an upward slope when on flat ground in real

Mingchi-Yeh, 2021-05-07:
My launch file about rtabmap_ros mapping as following .
`    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
	<arg name="args" default="--delete_db_on_start --Vis/MinInliers 15 --Mem/InitWMWithAllnodes true --Marker/Length 0.07 --RGBD/MarkerDetection true --Marker/Dictionary 20 --Optimizer/Strategy 1 --Marker/MaxDepthError 0.1"/>
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="true"/>
        <arg name="rviz" value="false"/>
	</include>`

And my launch file about rtabmap_ros localization as following .
`    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
	<arg name="args" default="--Vis/MinInliers 15 --Mem/IncrementalMemory false --Mem/InitWMWithAllNodes true --Marker/Length 0.07 --RGBD/MarkerDetection true --Marker/Dictionary 20 --Optimizer/Strategy 1 --Marker/MaxDepthError 0.1"/>
	<arg name="localization" value="true"/>
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="true"/>
        <arg name="rviz" value="false"/>
	</include>`
This is my slam result, when I localize the robot on flat ground, the slam result is an upward slope.
[image]
[image]

Can I get any support about this ?

matlabbe, 2021-05-07:
It looks like the camera is tilted in reality but not in TF tree. Make sure your camera is horizontal to ground or adjust TF to tell rtabmap that it is tilted.

Mingchi-Yeh, 2021-05-08:
In fact, this picture is actually a slam result with the camera not tilted.
When the camera is tilted downward, the angle of the slam tilt will be bigger.
So this result may be the error of rtabmap?
Is the adjust TF mean directly change the results such as /rtabmap/odom?

matlabbe, 2021-05-08:
can you share the database? If it was drift, I would have expected that z error increases over time.

Mingchi-Yeh, 2021-05-10:
This is my slam result bag file,but I don't think this is the drift error.
[bag_file.zip](https://github.com/introlab/rtabmap_ros/files/6450598/bag_file.zip)

My current solution is to rotate the coordinate axis directly, but if it’s an online slam, it’s impossible to know exactly how many degrees it has shifted upward.

Mingchi-Yeh, 2021-05-12:
https://drive.google.com/file/d/19Vs0c_P9sIo66pqMYoj6FnHIPW01puUe/view?usp=sharing
This is my Dataset, include /tf , /tf_static , and depth topic.

matlabbe, 2021-05-12:
Hi, the camera is definitely tilting towards the ground:
[image]

You could add a static transform like this:
`rosrun tf static_transform_publisher 0 0 0.6 0 0.48 0 base_link camera_link 100`

Then set rtabmap to use frame_id:=base_link

[image]

Mingchi-Yeh, 2021-05-13:
That looks fine !
Thanks for your help !