introlab/rtabmap_ros #582: the obstacle on the ground cannot be captured in grid map in demo_turtlebot_mapping demo

TouchDeeper, 2021-05-14:
Hello Mathieu,

I test the `demo_turtlebot_mapping` [demo](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot) in gazebo. 

I found that an obstacle on the ground cannot be captured in the grid map, as the image below shows:
[image]

[image]

I tuned the `scan_height` of `depthimage_to_laserscan`, but it has no effect.

Could you please provide some advice? Thanks a lot.

matlabbe, 2021-05-15:
To detect all objects over the ground in field of view of the camera, I would use `Grid/FromDepth=true`. For flat obstacles like that, the default ground plane segmentation may label it as ground. You could use `Grid/NormalsSegmentation=false` `Grid/MaxGoundHeight=0.05` if so.

TouchDeeper, 2021-05-16:
@matlabbe 
Thanks for your reply.

I add these parameters to `rtabmap_ros`:
```
  <param name="Grid/3D" value="false"/>
  <param name="Grid/NormalsSegmentation" value="false"/>
  <param name="Grid/MaxGroundHeight" value="0.02"/>\
  <param name="Grid/RayTracing" value="true"/>
```

But it make no effect.

Is it relevant to the `depthimage _to_laserscan`?

matlabbe, 2021-05-16:
Add also `<param name="Grid/FromDepth" value="true"/>` to `rtabmap` node.

TouchDeeper, 2021-05-17:
@matlabbe 
Thanks for your reply. it works.

closed 2021-05-17 by TouchDeeper