introlab/rtabmap_ros #584: use camera that depth is not aligned to color in demo_turtlebot_mapping example in gazebo

TouchDeeper, 2021-05-19:
Hello Mathieu,

I want to use realsense d435 in demo_turtlebot_mapping [example](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot) in gazebo. The realsense gazebo plugin I used is [here](https://github.com/pal-robotics/realsense_gazebo_plugin). 

I find that the simulation realsense plugin cannot align the depth to color or inverse. If I run the demo_turtlebot_mapping demo, error occur:
```
MsgConversion.cpp:1718::convertRGBDMsgs() Condition (imageWidth % depthWidth == 0 && imageHeight % depthHeight == 0 && imageWidth/depthWidth == imageHeight/depthHeight) not met! [rgb=1920x1080 depth=1280x720]
```

I think it is because of the depth without alignment.  But I do not turn on `rgbd_odometry`, I just need the pointcloud to navigation, so it should work using the depth without alignment. Could you please provide some advice to avoid the need for the aligned depth? Thanks a lot.

matlabbe, 2021-05-24:
Actually, I think that assert (from 4 years ago) could be now removed as the depth doesn't require anymore to be a multiple of rgb image (see this [commit](https://github.com/introlab/rtabmap/commit/012439aa0b3fdb3067023b33fa30ff29ed8895fe#diff-662ff84314b7cc07a291fae383f61ab5a69ecd654486376104c5a14cf66424caL4008-R4008) from rtabmap library).

TouchDeeper, 2021-05-24:
@matlabbe Thanks for your reply.

This commit has been accepted in my local repository. And I find the error is caused by this [assert](https://github.com/introlab/rtabmap_ros/blob/761d8d61292c5b219d53276336426527b25bd7f4/src/MsgConversion.cpp#L371) or this [assert](https://github.com/introlab/rtabmap_ros/blob/761d8d61292c5b219d53276336426527b25bd7f4/src/MsgConversion.cpp#L1733).

Should these two assertions be commented out?

matlabbe, 2021-05-24:
I removed the checks in the commit above. However, rtabmap still expect registered depth image to rgb camera. If you feed unaligned depth with RGB, the generated occupancy grid may be okay and point cloud color would be off, but rtabmap would generate wrong loop closure transforms.

TouchDeeper, 2021-05-25:
@matlabbe Thank you for your reply.

`rtabmap would generate wrong loop closure transforms` 
I think so.

Maybe we should try to provide aligned depth.

closed 2021-05-25 by TouchDeeper

matlabbe, 2021-05-30:
To align depth with rtabmap node, we have to set `subscribe_depth:=false`, set `subscribe_scan_cloud:=true` and feed the point cloud topic of the camera, then use `gen_depth:=true`, with possibility `gen_depth_decimation:=4` or more to have a dense depth image. However, this kind of transformation would be already be done when aligning depth from the camera driver directly.