introlab/rtabmap_ros #585: An error occurring after "catkin_make -j1": function creatCloud

Wangxz195, 2021-05-20:
[ 59%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o
/home/wang/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp: In member function ‘void MapsManager::publishMaps(const std::map<int, rtabmap::Transform>&, const ros::Time&, const string&)’:
/home/wang/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:1244:214: error: no matching function for call to ‘rtabmap::OctoMap::createCloud(int&, boost::shared_ptr<std::vector<int> >::element_type*, boost::shared_ptr<std::vector<int> >::element_type*, boost::shared_ptr<std::vector<int> >::element_type*, bool, boost::shared_ptr<std::vector<int> >::element_type*, int, bool&)’
 1244 |  groundIndices.get(), true, frontierIndices.get(),0,octomap_frontier_flood_fill_);
      |                                                                                 ^

In file included from /home/wang/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:53:
/opt/ros/noetic/lib/x86_64-linux-gnu/rtabmap-0.20/../../../include/rtabmap-0.20/rtabmap/core/OctoMap.h:189:41: note: candidate: ‘pcl::PointCloud<pcl::PointXYZRGB>::Ptr rtabmap::OctoMap::createCloud(unsigned int, std::vector<int>*, std::vector<int>*, std::vector<int>*, bool, std::vector<int>*, std::vector<double, std::allocator<double> >*) const’
  189 |  pcl::PointCloud<pcl::PointXYZRGB>::Ptr createCloud(
      |                                         ^~~~~~~~~~~
/opt/ros/noetic/lib/x86_64-linux-gnu/rtabmap-0.20/../../../include/rtabmap-0.20/rtabmap/core/OctoMap.h:189:41: note:   candidate expects 7 arguments, 8 provided
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/build.make:76: rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1104: rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j 1" failed

Above is my error message, it seems the function "createCloud" does not exist. I followed the installation instrcution step by step, not installing any optional dependencies. Thank you very much for anyone can help me solve my problem.

Wangxz195, 2021-05-20:
I have checked where the function "createCloud" is called: /home/wang/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp, line 1244, there are 8 arguments passed to the function, which should be right.
I have also checked where this function is defined: /usr/local/include/rtabmap-0.20/rtabmap/core/OctoMap.h, line 189. There also should be 8 arguments. 
However, when I run "catkin_make -j1", it seems that only 7 arguments in the definition are detected.

matlabbe, 2021-05-20:
Catkin_make is using this one:
`/opt/ros/noetic/lib/x86_64-linux-gnu/rtabmap-0.20/../../../include/rtabmap-0.20/rtabmap/core/OctoMap.h`

Uninstall rtabmap b8naries to make sure catkin_make is using the version built from source:
```
sudo apt remove ros-noetic-rtabmap
```

Wangxz195, 2021-05-20:
@matlabbe Thank you very much, I followed your advice and solved the problem. Although there was a new problem occurring, I followed the solution in issue#91 and also solved it.

closed 2021-05-20 by Wangxz195