introlab/rtabmap_ros #587: Include Issue after successfully compiling

Wangxz195, 2021-05-22:
[image]
[image]
[image]
[image]

I followed the steps to install RTAB-Map standalone libraries and to install the RTAB-Map_ros package. Every step was successfullly done. However, when I use VSCode open the *.cpp files, there are still include errors detected. Could anyone please help me solve the problem? Thank you very much in advanced.

matlabbe, 2021-05-22:
It is an error of vscode (indexing), not the compiler. You may have to tell vscode where are the includes. You may check the cmake plugin that can be installed in vscode, then use `cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON` ([ref](https://stackoverflow.com/questions/36122358/visual-studio-code-c-include-path))

Wangxz195, 2021-05-22:
@matlabbe Thank you very much for your hint. I have read the reference, and I have tried the solution mentioned while failing.  I am sorry for asking a stupid question: Where should I add this syntax? (Since I have no idea how is your code working and I am new to ubuntu&vscode...) Thank you again if you could help me with more details.

Best regards

matlabbe, 2021-05-23:
I don't use VSCode often, but based on this page: https://code.visualstudio.com/docs/cpp/cmake-linux
You can download "CMake Tools" extension (click on the gear icon at the bottom left, then click on Extensions, or jsut do "ctrl-shift-x", then search for CMake Tools and install it).

If you do File->Open Folder, select rtabmap folder. At the bottom of the window (blue toolbar), you can execute CMake on Release mode, then do Build afterwards. On my computer, the includes were correctly found and autocompletion working already.
[image]

For rtabmap_ros, you may download this extension: https://marketplace.visualstudio.com/items?itemName=ms-iot.vscode-ros, then open your `catkin_ws` folder in VSCode (as above, the includes and autocompletion should work):
[image]

Wangxz195, 2021-05-24:
@matlabbe Thank you very much, your hints are really helpful. And I have solved my problem by following your hints.
Best reagrds.

closed 2021-05-24 by Wangxz195