introlab/rtabmap_ros #590: low performance with orbslam3 backend

FPSychotic, 2021-05-27:
Hi I found that using orbslam3 with rtabmap ros (Strategy 5) detects a lot less features and tracking lost than orb3 standalone or orbslam2.
i think it is due to this values that you can find in orbslam:

```
#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500
```

The valors in rtabmap are, and there is two files with them (/documents/rtabmap and .ros/rtabmap that looks hardcoded because if I try change the values they again are overwritten:

```
Camera.fps: 0.0000000000000

ThDepth: 40.0000000000000

DepthMapFactor: 1000.0000000000000

ORBextractor.nFeatures: 1200

ORBextractor.scaleFactor: 2.0000000000000

ORBextractor.nLevels: 3

ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7
```

How I can put my own values? they are not available in rtabmap params
       ```
   <param name="Odom/Strategy" value="5"/> 
          <param name="OdomORBSLAM/VocPath" value="/home/iwire/ORB_SLAM3/Vocabulary/ORBvoc.txt"/> 
          <param name="OdomORBSLAM/Fps" value="60.0"/> 
          <param name="OdomORBSLAM/MapSize" value="3000"/> 
          <param name="OdomORBSLAM/MaxFeatures" value="1200"/> 
          <param name="OdomORBSLAM/ThDepth" value="45.0"/> 
```
thanks!!

matlabbe, 2021-05-30:
Hi, those values are not hard-coded. rtabmap generates an orbslam2 config file based from the actual camera_info received and RTAB-Map's parameters below. For ORBSLAM2 specific parameters, use:
```
rtabmap --params | grep ORBSLAM
Param: OdomORBSLAM/Bf = "0.076"                            [Fake IR projector baseline (m) used only when stereo is not used.]
Param: OdomORBSLAM/Fps = "0.0"                             [Camera FPS.]
Param: OdomORBSLAM/MapSize = "3000"                        [Maximum size of the feature map (0 means infinite).]
Param: OdomORBSLAM/MaxFeatures = "1000"                    [Maximum ORB features extracted per frame.]
Param: OdomORBSLAM/ThDepth = "40.0"                        [Close/Far threshold. Baseline times.]
Param: OdomORBSLAM/VocPath = ""                            [Path to ORB vocabulary (*.txt).]
```

To tune ORB related parameters:
```
rtabmap --params | grep ORB/
Param: ORB/EdgeThreshold = "19"                            [This is size of the border where the features are not detected. It should roughly match the patchSize parameter.]
Param: ORB/FirstLevel = "0"                                [It should be 0 in the current implementation.]
Param: ORB/Gpu = "false"                                   [GPU-ORB: Use GPU version of ORB. This option is enabled only if OpenCV is built with CUDA and GPUs are detected.]
Param: ORB/NLevels = "3"                                   [The number of pyramid levels. The smallest level will have linear size equal to input_image_linear_size/pow(scaleFactor, nlevels).]
Param: ORB/PatchSize = "31"                                [size of the patch used by the oriented BRIEF descriptor. Of course, on smaller pyramid layers the perceived image area covered by a feature will be larger.]
Param: ORB/ScaleFactor = "2"                               [Pyramid decimation ratio, greater than 1. scaleFactor==2 means the classical pyramid, where each next level has 4x less pixels than the previous, but such a big scale factor will degrade feature matching scores dramatically. On the other hand, too close to 1 scale factor will mean that to cover certain scale range you will need more pyramid levels and so the speed will suffer.]
Param: ORB/ScoreType = "0"                                 [The default HARRIS_SCORE=0 means that Harris algorithm is used to rank features (the score is written to KeyPoint::score and is used to retain best nfeatures features); FAST_SCORE=1 is alternative value of the parameter that produces slightly less stable keypoints, but it is a little faster to compute.]
Param: ORB/WTA_K = "2"                                     [The number of points that produce each element of the oriented BRIEF descriptor. The default value 2 means the BRIEF where we take a random point pair and compare their brightnesses, so we get 0/1 response. Other possible values are 3 and 4. For example, 3 means that we take 3 random points (of course, those point coordinates are random, but they are generated from the pre-defined seed, so each element of BRIEF descriptor is computed deterministically from the pixel rectangle), find point of maximum brightness and output index of the winner (0, 1 or 2). Such output will occupy 2 bits, and therefore it will need a special variant of Hamming distance, denoted as NORM_HAMMING2 (2 bits per bin). When WTA_K=4, we take 4 random points to compute each bin (that will also occupy 2 bits with possible values 0, 1, 2 or 3).]
```
Note that `ORBextractor.nFeatures` is mapped to `Vis/MaxFeatures` (default 1000) rtabmap parameter.

For FAST:
```
rtabmap --params | grep FAST/
Param: FAST/CV = "0"                                       [Enable FastCV implementation if non-zero (and RTAB-Map is built with FastCV support). Values should be 9 and 10.]
Param: FAST/Gpu = "false"                                  [GPU-FAST: Use GPU version of FAST. This option is enabled only if OpenCV is built with CUDA and GPUs are detected.]
Param: FAST/GpuKeypointsRatio = "0.05"                     [Used with FAST GPU.]
Param: FAST/GridCols = "0"                                 [Grid cols (0 to disable). Adapts the detector to partition the source image into a grid and detect points in each cell.]
Param: FAST/GridRows = "0"                                 [Grid rows (0 to disable). Adapts the detector to partition the source image into a grid and detect points in each cell.]
Param: FAST/MaxThreshold = "200"                           [Maximum threshold. Used only when FAST/GridRows and FAST/GridCols are set.]
Param: FAST/MinThreshold = "7"                             [Minimum threshold. Used only when FAST/GridRows and FAST/GridCols are set.]
Param: FAST/NonmaxSuppression = "true"                     [If true, non-maximum suppression is applied to detected corners (keypoints).]
Param: FAST/Threshold = "20"                               [Threshold on difference between intensity of the central pixel and pixels of a circle around this pixel.]
```
Note that `ORBextractor.iniThFAST` is mapped to `FAST/Threshold` rtabmap parameter..

Note also that I've found a bug on ORB_SLAM3 side for RGB-D mode (Stereo mode should be already fine). I've made a patch for ORBSLAM3 v4 integration with rtabmap: https://gist.github.com/matlabbe/f5cb281304a1305b2824a6ce19792e13

FPSychotic, 2021-05-30:
thank you a lot by your answer, I get lost very easy by lack of knowledge, I couldn't find that params started by ORB and they are clearly what  was looking for, but I just did grep to orbslam params.

Respect to the patch, yes I'm using RGBD in V4 but I never applied a patch, I followed a small tutorial, and the patch fail to apply in my installed orblam3 v4 but also with the Orbslam3 v4 master just downloaded. I guess the patch is applied in the main folder of orbslam3 where the cmakelists.txt is located. 

I'm sorry need ask you how to apply or where, I cannot understand the content to apply the changes manually.

thanks,, and sorry by take your time

matlabbe, 2021-05-30:
I added a usage example here: https://gist.github.com/matlabbe/f5cb281304a1305b2824a6ce19792e13#gistcomment-3763035

closed 2021-05-30 by FPSychotic