introlab/rtabmap_ros #591: Can the pose estimation at t+1 be calculated from the new result at t time?

Mingchi-Yeh, 2021-05-28:
Hi , the red line in the figure is my slam result , as shown in the figure , it has a serious drift.
[image]

So I want to use the blue line to limit its error range.
If the new slam result at t time is (x1,y1,z1), can I use the (x1,y1,z1) to estimate the slam pose (x2,y2,z2) at t+1 time ?
I searched the source code of rtabmap_ros but I don't know where to alter it.

matlabbe, 2021-05-30:
What the blue line represents? If you use rtabmap's odometry, look inside  [OdometryROS.cpp](https://github.com/introlab/rtabmap_ros/blob/master/src/OdometryROS.cpp).

Mingchi-Yeh, 2021-05-31:
The blue line is the other pose with 'geometry_msgs/PoseWithCovarianceStamped'.

Mingchi-Yeh, 2021-05-31:
I look inside the OdometryROS.cpp .

matlabbe, 2021-06-03:
Note sure to understand what represents the blue line: "The blue line is the other pose". Which other pose? how is it computed?

matlabbe, 2021-07-14:
Is the blue line in a global coordinate frame (not relative like odometry)? There a some options:
1) use the blue line as input odometry to rtabmap node.
2) use the blue line as guess odometry for odometry node (using `guess_frame_id` parameter).
3) do odometry fusion between blue and red lines with [robot_localization](http://docs.ros.org/en/noetic/api/robot_localization/html/index.html) package
4) use `global_pose` input topic of rtabmap node to feed your blue lines. Set `Optimizer/PriorsIgnored` to false, then rtabmap will use the blue line to optimize red line based on the blue line (used as priors).