introlab/rtabmap_ros #592: Feature Request

dineshrboson, 2021-06-01:
Can you please add demo_catvehicle_navigation.launch as well

matlabbe, 2021-06-01:
Hi,
Can you point out the repo, is it this one? https://github.com/jmscslgroup/catvehicle

dineshrboson, 2021-06-01:
Yes

dineshrboson, 2021-06-01:
Is it possible to do same as demo_turtlebot3_navigation.launch? For cat vehicle simulation as well

matlabbe, 2021-06-28:
I tried the repo, cannot get the sensor data working. I am on Noetic, I tried their [noetic_gazebo-11](https://github.com/jmscslgroup/catvehicle/tree/noetic_gazebo-11) branch. IF you had any success to see velodyne and camera data in RVIZ, show me what commandlines you used to start the simulator.

This is what I tried (I can see tf and control with joystick, but no sensor):
```bash
roslaunch catvehicle catvehicle_city.launch gui:=true
roslaunch catvehicle catvehicle_spawn.launch
roslaunch catvehicle joystick.launch
```
[image]

dineshrboson, 2021-06-29:
https://github.com/jmscslgroup/catvehicle/issues/1
[image]

dineshrboson, 2021-06-29:
I added a link on top of the image pls go through it. That has the information on how to retrieve sensor data

dineshrboson, 2022-01-24:
Hi had you done this?

matlabbe, 2022-01-26:
I think I moved to other issues and forgotten that issue, thx for the reminder. The lidar cloud you shown was not very useful for RTAB-Map, we have to modify the velodyne and triclops gazebo config. In commit above, I added a mapping example with instructions on how to modify the catvehicle urdf:
https://github.com/introlab/rtabmap_ros/blob/293cfcf6540a816361682308a917979e3d0387e5/launch/demo/demo_catvehicle_mapping.launch#L3-L29
For navigation, I didn't see if there is already something in catvehicle project. Integrating move_base could be done with ted_local_planner, which can handle local planning with ackermann geometry. I need to work on another tasks, but I may come back later for move_base example if no one else tried it.