introlab/rtabmap_ros #593: Error building on Jetson TX2

sriharis123, 2021-06-07:
Hello,
I'm trying to get this package running on the TX2  with an Intel Realsense d435i and have been running into some issues. First, I tried `apt install ros-melodic-rtabmap-ros` and followed [this tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping). Running the rtabmapviz command gave me the following error:

```
[rtabmap/rtabmapviz-3] process has died [pid 4164, exit code 127, cmd /opt/ros/melodic/lib/rtabmap_ros/rtabmapviz -d ~/.ros/rtabmap_gui.ini rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud scan_descriptor:=/scan_descriptor odom:=odom __name:=rtabmapviz __log:=/home/nvidia/.ros/log/e8cd147e-c531-11eb-88ef-00044bc6f69a/rtabmap-rtabmapviz-3.log].
log file: /home/nvidia/.ros/log/e8cd147e-c531-11eb-88ef-00044bc6f69a/rtabmap-rtabmapviz-3*.log
```

I was able to pull up the pointcloud in RViz, but the visualization wasn't as good as the rtabmapviz (I've previously run this on an Intel NUC, which worked fine). Next, I tried installing from source using the guide at the bottom of the readme. During the make, I got this error:

```
/usr/bin/ld: cannot find -lQt4::Widgets
collect2: error: ld returned 1 exit status
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:3415: recipe for target '../bin/librtabmap_core.so.0.20.11' failed
make[2]: *** [../bin/librtabmap_core.so.0.20.11] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

I also tried following the steps in [this post](https://github.com/introlab/rtabmap/issues/427#issuecomment-608052821), but I got `cc: error: ARGS: No such file or directory`.

Any help would be appreciated. Thank you!

matlabbe, 2021-07-13:
What is the jetpack version you are using? Can you try with the same version than in the post you linked? 

For jetpack 4.3:
> ROS
> The ros binaries are still working out of the box with the original issue (rtabmapviz blank 3D View). Recommendation is still to use rviz for visualization.

jamesbaker1, 2024-08-14:
Did you figure this out?