introlab/rtabmap_ros #595: Error building rtabmap_ros from source

adbidwai, 2021-06-16:
### Issue
I am using ROS Melodic on Ubuntu 18.04. I am trying to build the rtabmap_ros package from the source but I am receiving the following error. I am not able to understand if the problem is with the OpenCV version or some missing dependency. 
Help will be appreciated.
TIA

### Error Log : 

```
Errors     << rtabmap_ros:make /home/adbidwai/nus_ws/logs/rtabmap_ros/build.make.003.log          
/home/adbidwai/nus_ws/src/rtabmap_ros/src/OdometryROS.cpp: In member function ‘virtual void rtabmap_ros::OdometryROS::onInit()’:
/home/adbidwai/nus_ws/src/rtabmap_ros/src/OdometryROS.cpp:357:34: error: ‘class rtabmap::Odometry’ has no member named ‘canProcessIMU’; did you mean ‘canProcessAsyncIMU’?
  if(waitIMUToinit_ || odometry_->canProcessIMU())
                                  ^~~~~~~~~~~~~
                                  canProcessAsyncIMU
/home/adbidwai/nus_ws/src/rtabmap_ros/src/OdometryROS.cpp: In member function ‘void rtabmap_ros::OdometryROS::callbackIMU(const ImuConstPtr&)’:
/home/adbidwai/nus_ws/src/rtabmap_ros/src/OdometryROS.cpp:426:18: error: ‘class rtabmap::Odometry’ has no member named ‘canProcessIMU’; did you mean ‘canProcessAsyncIMU’?
   if(!odometry_->canProcessIMU() &&
                  ^~~~~~~~~~~~~
                  canProcessAsyncIMU
/home/adbidwai/nus_ws/src/rtabmap_ros/src/OdometryROS.cpp:454:18: error: ‘class rtabmap::Odometry’ has no member named ‘canProcessIMU’; did you mean ‘canProcessAsyncIMU’?
   if(!odometry_->canProcessIMU())
                  ^~~~~~~~~~~~~
                  canProcessAsyncIMU
/home/adbidwai/nus_ws/src/rtabmap_ros/src/OdometryROS.cpp: In member function ‘void rtabmap_ros::OdometryROS::processData(const rtabmap::SensorData&, const ros::Time&, const string&)’:
/home/adbidwai/nus_ws/src/rtabmap_ros/src/OdometryROS.cpp:525:16: error: ‘class rtabmap::Odometry’ has no member named ‘canProcessIMU’; did you mean ‘canProcessAsyncIMU’?
  if(odometry_->canProcessIMU())
                ^~~~~~~~~~~~~
                canProcessAsyncIMU
/home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘bool rtabmap_ros::convertScanMsg(const LaserScan&, const string&, const string&, const ros::Time&, rtabmap::LaserScan&, tf::TransformListener&, double, bool)’:
/home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:2105:74: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘rtabmap::LaserScan’)
   data = rtabmap::util3d::laserScan2dFromPointCloud(*pclScan, laserToOdom); // put back in laser frame
                                                                          ^
In file included from /usr/include/opencv2/core/mat.hpp:3518:0,
                 from /usr/include/opencv2/core.hpp:59,
                 from /usr/include/opencv2/opencv.hpp:52,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:41,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv2/core/mat.inl.hpp:596:6: note: candidate: cv::Mat& cv::Mat::operator=(const cv::Mat&)
 Mat& Mat::operator = (const Mat& m)
      ^~~
/usr/include/opencv2/core/mat.inl.hpp:596:6: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const cv::Mat&’
/usr/include/opencv2/core/mat.inl.hpp:3104:6: note: candidate: cv::Mat& cv::Mat::operator=(const cv::MatExpr&)
 Mat& Mat::operator = (const MatExpr& e)
      ^~~
/usr/include/opencv2/core/mat.inl.hpp:3104:6: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const cv::MatExpr&’
In file included from /usr/include/opencv2/core.hpp:59:0,
                 from /usr/include/opencv2/opencv.hpp:52,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:41,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv2/core/mat.hpp:1154:10: note: candidate: cv::Mat& cv::Mat::operator=(const Scalar&)
     Mat& operator = (const Scalar& s);
          ^~~~~~~~
/usr/include/opencv2/core/mat.hpp:1154:10: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘const Scalar& {aka const cv::Scalar_<double>&}’
In file included from /usr/include/opencv2/core/mat.hpp:3518:0,
                 from /usr/include/opencv2/core.hpp:59,
                 from /usr/include/opencv2/opencv.hpp:52,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:41,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/usr/include/opencv2/core/mat.inl.hpp:1204:6: note: candidate: cv::Mat& cv::Mat::operator=(cv::Mat&&)
 Mat& Mat::operator = (Mat&& m)
      ^~~
/usr/include/opencv2/core/mat.inl.hpp:1204:6: note:   no known conversion for argument 1 from ‘rtabmap::LaserScan’ to ‘cv::Mat&&’
/home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:2113:74: error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘rtabmap::LaserScan’)
   data = rtabmap::util3d::laserScan2dFromPointCloud(*pclScan, laserToOdom); // put back in laser frame
                                                                          ^
In file included from /usr/include/opencv2/core/mat.hpp:3518:0,
                 from /usr/include/opencv2/core.hpp:59,
                 from /usr/include/opencv2/opencv.hpp:52,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:41,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
[... 79 lines cut ...]
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/opt/ros/melodic/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/LaserScan.h:78:21: note: declared here
  RTABMAP_DEPRECATED(LaserScan(const LaserScan & data,
                     ^
/opt/ros/melodic/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/RtabmapExp.h:42:39: note: in definition of macro ‘RTABMAP_DEPRECATED’
 #define RTABMAP_DEPRECATED(func, msg) func __attribute__ ((deprecated(msg)))
                                       ^~~~
/home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:2273:147: warning: ‘rtabmap::LaserScan::LaserScan(const rtabmap::LaserScan&, int, float, rtabmap::LaserScan::Format, const rtabmap::Transform&)’ is deprecated: Use version without "format" argument. [-Wdeprecated-declarations]
    scan = rtabmap::LaserScan(rtabmap::util3d::laserScanFromPointCloud(*pclScan), maxPoints, maxRange, rtabmap::LaserScan::kXYZ, scanLocalTransform);
                                                                                                                                                   ^
In file included from /opt/ros/melodic/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Transform.h:31:0,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:45,
                 from /home/adbidwai/nus_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/opt/ros/melodic/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/LaserScan.h:78:21: note: declared here
  RTABMAP_DEPRECATED(LaserScan(const LaserScan & data,
                     ^
/opt/ros/melodic/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/RtabmapExp.h:42:39: note: in definition of macro ‘RTABMAP_DEPRECATED’
 #define RTABMAP_DEPRECATED(func, msg) func __attribute__ ((deprecated(msg)))
                                       ^~~~
make[2]: *** [CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o] Error 1
make[1]: *** [CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [all] Error 2
```

matlabbe, 2021-07-14:
Hi you are building latest rtabmap_ros version from source while it tries to link on rtabmap melodic binaries. You should build [rtabmap](https://github.com/introlab/rtabmap) from source too. Uninstall rtabmap melodic binaries to make sure rtabmap_ros is using the right rtabmap library that you would have built from source.
```
sudo apt remove ros-melodic-rtabmap
```

Other option is to build same rtabmap_ros version than the rtabmap melodic binaries version you installed. Example:
```bash
cd ~/catkin_ws/src/rtabmap_ros
git checkout melodic-devel
cd ../..
catkin_make
```

cheers,
Mathieu

adbidwai, 2021-07-15:
@matlabbe Thanks a lot for the solution. It works
I actually did it the other way round. I kept ROS Melodic's binaries, checked out to `melodic-devel` branch of `rtabmap_ros`, and then built it from the source. Then finally everything worked.

closed 2021-07-21 by adbidwai