introlab/rtabmap_ros #596: rtabmap_ros: catkin_make error: undefined reference to rtabmap::OctoMap::createCloud(...)

damianosa96, 2021-06-22:
Hi, 
I have a problem when building rtabmap_ros from source. After all the previous steps indicated in the github guide, when I launch the command catkin_make I receive the following error: 

Command:
`catkin_make -DRTABMAP_SYNC_MULTI_RGBD=ON -DRTABMAP_SYNC_USER_DATA=ON`

[image]
[image]

My system configuration is:
Ubuntu 20.04
Ros Noetic

I have already tried the following solutions, but the error still remains:
- forcing cmake to compile with c++11, with the command `set(CMAKE_CXX_STANDARD 11)`
- install libpointmatcher, then remove the packages and rebuild all from the beginning
- `sudo apt remove ros-noetic-rtabmap ros-noetic-rtabmap-ros`, then catkin_make

I installed all the packages from source, so I guess that it's all set to the last version.

I don't know what else I can try, thank you for any suggestion.

sumitsarkar1, 2021-07-11:
@damianosa96 did you solve it? I am also facing same problem...I build Octomap from source and build rtabmap standalone with Octopmap and made sure `sudo apt remove ros-noetic-rtabmap` . Stil no luck

matlabbe, 2021-07-28:
It seems linking on a different version of rtabmap library. Try building with `catkin_make VERBOSE=1` to show to which libraries it is linking on. The function is there: https://github.com/introlab/rtabmap/blob/a5685c3e3104cb7f620fd5ad37e073521962b41e/corelib/src/OctoMap.cpp#L1031-L1038
and the buildfarm is building correctly:  https://travis-ci.com/github/introlab/rtabmap_ros/jobs/525757785