introlab/rtabmap_ros #601: Save a 3dpointcloud taken from Kinect 360 camera with rtabamp_ros

karishmathumu, 2021-06-30:
I need to save a entire 3d pointcloud and have a format or way to be able to export it onto Cloudcompare or Meshlab. 
I am using rtabmap_ros and a turtlebot2 with a Kinect camera.

I have a rosbag file which is saved version of the 3d mapping with the rtabmap_ros. I could visualise the pointcloud on RVIZ and RTABVIZ, but I am not sure how to save this pointcloud as a whole and be able to view and export it on to cloudcompare or meshlab applications.

A. On RTABMAPVIZ (does not work for me) " The option in RTABMAPVIZ with File->Export 3D Cloud" doesnot work for me, because all the options on RTABMAPVIZ seem to be disabled, though I have first selected "download all the clouds" option under Edit menu on thr RTabmapviz application.

And

B. I am saving a rosbag file and playing it with the goal to save the pointcloud with pcl_ros package. But all I could save is a set of multiple pcd files (kind of image files), which when viewed with pcl_viewer, looks like an image outline ( made of red dots) of a map , where as the actual pointcloud saving and viewing is the goal. Kindly please help me.

--On terminal (also does not work) rosrun pcl_ros pointcloud_to_pcd input:=/rtabmap/cloud_map and changing the boolean value to true
--All i could get are a red dotted outline on the ground view.

Kindly please help me out. I am on the verge of completing my project. This is one last thing to be done.

Enclosed are the screenshots for your reference.

Thank you very much,

Karishma.

[image]
[image]
[image]
[image]
[image]

karishmathumu, 2021-07-08:
I think I might identified the where the issue is, but not sure how to rectify it.  Any kind of help is greatly appreciated. 
I have realised the difference when a demo_mapping.bag file from http://wiki.ros.org/rtabmap_ros, and my own ros bag file is the these lines : **_[ INFO] [1625733351.613866953, 1368730028.617908510]: rtabmap (19): Rate=1.00s, Limit=0.000s, Conversion=0.0006s, RTAB-Map=0.0652s, Maps update=0.0088s pub=0.0008s (local map=17, WM=17)_**

Trial with demo bag file:
**1. rosbag play --clock demo_mapping.bag
2. roslaunch rtabmap_ros demo_robot_mapping.launch rviz:=true rtabmapviz:=true**
... logging to /home/tester/.ros/log/8810ab28-dfc1-11eb-ab04-34c93dbedb6f/roslaunch-tester-XMG-CORE-15-M20-GTX-1650xx-20424.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/tester/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://tester-XMG-CORE-15-M20-GTX-1650xx:38989/

SUMMARY
========

PARAMETERS
 * /points_xyzrgb/depth/image_transport: compressedDepth
 * /points_xyzrgb/queue_size: 10
 * /points_xyzrgb/rgb/image_transport: compressed
 * /points_xyzrgb/voxel_size: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /rtabmap/rtabmap/Grid/FromDepth: true
 * /rtabmap/rtabmap/Icp/CorrespondenceRatio: 0.2
 * /rtabmap/rtabmap/Icp/MaxCorrespondenceDistance: 0.05
 * /rtabmap/rtabmap/Icp/PM: false
 * /rtabmap/rtabmap/Icp/PointToPlane: false
 * /rtabmap/rtabmap/Icp/VoxelSize: 0.05
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/Mem/STMSize: 30
 * /rtabmap/rtabmap/RGBD/LocalRadius: 5
 * /rtabmap/rtabmap/RGBD/NeighborLinkRefining: true
 * /rtabmap/rtabmap/RGBD/OptimizeFromGraphEnd: false
 * /rtabmap/rtabmap/RGBD/OptimizeMaxError: 4
 * /rtabmap/rtabmap/RGBD/ProximityBySpace: true
 * /rtabmap/rtabmap/RGBD/ProximityByTime: false
 * /rtabmap/rtabmap/RGBD/ProximityPathMaxNeighbors: 10
 * /rtabmap/rtabmap/Reg/Force3DoF: true
 * /rtabmap/rtabmap/Reg/Strategy: 1
 * /rtabmap/rtabmap/Vis/MinInliers: 12
 * /rtabmap/rtabmap/depth/image_transport: compressedDepth
 * /rtabmap/rtabmap/frame_id: base_footprint
 * /rtabmap/rtabmap/odom_frame_id: odom
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/rgb/image_transport: compressed
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: True
 * /rtabmap/rtabmap/wait_for_transform: True
 * /rtabmap/rtabmapviz/depth/image_transport: compressedDepth
 * /rtabmap/rtabmapviz/frame_id: base_footprint
 * /rtabmap/rtabmapviz/rgb/image_transport: compressed
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_scan: True
 * /rtabmap/rtabmapviz/wait_for_transform: True
 * /use_sim_time: True

NODES
  /
    points_xyzrgb (nodelet/nodelet)
    rviz (rviz/rviz)
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

ROS_MASTER_URI=http://localhost:11311

[... 332 lines cut ...]
[ INFO] [1625733539.947438499]: /rtabmap/rtabmapviz: approx_sync   = true
[ INFO] [1625733539.947464827]: Setup depth callback
[ INFO] [1625733540.029959284]: 
/rtabmap/rtabmapviz subscribed to (approx sync):
   /az3/base_controller/odom \
   /data_throttled_image/compressed \
   /data_throttled_image_depth/compressedDepth \
   /data_throttled_camera_info \
   /jn0/base_scan
**[ INFO] [1625733540.030208451]: rtabmapviz started.**
[rviz-3] process has finished cleanly
log file: /home/tester/.ros/log/8810ab28-dfc1-11eb-ab04-34c93dbedb6f/rviz-3*.log
[ INFO] [1625734366.862885764]: rtabmap: Getting map (global=true optimized=true graphOnly=false)...
[ INFO] [1625734373.729446020]: rtabmap: Getting map (global=true optimized=true graphOnly=false)...
[ INFO] [1625734377.521129803]: rtabmapviz stopping spinner...
[ INFO] [1625734377.527014276]: rtabmapviz deleting qt stuff...
^C[points_xyzrgb-4] killing on exit
[rtabmap/rtabmapviz-2] killing on exit
[ INFO] [1625736022.656519729]: rtabmapviz: ctrl-c catched! Exiting Qt app...
[rtabmap/rtabmap-1] killing on exit
rtabmap: Saving database/long-term memory... (located at /home/tester/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/tester/.ros/rtabmap.db, 0 MB)
shutting down processing monitor...
... shutting down processing monitor
done

karishmathumu, 2021-07-08:
$ rosbag info demo_mapping.bag
path:        demo_mapping.bag
version:     2.0
duration:    4:05s (245s)
start:       May 16 2013 20:46:43.35 (1368730003.35)
end:         May 16 2013 20:50:49.28 (1368730249.28)
size:        295.0 MB
messages:    84745
compression: none [382/382 chunks]
types:       nav_msgs/Odometry           [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/LaserScan       [90c7ef2dc6895d81024acba2ac42f369]
             tf/tfMessage                [94810edda583a504dfda3829e70d7eec]
topics:      /az3/base_controller/odom                    11936 msgs    : nav_msgs/Odometry          
             /jn0/base_scan                                9843 msgs    : sensor_msgs/LaserScan      
             /tf                                          56909 msgs    : tf/tfMessage               
             data_throttled_camera_info                    2019 msgs    : sensor_msgs/CameraInfo     
             data_throttled_image/compressed               2019 msgs    : sensor_msgs/CompressedImage
             data_throttled_image_depth/compressedDepth    2019 msgs    : sensor_msgs/CompressedImage

whereas for my bagfile, 

 rosbag info rtabmap2.bag
path:        rtabmap2.bag
version:     2.0
duration:    5:47s (347s)
start:       Jun 28 2021 12:52:37.24 (1624877557.24)
end:         Jun 28 2021 12:58:24.77 (1624877904.77)
size:        21.6 GB
messages:    414449
compression: none [19805/19805 chunks]
types:       actionlib_msgs/GoalStatusArray          [8b2b82f13216d0a8ea88bd3af735e619]
             bond/Status                             [eacc84bf5d65b6777d4c50f463dfb9c8]
             diagnostic_msgs/DiagnosticArray         [60810da900de1dd6ddd437c3503511da]
             diagnostic_msgs/DiagnosticStatus        [d0ce08bc6e5ba34c7754f563a9cabaf1]
             dynamic_reconfigure/Config              [958f16a05573709014982821e6822580]
             dynamic_reconfigure/ConfigDescription   [757ce9d44ba8ddd801bb30bc456f946f]
             geometry_msgs/PolygonStamped            [c6be8f7dc3bee7fe9e8d296070f53340]
             geometry_msgs/PoseWithCovarianceStamped [953b798c0f514ff060a53a3498ce6246]
             geometry_msgs/Twist                     [9f195f881246fdfa2798d1d3eebca84a]
             kobuki_msgs/DockInfraRed                [a3d77f7d15e50579473dcc24e9b97b34]
             kobuki_msgs/RobotStateEvent             [c6eccd4cb1f95df95635b56d6226ea32]
             kobuki_msgs/SensorState                 [430a4bfd78449c8740bfef32b26613a6]
             kobuki_msgs/VersionInfo                 [3dc0ce6edd49be1fc86730f4a4a21861]
             map_msgs/OccupancyGridUpdate            [b295be292b335c34718bd939deebe1c9]
             nav_msgs/OccupancyGrid                  [3381f2d731d4076ec5c71b0759edbe4e]
             nav_msgs/Odometry                       [cd5e73d190d741a2f92e81eda573aca7]
             nav_msgs/Path                           [6227e2b7e9cce15051f669a5e197bbf7]
             octomap_msgs/Octomap                    [9a45536b45c5e409cd49f04bb2d9999f]
             rosgraph_msgs/Log                       [acffd30cd6b6de30f120938c17c593fb]
             rtabmap_ros/Info                        [1348655e70ba2e47fa77d7c83e936d33]
             rtabmap_ros/MapData                     [c08cdd9bd760e475724e80eaeb71af47]
             rtabmap_ros/MapGraph                    [903c2e44ed7e275e08ef323188cb38ee]
             sensor_msgs/BatteryState                [476f837fa6771f6e16e3bf4ef96f8770]
             sensor_msgs/CameraInfo                  [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage             [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Image                       [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu                         [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/JointState                  [3066dcd76a6cfaef579bd0f34173e9fd]
             sensor_msgs/LaserScan                   [90c7ef2dc6895d81024acba2ac42f369]
             sensor_msgs/PointCloud                  [d8e9c3f5afbdd8a130fd1d2763945fca]
             sensor_msgs/PointCloud2                 [1158d486dd51d683ce2f1be655c3c181]
             std_msgs/Int16MultiArray                [d9338d7f523fcb692fae9d0a0e9f067c]
             std_msgs/String                         [992ce8a1687cec8c8bd883ec73ca41d1]
             stereo_msgs/DisparityImage              [04a177815f75271039fa21f16acad8c9]
             tf2_msgs/TFMessage                      [94810edda583a504dfda3829e70d7eec]
             theora_image_transport/Packet           [33ac4e14a7cff32e7e0d65f18bb410f3]
             visualization_msgs/MarkerArray          [d155b9ce5188fbaf89745847fd5882d7]
topics:      /camera/camera_nodelet_manager/bond                                                              696 msgs    : bond/Status                             (2 connections)
[... 204 lines cut ...]
             /rtabmap/cloud_ground                                                                             69 msgs    : sensor_msgs/PointCloud2                
             /rtabmap/cloud_map                                                                                69 msgs    : sensor_msgs/PointCloud2                
[... cut ...]
             /rtabmap/scan_map                                                                                 68 msgs    : sensor_msgs/PointCloud2                
             /scan                                                                                           1099 msgs    : sensor_msgs/LaserScan                  
             /tf                                                                                            26007 msgs    : tf2_msgs/TFMessage                      (3 connections)
             /tf_static                                                                                         1 msg     : tf2_msgs/TFMessage

matlabbe, 2021-07-22:
Hi,

To export with rtabmapviz, you should click on "Pause" to enable Export Clouds menu option.

Your cloud_map seems created from 2d laser scan instead of RGB-D clouds. Set `Grid/FromDepth=true` explicitly if you want to create the map from RGB-D.