introlab/rtabmap_ros #602: Use rtabmap.db generated in handheld mode with external odometry later on

taletovic, 2021-07-02:
Hello Mathieu,

I have the following RTAB-Map setup on my robot:

- 2x D435 (one is placed lower and one on top of the robot, 90° rotated and there is a TF between these two) - mapping with two cameras works nice
- 1x 2D LiDAR LDS-01 (was included on TB3)
- Using wheel odometry coming from OpenCR board

Here is my launch file:

```
<launch>

   <group ns="d435_bottom">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager1" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync1" args="load rtabmap_ros/rgbd_sync nodelet_manager1">
      <remap from="rgb/image"       to="color/image_raw"/>
      <remap from="depth/image"     to="aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="color/camera_info"/>
      <param name="approx"          value="false"/>
    </node>
   </group>
   
   <group ns="d435_top">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager2" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync2" args="load rtabmap_ros/rgbd_sync nodelet_manager2">
      <remap from="rgb/image"       to="color/image_raw"/>
      <remap from="depth/image"     to="aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="color/camera_info"/>
      <param name="approx"          value="false"/>
    </node>
   </group>

  <group ns="rtabmap">

    <!-- Visual SLAM (robot side) -->
    <!-- args: "delete_db_on_start" and "udebug" -->
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="subscribe_scan" type="bool" value="true"/>
      <param name="rgbd_cameras"     type="int"    value="2"/>
      <param name="frame_id"         type="string" value="base_footprint"/>
      <!-- <param name="gen_scan"         type="bool"   value="true"/> -->
      <param name="map_negative_poses_ignored" type="bool"   value="false"/>        <!-- refresh grid map even if we are not moving-->
      <param name="map_negative_scan_empty_ray_tracing" type="bool" value="false"/> <!-- don't fill empty space between the generated scans-->

      <remap from="odom"          to="/odom"/>
      <remap from="rgbd_image0"   to="/d435_bottom/rgbd_image"/>
      <remap from="rgbd_image1"   to="/d435_top/rgbd_image"/>
      <remap from="scan"          to="/scan_filtered"/>

      <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
      <param name="RGBD/ProximityBySpace" type="string" value="true"/>
      <param name="RGBD/AngularUpdate" type="string" value="0.01"/>
      <param name="RGBD/LinearUpdate" type="string" value="0.01"/>
      <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
      <param name="Grid/FromDepth" type="string" value="false"/>
      <param name="Grid/CellSize" type="string" value="0.02"/>
      <param name="Vis/EstimationType" type="string" value="0"/> <!-- should be 0 for multi-cameras -->

      <param name="Reg/Force3DoF" type="string" value="true"/>
      <param name="Reg/Strategy" type="string" value="1"/>

      <param name="Icp/VoxelSize" type="string" value="0.05"/>
      <param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/>
    </node>

  </group>

</launch>
```

I had this idea to use one D435 on my laptop and generate a map using RGBD visual odometry in handheld mode and later on 
to transfer the generated rtabmap.db file to the previous mentioned setup. I was using the following commands from [RGB-D Handheld Mapping](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping):

`roslaunch realsense2_camera rs_camera.launch align_depth:=true`

```
roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="--delete_db_on_start" \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false
```

I tried it already but when I click on Download button it ignores the previously saved map. In case you ask, I removed `--delete_db_on_start` when I try to load the previously saved map.

I wanted to ask will it work that way before I invest more time trying to make it work?
If this concept would work I would move to the iOS RTAB-Map and try the same with an iPad. Thanks in advance.

Cheers, Mujo

matlabbe, 2021-07-22:
If you restart in SLAM mode with a previous database, it will start a new session in the same database. They will be disjoint at first, so you cannot Download the first session in rtabmapviz. When a loop closure is detected, the sessions will be merged, then the first map should reappear in rtabmapviz.

For example, if you launch in localziation mode:
```bash
roslaunch realsense2_camera rs_camera.launch align_depth:=true
roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="" \
    localization:=true \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false
```
You should be able to download the map in rtabmapviz.

taletovic, 2021-07-26:
Thank you for your answer.
However I think there was a misunderstanding.

When I use the same method you posted it works like a charm, here is a video that I've made:
[https://www.youtube.com/watch?v=Fbq1koe1030](https://www.youtube.com/watch?v=Fbq1koe1030)

I've built a bigger map using the same method, here is the .db file:
[d435_test_github.db](https://drive.google.com/file/d/1o7TSpJYLKMsZjTwYw9CTnWpFC6uFCw27/view?usp=sharing)

When I use this file on my robot with 2x D435 cameras it ignores the map when I set the `localization:=true`. Here is a video regarding this behaviour:
[https://www.youtube.com/watch?v=i0CceodixvU](https://www.youtube.com/watch?v=i0CceodixvU)

On the robot I am using external odometry from the robot wheels, could that be a problem ?

matlabbe, 2021-09-01:
It depends on how you launch rtabmap on the robot. Is localization argument exist in your launch file? I tried with demo robot mapping and your database, and your map stay there:
```bash
cp ~/Downloads/d435_test_github.db ~/.ros/rtabmap.db
roslaunch rtabmap_ros demo_robot_mapping.launch localization:=true
rosbag play --clock demo_mapping.bag
```
[image]

Not sure what is your target application, but if the goal is to use the map only with the robot, I suggest to do mapping with the robot, this will help for localization (I am pretty sure that the robot would not be able to localize in that map because of different camera point of view) and the map would be correctly aligned with the ground.