introlab/rtabmap_ros #608: Standalone Installation Error

jacklu333333, 2021-07-12:
Hi, I am trying to test different visual odom provided rtabmap. I tried to install directly from the ubuntu repo, but the output shows that I need to build the standalone version from the source. 
When I tried to install from the source, different errors occur for different package dependencies.  I was wondering is there recommend version of ubuntu for installation so that there would be less?

This is my environment configuration.
`Ubuntu: 18.04`
`ROS: melodic`

This is my CMake output.

```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- MOBILE_BUILD=OFF
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.1
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.9.1
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL_OCTREE: /usr/lib/x86_64-linux-gnu/libpcl_octree.so  
-- Found PCL_IO: /usr/lib/x86_64-linux-gnu/libpcl_io.so  
-- Found PCL_KDTREE: /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/lib/x86_64-linux-gnu/libpcl_search.so  
-- Found PCL_SURFACE: /usr/lib/x86_64-linux-gnu/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/lib/x86_64-linux-gnu/libpcl_filters.so  
-- Found PCL_2D: /usr/include/pcl-1.8  
-- Found PCL_FEATURES: /usr/lib/x86_64-linux-gnu/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/lib/x86_64-linux-gnu/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.8  
-- Found PCL_ML: /usr/lib/x86_64-linux-gnu/libpcl_ml.so  
-- Found PCL_SEGMENTATION: /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/lib/x86_64-linux-gnu/libpcl_visualization.so  
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
[... 309 lines cut ...]
     vertex->oplus(add_vertex.data());
                   ~~~~~~~~~~~^~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp: In instantiation of ‘void g2o::BaseFixedSizedEdge<D, E, VertexTypes>::linearizeOplusN() [with int N = 1; int D = 6; E = g2o::SE3Quat; VertexTypes = {g2o::VertexCam, g2o::VertexCam}]’:
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:191:41:   required from ‘void g2o::BaseFixedSizedEdge<D, E, VertexTypes>::linearizeOplusNs(std::index_sequence<Ints ...>) [with long unsigned int ...Ints = {0, 1}; int D = 6; E = g2o::SE3Quat; VertexTypes = {g2o::VertexCam, g2o::VertexCam}; std::index_sequence<Ints ...> = std::integer_sequence<long unsigned int, 0, 1>]’
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:199:19:   required from ‘void g2o::BaseFixedSizedEdge<D, E, VertexTypes>::linearizeOplus() [with int D = 6; E = g2o::SE3Quat; VertexTypes = {g2o::VertexCam, g2o::VertexCam}]’
/usr/local/include/g2o/types/sba/edge_sba_cam.h:39:25:   required from here
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:165:14: error: ‘FixedArray {aka class ceres::internal::FixedArray<double, 6>}’ has no member named ‘fill’
   add_vertex.fill(0.);
   ~~~~~~~~~~~^~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:172:30: error: ‘FixedArray {aka class ceres::internal::FixedArray<double, 6>}’ has no member named ‘data’
     vertex->oplus(add_vertex.data());
                   ~~~~~~~~~~~^~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:178:30: error: ‘FixedArray {aka class ceres::internal::FixedArray<double, 6>}’ has no member named ‘data’
     vertex->oplus(add_vertex.data());
                   ~~~~~~~~~~~^~~~
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1483: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o] Error 1
CMakeFiles/Makefile2:2404: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```
I also tried to switch all packages to the melodic version. However, the `Eigen3` and `Ceres` start to have compile errors.

Thanks in advance.

jacklu333333, 2021-07-14:
Nevermind.
I solved it by roll back to the kinetic version.
if anyone needs the step guides, leave the comments. I will share the steps.

closed 2021-07-14 by jacklu333333

matlabbe, 2021-07-30:
Note: The buildfarm is building with g2o on both Ubuntu bionic and focal: https://travis-ci.com/github/introlab/rtabmap_ros/builds/233240221