introlab/rtabmap_ros #610: rtabmap_ros faild to mappin with d435i

matlabmann, 2021-07-13:
hi ,i m a beginner in ros ,
so , i have installed ros kinetic on ubuntu 16.04 with rtabmap_ros (latest version) and realsens-ros-2.3.0 .and the camera is realsense D435i.
after that when i launch my launch file (roslaunch realsense2_camera opensource_tracking.launch) i have no map in rviz and rtabmapviz, and when i change the global frame in rviz from map to camera_link or anythink ,i can get a point claud but not a map.
so this is all my parametre attached here,
the launch file:
[opensource_tracking.txt](https://github.com/introlab/rtabmap_ros/files/6808906/opensource_tracking.txt)
the rtabmap.launch:
[rtabmap.txt](https://github.com/introlab/rtabmap_ros/files/6808918/rtabmap.txt)
the ros graph:
[image]
the frame graph:
[frames.pdf](https://github.com/introlab/rtabmap_ros/files/6808937/frames.pdf)

matlabbe, 2021-07-29:
`rgbd_odometry` node is no connected to camera topics. Do you have warnings in terminal telling that input topics are not synchronized? Is the `rtabmap.launch` file modified from the default one (here [kinetic version](https://github.com/introlab/rtabmap_ros/blob/kinetic-devel/launch/rtabmap.launch))?

matlabmann, 2021-07-29:
Hi @matlabbe you are right, the rgbd_odometry node is not connected,and it is died directly when i launch the file.
this is the output of the terminal,to see the warning msg,
thank your for replay,

>  roslaunch realsense2_camera opensource_tracking.launch 
... logging to /home/hassen/.ros/log/a3dbab92-f087-11eb-b62b-20689d328d50/roslaunch-thewalf-19584.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://thewalf:39665/

SUMMARY
========

CLEAR PARAMETERS
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: 63
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
[... 299 lines cut ...]
[ INFO] [1627575100.653560274]: Setup depth callback
[ INFO] [1627575100.731147246]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info
[ INFO] [1627575101.287851670]: rtabmap 0.20.12 started...
[ WARN] [1627575105.731486237]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info
[ WARN] [1627575110.731774216]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info

### note:i l not modify anything from the original rtabmap_ros launch file (xml) im just put sum modification in realsense launch file to adjust my camea name space or topics

matlabbe, 2021-09-02:
See this issue to debug "process has died" errors: https://github.com/introlab/rtabmap_ros/issues/28

SahanGura, 2022-03-03:
**Just wanted to clarify another issue in this implementation.**

You have not made use of the IMU data in this configuration. Because, IMU data and visual odometry are fused inside the ukf, but the output of ukf has not fed into the rtabmap node. rtabmap node should be given the filtered odometry (/odometry/filtered), not just the visual odometry (/rtabmap/odom) if you want to make use of the IMU. This is a common mistake in the community as I have identified. But, using an intermediate filter to fuse odometry is really difficult to configure. I am still trying to create an accurate map with fused odometry, than just using visual odometry. But unfortunately, rtabmap creates a distorted map with fused odometry.