introlab/rtabmap_ros #611: Detected not valid consecutive stamps New stamp should be always greater than previous stamp. This new data is ignored. This message will appear

eremteknoloji, 2021-07-13:
I am using realsense d435 camera on ubuntu 16.04 kinetic. When I run the roslaunch realsense2_camera opensource_tracking.launch line, I get the output of /rtabmap/odom at intervals. I get this error when I can't. How could it be the solution. Thanks.

matlabbe, 2021-07-29:
There is a problem with stamps set in image topics. They should be incrementing. It means that some images are skipped because they have the exact timestamp than the previous image. You can check the timestamps of the topics with `rostopic echo /camera/rgb/image_color_rect/header`

RalphCodesTheInternet, 2021-08-10:
I have the same problem when using a bag file to playback a scan. Using L515 with Noetic. Any idea on how to fix / bypass?

matlabbe, 2021-09-03:
If just by playing the bag there are indeed consecutive frames with same stamp, there was a problem with the camera publisher when it was recorded.