introlab/rtabmap_ros #613: Error building rtabmap_ros for ROS2 Foxy

Aditya02m, 2021-07-18:
Hi,
I'm really new here and this is my first post.
I'm using a live persistent SSD on which I freshly installed Ubuntu Focal 20.04 and ROS2 Foxy.
Trying to install rtabmap_ros for ROS2 but getting error
[image]

Aditya02m, 2021-07-18:
I solved that problem, it was due to lack of space on my RAM so I executed the code 'export MAKEFLAGS="-j1' but then another issue immediately rose.

[image]

matlabbe, 2021-07-30:
Make sure you cloned the ros2 branch of rtabmap_ros.

matlabbe, 2021-08-28:
Those are warnings, are there errors further down the line? Can they build to 100%?

MarcoAndrade26, 2021-08-29:
Hello @matlabbe this is the full error of rtabmap:

[image]
[image]

And this is error of rtabmap_ros:
[image]
[image]
[image]
[image]
[image]
[image]

Summary: 2 packages finished [1h 50min 29s]
  2 packages had stderr output: rtabmap rtabmap_ros

matlabbe, 2021-08-30:
I think "2 packages had stderr output: rtabmap rtabmap_ros" is a warning (I think colcon default settings make all warnings appearing as "error" but still build the packages), the packages have been built.

Lunatik00, 2021-09-19:
> I solved that problem, it was due to lack of space on my RAM so I executed the code 'export MAKEFLAGS="-j1' but then another issue immediately rose.
> 
> [image]

i am having a similar issue but with the other package, the output is:

Starting >>> rtabmap 
--- stderr: rtabmap
CMake Deprecation Warning at CMakeLists.txt:72 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Error at CMakeLists.txt:224 (FIND_PACKAGE):
  By not providing "FindPCL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PCL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PCL" (requested
  version 1.7) with any of the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< rtabmap [0.08s, exited with code 1]

Summary: 0 packages finished [0.20s]
  1 package failed: rtabmap
  1 package had stderr output: rtabmap
  1 package not processed

i am using the official dockerfile of ros foxy, so ubuntu 20.04 and ros 2 foxy, i have tried using the main branch and the foxy-devel branch for rtabmap so:

git clone https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
colcon build

and:

git clone --branch foxy-devel https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
colcon build

both have the exact same issue

Lunatik00, 2021-09-19:
i solved my issue, i had to use rosdep before colcon so:

rosdep update && rosdep install --from-paths src --ignore-src -r -y
colcon build

at least with that i was able to install, now i have to try the stereo cameras