introlab/rtabmap_ros #614: Error with catkin_make

jakb2403, 2021-07-20:
Hello, I have tried to build `rtabmap_ros` on this system:

- Ubuntu 20.04 LTS
- ROS Noetic

I used the following commands:
```
source /opt/ros/noetic/setup.bash

cd ~
git clone git@github.com:introlab/rtabmap.git rtabmap
cd rtabmap/build
cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..
make -j4
sudo make install

source ~/catkin_ws/devel/setup.bash

cd ~/catkin_ws
git clone git@github.com:introlab/rtabmap_ros.git src/rtabmap_ros
catkin_make
```
At the final step (`catkin_make`), I get the following error:
```
[ 79%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o
/home/john/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp: In constructor ‘MapsManager::MapsManager()’:
/home/john/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:72:16: error: invalid use of incomplete type ‘class rtabmap::OctoMap’
   72 |   octomap_(new OctoMap),
      |                ^~~~~~~
In file included from /home/john/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:27:
/home/john/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/MapsManager.h:40:7: note: forward declaration of ‘class rtabmap::OctoMap’
   40 | class OctoMap;
      |       ^~~~~~~
[ 79%] Linking CXX executable /home/john/catkin_ws/devel/lib/rtabmap_ros/camera
/home/john/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘rtabmap::Landmarks rtabmap_ros::landmarksFromROS(const std::map<int, geometry_msgs::PoseWithCovarianceStamped_<std::allocator<void> > >&, const string&, const string&, const ros::Time&, tf::TransformListener&, double, double, double)’:
/home/john/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:1625:101: warning: ‘rtabmap::Landmark::Landmark(const int&, const rtabmap::Transform&, const cv::Mat&)’ is deprecated: Use constructor with size=0 instead. [-Wdeprecated-declarations]
 1625 | sert(std::make_pair(iter->first, rtabmap::Landmark(iter->first, baseToTag, covariance)));
      |                                                                                      ^

In file included from /home/john/catkin_ws/devel/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/SensorData.h:42,
                 from /home/john/catkin_ws/devel/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Signature.h:42,
                 from /home/john/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:47,
                 from /home/john/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/home/john/catkin_ws/devel/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Landmark.h:80:8: note: declared here
   80 | inline Landmark::Landmark(const int & id, const Transform & pose, const cv::Mat & covariance) :
      |        ^~~~~~~~
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/build.make:76: rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 79%] Built target rtabmap_camera
make[1]: *** [CMakeFiles/Makefile2:5971: rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed

```

I have built using `catkin_make` on this same setup before and did not encounter this error. 
Your help would be much appreciated!
Thanks

matlabbe, 2021-07-20:
Make sure rtabmap and rtabmap_ros repos are updated to latest version. It builds correctly on my machine. Just saw that travis builds are not working for a while, we are correcting the problem.

jakb2403, 2021-07-21:
Thank you for you quick response @matlabbe. I ended up using these instructions instead and it worked for me.
```
$ source /opt/ros/noetic/setup.bash

$ cd ~
$ git clone git@github.com:introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake ..
$ make
$ sudo make install

$ source ~/catkin_ws/devel/setup.bash

$ cd ~/catkin_ws
$ git clone git@github.com:introlab/rtabmap_ros.git src/rtabmap_ros
$ catkin_make -j1
```

closed 2021-07-21 by jakb2403