introlab/rtabmap_ros #616: Rtabmap does not receive data

GiuliaB97, 2021-07-21:
**Behavior**
Launching a simulation which has a zed2 stereocamera I am able to publish both the odom topic (/gazebo/zed2/odom) and the rectified left and right image as well as left and right camera_info however i get the following error:

```shell_session
ruggio@Hologram:~$ roslaunch rtabmap_ros rtabmap.launch 
... logging to /home/ruggio/.ros/log/6669b6b8-e9fd-11eb-8619-087190d77481/roslaunch-Hologram-4073.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Hologram:39941/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: base_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 100
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_rgb: False
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: True
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)

ROS_MASTER_URI=http://localhost:11311

process[rtabmap/rtabmap-1]: started with pid [4089]
[ INFO] [1626856536.944783881]: Starting node...
[ INFO] [1626856536.969043590]: Initializing nodelet with 12 worker threads.
[ INFO] [1626856537.067242528]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1626856537.067281875]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1626856537.067299774]: /rtabmap/rtabmap(maps): map_cleanup                = true
[ INFO] [1626856537.067315873]: /rtabmap/rtabmap(maps): map_always_update          = false
[ INFO] [1626856537.067331392]: /rtabmap/rtabmap(maps): map_empty_ray_tracing      = true
[ INFO] [1626856537.067348361]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1626856537.067364320]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1626856537.067380319]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[... 554 lines cut ...]
  
  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
  </node>

</launch>

 ```

**My Environment**
- Ubuntu 18.04

closed 2021-07-25 by GiuliaB97

matlabbe, 2021-09-01:
Is it fixed? seems like timestamps between the cameras were not exactly the same.

GiuliaB97, 2021-09-02:
Yes, After a while i notice that in the simulation i was using rtabmap were already installed and running (even the documentation said nothing about it).
Therefore, i was basically trying to run it multiple times and this caused some problems in the system.
By stopping to do so everything works fine, thus i closed the issue.

Thank you very much.