introlab/rtabmap_ros #617: docker GPU support version and acceleration

jacklu333333, 2021-07-22:
Hi, I saw [this](https://github.com/introlab/rtabmap/issues/561) on the issue from rtabmap github repo. It says if I want to accelerate the processing speed with GPU. I can build the dependency library to increase them. However, I encounter some issues with opencv3 with my hardware at build time(old version OpenCV did not support new version architecture as mentioned [here](https://github.com/opencv/opencv/issues/9932)). 
And the new version of OpenCV will have a dependency with eigen3 and Ceres, if I update both of them all the dependency packages will need to update as well.

Therefore, I turn to the docker version for a solution. [This](https://github.com/introlab/rtabmap_ros/tree/master/docker) is the instruction I followed. As mentioned in the bottom half, if want to use GPU acceleration, I can follow the step. 
I am wondering the following

1.  Are they accelerate because of OpenCV or other dependency libraries?
2. What are versions of docker images support such acceleration?

Thanks in advance.

matlabbe, 2021-08-20:
This docker image uses nvidia as example to show OpenGL windows (just for visualization). Rtabmap or other libraries are not rebuilt with cuda support in that image.

I am not used to create cuda-enabled docker images, but if you are able to start from an [ubuntu-nvidia-cuda](https://hub.docker.com/r/nvidia/cuda) image, you may rebuild rtabmap from source and its dependencies (like opencv) while enabling cuda support for each dependency that can support it.

jacklu333333, 2021-08-23:
Got it, thanks for your reply.

closed 2021-08-23 by jacklu333333