introlab/rtabmap_ros #619: rtabmap with two d435i

aditdoshi333, 2021-07-26:
Hello,
I want to run two d435i with rtabmap. I tried (https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch) with some changes for d435i but I am not able to run. It would be great if someone can share the launch file for two d435i with rtabmap. 

Thank yu

taletovic, 2021-07-28:
Hello, you need to launch the cameras properly with a TF informations between them. If you don't have an URDF model you can just use static_transform_publisher like in this example you posted:

```
<node pkg="tf" type="static_transform_publisher" name="base_to_camera1_tf"
      args="0.0 0.0 0.0 0.0 0.0 0.0 /base_link /camera1_link 100" />
<node pkg="tf" type="static_transform_publisher" name="base_to_camera2_tf"
    args="-0.1325 -0.1975 0.0 -1.570796327 0.0 0.0 /base_link /camera2_link 100" />
```
Here is how to launch multiple D435 cameras:
```
<launch>
   
  <!-- Cameras -->
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="SERIALNUMBER_OF_YOUR_CAMERA" />
    <arg name="align_depth" value="true" />
    <arg name="camera" value="d435_bottom" />
    <arg name="filters" value="spatial" />
  </include>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="SERIALNUMBER_OF_YOUR_CAMERA" />
    <arg name="align_depth" value="true" />
    <arg name="camera" value="d435_top" />
    <arg name="filters" value="spatial" />
  </include>

</launch>
```
And finally, this is how you can start the mapping with RTAB-Map:
```
<launch>

   <group ns="d435_bottom">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager1" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync1" args="load rtabmap_ros/rgbd_sync nodelet_manager1">
      <remap from="rgb/image"       to="color/image_raw"/>
      <remap from="depth/image"     to="aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="color/camera_info"/>
      <param name="approx"          value="false"/>
    </node>
   </group>
   
   <group ns="d435_top">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager2" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync2" args="load rtabmap_ros/rgbd_sync nodelet_manager2">
      <remap from="rgb/image"       to="color/image_raw"/>
      <remap from="depth/image"     to="aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="color/camera_info"/>
      <param name="approx"          value="false"/>
    </node>
   </group>

  <group ns="rtabmap">

    <!-- Visual SLAM (robot side) -->
	
   <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen">
      <remap from="rgbd_image0"   to="/d435_bottom/rgbd_image"/>
      <remap from="rgbd_image1"   to="/d435_top/rgbd_image"/>
	  
      <param name="subscribe_rgbd"           type="bool"   value="true"/>
      <param name="frame_id"                 type="string" value="base_link"/>
      <param name="rgbd_cameras"             type="int"    value="2"/>
      <param name="wait_for_transform"       type="bool"   value="true"/>
      <param name="Odom/Strategy"            type="string" value="0"/> 
      <param name="OdomF2M/BundleAdjustment" type="string" value="0"/> <!-- should be 0 for multi-cameras -->
      <param name="Vis/EstimationType"      type="string" value="0"/> <!-- should be 0 for multi-cameras -->
      <param name="Vis/FeatureType"         type="string" value="6"/> 
      <param name="Vis/CorGuessWinSize"     type="string" value="0"/> 
      <param name="Vis/CorNNType"           type="string" value="3"/>
      <param name="Vis/MaxDepth"            type="string" value="4.0"/>  
      <param name="Vis/MinInliers"          type="string" value="20"/> 
      <param name="Vis/InlierDistance"      type="string" value="0.02"/>       
      <param name="OdomF2M/MaxSize" type="string" value="1000"/> 
      <param name="Odom/FillInfoData"        type="string" value="true"/>   
    </node>
	
    <!-- args: "delete_db_on_start" and "udebug" -->
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="subscribe_scan" type="bool" value="false"/>
      <param name="rgbd_cameras"     type="int"    value="2"/>
      <param name="frame_id"         type="string" value="base_link"/>
      <!-- <param name="gen_scan"         type="bool"   value="true"/> -->
      <param name="map_negative_poses_ignored" type="bool"   value="false"/>        <!-- refresh grid map even if we are not moving-->
      <param name="map_negative_scan_empty_ray_tracing" type="bool" value="false"/> <!-- don't fill empty space between the generated scans-->

      <remap from="odom"          to="/rtabmap/odom"/>
      <remap from="rgbd_image0"   to="/d435_bottom/rgbd_image"/>
      <remap from="rgbd_image1"   to="/d435_top/rgbd_image"/>

      <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
      <param name="RGBD/ProximityBySpace" type="string" value="true"/>
      <param name="RGBD/AngularUpdate" type="string" value="0.01"/>
      <param name="RGBD/LinearUpdate" type="string" value="0.01"/>
      <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
      <param name="Grid/FromDepth" type="string" value="true"/>
      <param name="Grid/CellSize" type="string" value="0.02"/>
      <param name="Vis/EstimationType" type="string" value="0"/> <!-- should be 0 for multi-cameras -->

      <param name="Reg/Force3DoF" type="string" value="false"/>
      <param name="Reg/Strategy" type="string" value="1"/>

      <param name="Icp/VoxelSize" type="string" value="0.05"/>
      <param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/>
    </node>

  </group>

</launch>
```

aditdoshi333, 2021-07-28:
Hello,

Thanks for the response. I tried the following launch file for two d435i

Launch file: 
[two_rs.txt](https://github.com/introlab/rtabmap_ros/files/6894609/two_rs.txt)

But there is an issue with odometry lost. It is very frequent with two realsense devices then one. Any light on the same? Any issue with the launch file?

taletovic, 2021-07-29:
I suppose you need more features in your mapping environment.

Try printing and adding some ArUco markers maybe?