introlab/rtabmap_ros #622: Not able to use compressed topic of realsense

aditdoshi333, 2021-08-03:
Hello,

For mapping, I am using intel realsense l515. Hello,

I want to record the .bag file for around 20 to 30 mins and running rtabmap ros on the same. If I record raw image topics like the following:
`
/camera/aligned_depth_to_color/image_raw  
/camera/color/image_raw/compressed 
/camera/color/camera_info
`
The size of the bag file is crossing ~200gb. I can't use such a huge bag file. So I am thinking to use compressed image topics like the following:

`
/camera/color/image_raw/compressed /camera/aligned_depth_to_color/image_raw/compressed /camera/color/camera_info
`
But I am getting some errors while running the realsense node

**Realsense Node**
`[ERROR] [1627983160.926561915]: Client [/rtabmap/rgbd_odometry] wants topic /camera/color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.926853260]: Client [/rtabmap/rgbd_odometry] wants topic /camera/aligned_depth_to_color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.929256854]: Client [/rtabmap/rtabmap] wants topic /camera/color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.929974768]: Client [/rtabmap/rtabmap] wants topic /camera/aligned_depth_to_color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.934051363]: Client [/rtabmap/rtabmapviz] wants topic /camera/color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.934547045]: Client [/rtabmap/rtabmapviz] wants topic /camera/aligned_depth_to_color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
`

**Rtabmap Node**

`[ WARN] [1627984776.467773886]: [image_transport] It looks like you are trying to subscribe directly to a transport-specific image topic '/camera/color/image_raw/compressed', in which case you will likely get a connection error. Try subscribing to the base topic '/camera/color/image_raw' instead with parameter ~image_transport set to 'compressed' (on the command line, _image_transport:=compressed). See http://ros.org/wiki/image_transport for details.
[ WARN] [1627984776.469969621]: [image_transport] It looks like you are trying to subscribe directly to a transport-specific image topic '/camera/aligned_depth_to_color/image_raw/compressed', in which case you will likely get a connection error. Try subscribing to the base topic '/camera/aligned_depth_to_color/image_raw' instead with parameter ~image_transport set to 'compressed' (on the command line, _image_transport:=compressed). See http://ros.org/wiki/image_transport for details.
`

The following are the launch cmds 

**Rtabmap**
`roslaunch rtabmap_ros rtabmap.launch     rtabmap_args:="--delete_db_on_start"     depth_topic:=/camera/aligned_depth_to_color/image_raw/compressed     rgb_topic:=/camera/color/image_raw/compressed     camera_info_topic:=/camera/color/camera_info     approx_sync:=false
`
**Realsense**

`roslaunch realsense2_camera rs_camera.launch align_depth:=true`

Any light is appreciated. 
Thank you

matlabbe, 2021-09-02:
For the depth image, you should use `compressedDepth` subtopic instead of `compressed`.

matlabbe, 2021-09-02:
Also, you should launch rtabmap.launch like this:
```
roslaunch rtabmap_ros rtabmap.launch \
   rtabmap_args:="--delete_db_on_start" \
   compressed:=true\
   depth_topic:=/camera/aligned_depth_to_color/image_raw \
   rgb_topic:=/camera/color/image_raw \
   camera_info_topic:=/camera/color/camera_info \
   approx_sync:=false 
```