introlab/rtabmap_ros #623: [rtabmap/rgbd_odometry-4] process has died [pid 16738, exit code -11, 　 /rtabmap/rtabmap: Did not receive data since 5 seconds!

xiaoqiuslambiji, 2021-08-09:
[rtabmap/rgbd_odometry-4] process has died [pid 16738, exit code -11, cmd /home/q/rtabmap_ros_catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay odom:=odom imu:=/imu/data __name:=rgbd_odometry __log:=/home/q/.ros/log/2c752b04-f8f8-11eb-a7c9-681401874ca7/rtabmap-rgbd_odometry-4.log].
log file: /home/q/.ros/log/2c752b04-f8f8-11eb-a7c9-681401874ca7/rtabmap-rgbd_odometry-4*.log

[ INFO] [1628506782.333386886]: rtabmap 0.20.7 started...
[ WARN] [1628506786.833814281]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info

need help

matlabbe, 2021-09-03:
See https://github.com/introlab/rtabmap_ros/issues/28 to get more information about why rgbd_odometry crashed

albdlackzrc, 2023-04-06:
Sorry I'm not familiar with GDB and ROS, but this is what rtabmap_ws/devel/lib/stereo_odometry looked like when I opened it with GDB.
[image]
So how do I get this up and running, thank you for your answer