introlab/rtabmap_ros #627: Able to load map immediately at localization start-up and odometry loses track easily but never recover

wrmichaelmak, 2021-08-12:
Hi, I have managed to successfully build RTAB-MAP with ORB SLAM3 support. I have also managed to map an environment and obtain a grid map that I am satisfied with. However, when I set it to localization mode, the map loads up immediately and is displayed on rviz without me doing any loop closure. I don't think that is what it should be happening. Additionally, I realise that the odometry is easily lost and I am not able to recover from it. 

I am using a D435i to do my RGB-D mapping. 
```
<!-- -->
<launch>
  <!-- Convenience launch file to launch odometry, rtabmap and rtabmapviz nodes at once -->

  <!-- For stereo:=false
        Your RGB-D sensor should be already started with "depth_registration:=true".
        Examples:
           $ roslaunch freenect_launch freenect.launch depth_registration:=true 
           $ roslaunch openni2_launch openni2.launch depth_registration:=true -->
           
  <!-- For stereo:=true
        Your camera should be calibrated and publishing rectified left and right 
        images + corresponding camera_info msgs. You can use stereo_image_proc for image rectification.
        Example:
           $ roslaunch rtabmap_ros bumblebee.launch -->
     
  <!-- Choose between depth and stereo, set both to false to do only scan -->      
  <arg name="stereo"                    default="false"/>
  <arg     if="$(arg stereo)" name="depth"  default="false"/>
  <arg unless="$(arg stereo)" name="depth"  default="true"/>
 
  <!-- Choose visualization -->
  <arg name="rtabmapviz"              default="false" /> 
  <arg name="rviz"                    default="true" />
  
  <!-- Localization-only mode -->
  <arg name="localization"            default="true"/>
  
  <!-- sim time for convenience, if playing a rosbag -->
  <arg name="use_sim_time"            default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>
  
  <!-- Corresponding config files -->
  <arg name="cfg"                     default="" /> <!-- To change RTAB-Map's parameters, set the path of config file (*.ini) generated by the standalone app -->
  <arg name="gui_cfg"                 default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="$(find rtabmap_ros)/launch/config/rgbd.rviz" />
  
  <arg name="frame_id"                default="camera_link"/>     <!-- Fixed frame id, you may set "base_link" or "base_footprint" if they are published -->
  <arg name="odom_frame_id"           default=""/>                <!-- If set, TF is used to get odometry instead of the topic -->
  <arg name="odom_frame_id_init"      default=""/>                <!-- If set, TF map->odom is published even if no odometry topic has been received yet. The frame id should match the one in the topic. -->
  <arg name="map_frame_id"            default="map"/>
  <arg name="ground_truth_frame_id"   default=""/>     <!-- e.g., "world" -->
  <arg name="ground_truth_base_frame_id" default=""/>  <!-- e.g., "tracker", a fake frame matching the frame "frame_id" (but on different TF tree) -->
  <arg name="namespace"               default="rtabmap"/>
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="queue_size"              default="10"/>
  <arg name="wait_for_transform"      default="0.2"/>
  <arg name="args"                    default=""/>              <!-- delete_db_on_start, udebug -->
  <arg name="rtabmap_args"            default="$(arg args)"/>   <!-- deprecated, use "args" argument -->
  <arg name="gdb"                     default="false"/>         <!-- Launch nodes in gdb for debugging (apt install xterm gdb) -->
  <arg     if="$(arg gdb)" name="launch_prefix" default="xterm -e gdb -q -ex run --args"/>
  <arg unless="$(arg gdb)" name="launch_prefix" default=""/>
  <arg name="output"                  default="screen"/>        <!-- Control node output (screen or log) -->
  <arg name="publish_tf_map"          default="true"/>

  <!-- if timestamps of the input topics are synchronized using approximate or exact time policy-->
  <arg     if="$(arg stereo)" name="approx_sync"  default="false"/>
  <arg unless="$(arg stereo)" name="approx_sync"  default="$(arg depth)"/>         
   
  <!-- RGB-D related topics -->
  <arg name="rgb_topic"               default="/camera/color/image_raw" />
  <arg name="depth_topic"             default="/camera/aligned_depth_to_color/image_raw" />
  <arg name="camera_info_topic"       default="/camera/color/camera_info" />
  <arg name="depth_camera_info_topic" default="/camera/aligned_depth_to_color/camera_info" />
  
  <!-- stereo related topics -->
[... 348 lines cut ...]
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
  </node>
  
  <node name="imu_filter_madgwick" pkg="imu_filter_madgwick" type="imu_filter_node">
  
  	<remap from="/imu/data_raw" to="/camera/imu"/>
  	<remap from="/imu/data" to="/rtabmap/imu"/>
  
	<param name="use_mag" value="false"/>
	<param name="publish_tf" value="false"/>
	<param name="world_frame" value="enu"/>	
  </node>

</launch>
```

matlabbe, 2021-09-03:
Hi, when using rtabmap.launch, include it in another launch file to make it easier to see which parameters you changed. The trick to change odometry strategy is to use the arguments interface:
```xml
<launch>
<arg name="localization" default="false"/>
<include file="$(find rtabmap_ros)/launch/rtabmap.launch">
   <arg     if="$(arg localization)" name="args" value="--Optimizer/Strategy 2 --g2o/Solver 3 --Odom/Strategy 5 --OdomORBSLAM/VocPath /home/nuc-develop/build/ORB_SLAM3/Vocabulary/ORBvoc.txt"/>
   <arg unless="$(arg localization)" name="args" value="-d --Optimizer/Strategy 2 --g2o/Solver 3 --Odom/Strategy 5 --OdomORBSLAM/VocPath /home/nuc-develop/build/ORB_SLAM3/Vocabulary/ORBvoc.txt"/>
   <arg name="localization" value="$(arg localization)"/>
   ...
</include>
</launch>
```
or even show an example launching directly rtabmap.launch:
```bash
roslaunch rtabmap_ros rtabmap.launch \
   args:="-d \
      --Optimizer/Strategy 2 \
      --g2o/Solver 3 \
      --Odom/Strategy 5 \
      --OdomORBSLAM/VocPath /home/nuc-develop/build/ORB_SLAM3/Vocabulary/ORBvoc.txt"
```

Not showing the map on localization mode at the beginning was the default behavior in previous rtabmap versions. However, a lot of people were confused on why the map was not showing up, requiring me to explain everytime that is it because the robot is not localized in the map yet.

The new default initial behavior in localization mode is to restart the robot at the last location from the previous session. So if you close rtabmap, then restart it in localization mode without moving the robot while it was offline, the robot will restart correctly at its last known location. With this new behavior, proximity detections can work already without having a global localization (in case the robot is seeing a white wall, it can still relocalize with its lidar). If the robot has been moved (kidnapped robot problem), the robot still think it is restarting from the last saved location, but it can still recover when a global localization occurs.

In other applications, it is known that the robot is always restarting from the origin of the map, for such case, we can set parameter `RGBD/StartAtOrigin` to `true`. Again, if the robot was not really started at the origin, a global localization should occur to relocalize correctly the robot.

To simulate the old behavior to know when the robot is correctly localized (before if the map doesn't exist, it meant we are not yet localized), set `RGBD/StartAtOrigin` to true, then watch `/rtabmap/mapGraph` message until `mapToOdom` is not Identity matrix (or when tf /map ->/odom is not Identity anymore), so you will know until then, that even you are receiving a map, you know you are not yet localized in it.

cheers,
Mathieu

wrmichaelmak, 2021-09-06:
Thanks a lot! I got it now.

closed 2021-09-06 by wrmichaelmak